h2zero/NimBLE-Arduino #13: Migrating ESP32-BLECollector

tobozo, 2020-04-20:
hello and thank you for this great library!

It totally removed the burden of having to deal with "does not fit in ram" situations.

So after migrating my code to the new NimBLE syntax, I found out I was initially using a modified version of the official BLE library and manually applied a few patches.

I was pleasantly surprised to see that most of those changes are already adressed in your library :+1: 

However I couldn't find a matching method for one missing method `pChar->getDataLength()`.

In my project getDataLength() is used for file transfer situations with variable MTU.

In order to get my sketch working and go on with more tests, I added this to `src/NimBLECharacteristic.h`.

    size_t getDataLength() { return m_value.getLength(); }

So my question is: do I really need to modify `src/NimBLECharacteristic.h` to get the lenght of the data or is there a way to do that more elegantly ?

h2zero, 2020-04-20:
That's great, i'm glad you find it useful 😄 . 
As this project was derived mainly from the original library the only intent so far was to replicate it for NimBLE as best as possible and make changes/fixes that I thought made sense. 
As there was not existing API for that particular function in the original I have not created a more elegant solution. 

I'd be glad to add it as a feature as others may find it useful also. I'm currently working on an update to address a few other issues as well so I could incorporate this at the same time. Feel free to submit a PR if you'd like.

tobozo, 2020-04-20:
PR #14 it is :-)

closed 2020-04-22 by tobozo