h2zero/NimBLE-Arduino #16: Type redefinition errors on latest Arduino IDE master

Jeroen88, 2020-04-28:
First: compliments for the excellent work to reduce the memory footprint for BLE!

Today I updated my ESP32 Arduino IDE to the latest master, and I got compilation errors (type redefinition errors). The cause was in NimbleAddress.h:

```
typedef enum {
    BLE_ADDR_TYPE_PUBLIC        = 0x00,
    BLE_ADDR_TYPE_RANDOM        = 0x01,
    BLE_ADDR_TYPE_RPA_PUBLIC    = 0x02,
    BLE_ADDR_TYPE_RPA_RANDOM    = 0x03,
} esp_nimble_addr_type_t;

typedef uint8_t esp_nimble_addr_type_t ;

```
I changed it to:

```
typedef enum {
    NIMBLE_ADDR_TYPE_PUBLIC        = 0x00,
    NIMBLE_ADDR_TYPE_RANDOM        = 0x01,
    NIMBLE_ADDR_TYPE_RPA_PUBLIC    = 0x02,
    NIMBLE_ADDR_TYPE_RPA_RANDOM    = 0x03,
} esp_nimble_addr_type_t;

//typedef uint8_t esp_nimble_addr_type_t ;

```
Now I also had to change NimBLEClient.h into:

`bool                                       connect(NimBLEAddress address, esp_nimble_addr_type_t type = NIMBLE_ADDR_TYPE_PUBLIC, bool refreshServices = true);   // Connect to the remote BLE Server
`

and NimBLEClient.cpp:

`bool NimBLEClient::connect(NimBLEAddress address, esp_nimble_addr_type_t type, bool refreshServices) {
`
and lastly in this file:
`
    return connect(address, (esp_nimble_addr_type_t)type, refreshServices);
`

h2zero, 2020-04-28:
Thanks for reporting this I'll have a look at it soon. Those typedefs only exist for compatibility and I'm not even sure they are necessary, with any luck they can be removed completely as I'd prefer not to use esp specific code at all if possible.

h2zero, 2020-04-28:
Had a quick look into this and I don't see any issue unless `#include "esp_bt_defs.h" is included in your sketch. So unless there is some need for that include, you can remove it,  I will put a guard on the defines later.

thorrak, 2020-04-28:
> Had a quick look into this and I don't see any issue unless `#include "esp_bt_defs.h" is included in your sketch. So unless there is some need for that include, you can remove it, I will put a guard on the defines later.

I think the issue is that this file seemingly gets included by default when including the WiFi/HTTPClient clients:

```
In file included from /Users/jbeeler/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/bt/esp_gap_ble_api.h:22:0,
                 from /Users/jbeeler/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/protocomm/protocomm_ble.h:17,
                 from /Users/jbeeler/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/wifi_provisioning/wifi_provisioning/scheme_ble.h:18,
                 from /Users/jbeeler/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiProv.h:21,
                 from /Users/jbeeler/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:40,
                 from /Users/jbeeler/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.h:25,
                 from /Users/jbeeler/.platformio/packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.h:35,
                 from src/SecureWithRedirects.h:8,
                 from src/SecureWithRedirects.cpp:6:
/Users/jbeeler/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/bt/esp_bt_defs.h:109:35: error: redeclaration of 'BLE_ADDR_TYPE_PUBLIC'

```

Jeroen88, 2020-04-28:
The errors appear if I #include <Arduino.h> or #include <WiFi.h>. I do not #include "esp_bt_defs.h

h2zero, 2020-04-28:
Thanks, good to know, I'll try to get this fixed today.

thorrak, 2020-04-28:
_A_ fix is available on the branch GitHub linked above (renaming the defines as @Jeroen88 suggested) if you want to pull it and be done with it, though I can't comment as to whether it's the _right_ fix... :)

h2zero, 2020-04-28:
@thorrak Thanks, the reason these are there is for backward compatibility if someone used them in their sketch from the old Library so I needed to keep the names the same. I'm thinking of just eliminating them instead and just directing users to use the NimBLE types.

h2zero, 2020-04-28:
Fixed in commit ff99200.

Jeroen88, 2020-04-29:
Fixed by PR

closed 2020-04-29 by Jeroen88