h2zero/NimBLE-Arduino #17: Feature request on NimBLEUUID

Jeroen88, 2020-04-28:
Could you please add to NimBLEUUID.h:
```
    NimBLEUUID(uint32_t first, uint16_t second, uint16_t third, uint64_t fourth);
```
and to NimBLEUUID.cpp:

```
NimBLEUUID::NimBLEUUID(uint32_t first, uint16_t second, uint16_t third, uint64_t fourth) {
    m_uuid.u.type        = BLE_UUID_TYPE_128;
    memcpy(m_uuid.u128.value + 12, &first,  4);
    memcpy(m_uuid.u128.value + 10, &second, 2);
    memcpy(m_uuid.u128.value + 8,  &third,  2);
    memcpy(m_uuid.u128.value,      &fourth, 8);
    m_valueSet = true;
}
```
This does not save any library memory, but is does save sketch memory. In stead of
```
BLEUUID("ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6")
```
you can do
```
BLEUUID(0xebe0ccb0, 0x7a0a, 0x4b0c, 0x8a1a6ff2997da3a6)
```
thus trading 37 bytes for the string literal for 16 bytes uint literals, saving 21 bytes _per UUID literal!_

h2zero, 2020-04-28:
Looks good to me, could you make a PR for this?

Jeroen88, 2020-04-28:
Maybe I will make PRs later, for now I just copied the library to my library folder and dit not set up git for it. For now it would be easy if you integrate it :)

Jeroen88, 2020-04-28:
Made PR #21

closed 2020-04-28 by Jeroen88