h2zero/NimBLE-Arduino #18: Crash on BLEDevice::deleteClient(pClient)

Jeroen88, 2020-04-28:
I am trying to reuse the client, so I call `BLEDevice::deleteClient(pClient)`. This causes a crash:

```
Decoding stack results
0x4008e320: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008e59d: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x400efb6b: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x400837cf: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 267
0x4008307d: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 42
0x4017de81: operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc line 46
0x400d6465: NimBLEClientCallbacks::~NimBLEClientCallbacks() at /home/jeroen/Arduino/libraries/NimBLE-Arduino-master/src/NimBLEClient.h line 101
0x400d6be5: NimBLEClient::~NimBLEClient() at /home/jeroen/Arduino/libraries/NimBLE-Arduino-master/src/NimBLEClient.cpp line 78
0x400d73e6: NimBLEDevice::deleteClient(NimBLEClient*) at /home/jeroen/Arduino/libraries/NimBLE-Arduino-master/src/NimBLEDevice.cpp line 160
0x400d2bf1: readBLEClients() at /home/jeroen/Arduino/test_BLE_02/test_BLE02/test_BLE02.ino line 781
0x400d2c4e: loop() at /home/jeroen/Arduino/test_BLE_02/test_BLE02/test_BLE02.ino line 788
0x400ed9c5: loopTask(void*) at /home/jeroen/Downloads/arduino-1.8.10-linux64/arduino-1.8.10/hardware/espressif/esp32/cores/esp32/main.cpp line 19
0x40093555: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```
This is because `BLEClient::m_deleteCallbacks` is default `true` and `BLEClient::BLEClient()` sets the pointer `m_pClientCallbacks` to `&defaultCallbacks`. So before deleting this pointer, it should be checked that it does not point to `&defaultCallbacks`:

```
NimBLEClient::~NimBLEClient() { 
    // We may have allocated service references associated with this client.  
    // Before we are finished with the client, we must release resources.
    clearServices();
    
    if(m_deleteCallbacks && m_pClientCallbacks != &defaultCallbacks) {
        delete m_pClientCallbacks;
    }

} // ~NimBLEClient
```

h2zero, 2020-04-28:
Thanks! I totally forgot about this when I added the default callbacks. Will fix asap.

One comment, if you are just wanting to connect the client instance to a different device you don't need to delete the client, just call connect with `refreshServices = true`.

Jeroen88, 2020-04-28:
Thnx for the comment, I can use it! Still I need to clean up everything, to use the ESP for other tasks after reading the BLE peripherals.

h2zero, 2020-04-28:
Fixed in commit 846d690

Thanks!

Jeroen88, 2020-04-29:
Fixed by commit

closed 2020-04-29 by Jeroen88