h2zero/NimBLE-Arduino #19: Question

Jeroen88, 2020-04-28:
@h2zero I am new to BLE, and I think this is more a question than an issue, but maybe you can help.

When I read a remote characteristic with the notify callback everything is okay, but if I try to read it directly, I do not get the right values, while using the same address, service UUID and characteristic UUID. Am I forgetting something? This is the snippet:

```
void readBLEClient() {
  BLEClient * pClient = BLEDevice::createClient();

  Serial.println("Connecting...");
  bool success = pClient->connect(BLEAddress("a4:c1:38:5d:ef:16"));
  if(!success) {
    BLEDevice::deleteClient(pClient);

    Serial.println("Failed to connect");
    return;
  }
  Serial.println("Connected");
  BLERemoteService * pRemoteService = pClient->getService(BLEUUID(0xebe0ccb0, 0x7a0a, 0x4b0c, 0x8a1a6ff2997da3a6));
  if(pRemoteService == nullptr) {
    Serial.print("Failed to find our service UUID: ");
    pClient->disconnect();
  } else {
    Serial.println(" - Found our service");  
    // Obtain a reference to the characteristic in the service of the remote BLE server.
    BLERemoteCharacteristic * pRemoteCharacteristic = pRemoteService->getCharacteristic(BLEUUID(0xebe0ccc1, 0x7a0a, 0x4b0c, 0x8a1a6ff2997da3a6));
    if(pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
    } else {
      Serial.println(" - Found our characteristic");
      if(pRemoteCharacteristic->canRead()) {
        Serial.println("Can read");
        Serial.printf("string value is '%s'\n", pRemoteCharacteristic->readValue().c_str());
        Serial.printf("uint8_t value is %u\n", pRemoteCharacteristic->readUInt8());
        Serial.printf("uint16_t value is %u\n", pRemoteCharacteristic->readUInt16());
        const uint8_t * pData = pRemoteCharacteristic->readRawData();
        float temperature = (pData[0] | (pData[1] << 8)) * 0.01; //little endian 
        uint8_t humidity = pData[2];
        Serial.printf("Temperature = %.2f : humidity = %u\n", temperature, humidity);    
      }
    }
  }
  pClient->disconnect();
  BLEDevice::deleteClient(pClient);
}

```

The output is an empty string. If I test the length() and the contents of string returned by readValue(), it is 3 long all 0x00. In the notify callback pData is 5 long, containing 2 bytes temperature, 1 byte humidity and two (for me) unknown bytes.

Any suggestions?

h2zero, 2020-04-28:
Not sure why this is not working for you, I will test and get back to you.

Jeroen88, 2020-04-29:
Closing, taken over by #20

closed 2020-04-29 by Jeroen88