h2zero/NimBLE-Arduino #20: BLERemoteCharacteristic::readValue() does not return the correct value

Jeroen88, 2020-04-28:
I would love to get this library running, but I think I found another issue...

I already thought it was my problem so I asked question #19 today, but it is an error of the library.

MVCE:

```
#include <NimBLEDevice.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("/n/nStarted");

  BLEDevice::init("ESP32");

  readBLEClients();
}

void readBLEClients() {
// LYWSD03MMC
  static BLEClient * pClient = BLEDevice::createClient();

  Serial.println("Connecting...");
  bool success = pClient->connect(BLEAddress("a4:c1:38:5d:ef:16"));
  if(!success) {
    BLEDevice::deleteClient(pClient);

    Serial.println("Failed to connect");
    return;
  }
  Serial.println("Connected");

  BLERemoteService * pRemoteService = pClient->getService(BLEUUID("ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6"));
  if(pRemoteService == nullptr) {
    Serial.print("Failed to find our service UUID: ");
  } else {
    Serial.println(" - Found our service");

    // Obtain a reference to the characteristic in the service of the remote BLE server.
    BLERemoteCharacteristic * pRemoteCharacteristic = pRemoteService->getCharacteristic(BLEUUID("ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6"));
    if(pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
    } else {
      Serial.println(" - Found our characteristic");
      if(pRemoteCharacteristic->canRead()) {
        Serial.println("Can read");
//        Serial.printf("string value is '%s'\n", pRemoteCharacteristic->readValue().c_str());
//        Serial.printf("uint8_t value is %u\n", pRemoteCharacteristic->readUInt8());
//        Serial.printf("uint16_t value is %u\n", pRemoteCharacteristic->readUInt16());
//        const uint8_t * pData = pRemoteCharacteristic->readRawData();
        const uint8_t * pData = reinterpret_cast<const uint8_t*>(pRemoteCharacteristic->readValue().c_str());

        float temperature = (pData[0] | (pData[1] << 8)) * 0.01; //little endian 
        uint8_t humidity = pData[2];
        Serial.printf("LYWSD03MMC temperature = %.2f : humidity = %u\n", temperature, humidity);    
      }
      if(pRemoteCharacteristic->canNotify()) {
        Serial.println("Can notify");
//        pRemoteCharacteristic->registerForNotify(notifyCallback);
      }
    }
  }
  pClient->disconnect();
  BLEDevice::deleteClient(pClient);
}

void loop() {
}
```
returns 0.00 for temperature and 0 for humidity.

Digging for hours in the library resulted in:

[Here](https://github.com/h2zero/NimBLE-Arduino/blob/15448515d4bda4a042929f7b6546917d8df6e75a/src/NimBLEClient.cpp#L672) the data is still correct. If I add the following line at this exact position:

`characteristic->second->m_value = std::string((char*) event->notify_rx.om->om_data, event->notify_rx.om->om_len);
`

And comment out [this](https://github.com/h2zero/NimBLE-Arduino/blob/15448515d4bda4a042929f7b6546917d8df6e75a/src/NimBLERemoteCharacteristic.cpp#L395) line (otherwise the value just sets gets overwritten), the MVCE actually works correctly.

HOWEVER it does not feel good. My gut feeling says that `onReadCB` should indeed read the value from `struct ble_gatt_attr *attr`, but that value is not correct. At least the data from event->notify_rx.om in NimBLEClient differs from attr->om in NimbleRemoteCharacteristic.

I am not familiair enough with BLE and the library to solve this, and I am stuck now, any suggestions would be very welcome!.

h2zero, 2020-04-28:
Troubling indeed, would it be possible to see a log?

Jeroen88, 2020-04-28:
```
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
Connecting...
D NimBLEClient: ">> connect(a4:c1:38:5d:ef:16)"
D FreeRTOS: "Semaphore taking: name: OpenEvt (0x3ffc8b0c), owner: <N/A> for connect"
D FreeRTOS: "Semaphore taken:  name: OpenEvt (0x3ffc8b0c), owner: connect"
D FreeRTOS: ">> wait: Semaphore waiting: name: OpenEvt (0x3ffc8b0c), owner: connect for connect"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_CONNECT "
D NimBLEClient: "Connection established"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8b0c), owner: connect"
D FreeRTOS: "<< wait: Semaphore released: name: OpenEvt (0x3ffc8b0c), owner: <N/A>"
D NimBLEClient: "Refreshing Services for: (a4:c1:38:5d:ef:16)"
D NimBLEClient: ">> clearServices"
D NimBLEClient: "<< clearServices"
D NimBLEClient: ">> retrieveServices"
D FreeRTOS: "Semaphore taking: name: SearchCmplEvt (0x3ffc8b6c), owner: <N/A> for retrieveServices"
D FreeRTOS: "Semaphore taken:  name: SearchCmplEvt (0x3ffc8b6c), owner: retrieveServices"
D FreeRTOS: ">> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffc8b6c), owner: retrieveServices for retrieveServices"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_MTU"
I NimBLEClient: "mtu update event; conn_handle=0 mtu=23"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8b0c), owner: <N/A>"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 14 handle: 0"
D FreeRTOS: "Semaphore giving: name: SearchCmplEvt (0x3ffc8b6c), owner: retrieveServices"
D NimBLEClient: "<< Service Discovered. status: 0"
D FreeRTOS: "<< wait: Semaphore released: name: SearchCmplEvt (0x3ffc8b6c), owner: <N/A>"
D NimBLERemoteService: ">> retrieveCharacteristics() for service: 00000100-0065-6c62-2e74-6f696d2e696d"
D FreeRTOS: "Semaphore taking: name: GetCharEvt (0x3ffc96d4), owner: <N/A> for retrieveCharacteristics"
D FreeRTOS: "Semaphore taken:  name: GetCharEvt (0x3ffc96d4), owner: retrieveCharacteristics"
D FreeRTOS: ">> wait: Semaphore waiting: name: GetCharEvt (0x3ffc96d4), owner: retrieveCharacteristics for retrieveCharacteristics"
D NimBLERemoteService: "Characteristic Discovered >> status: 0 handle: 0"
D NimBLERemoteService: "<< Characteristic Discovered. status: 0"
D NimBLERemoteService: "Characteristic Discovered >> status: 0 handle: 0"
D NimBLERemoteService: "<< Characteristic Discovered. status: 0"
D NimBLERemoteService: "Characteristic Discovered >> status: 14 handle: 0"
D FreeRTOS: "Semaphore giving: name: GetCharEvt (0x3ffc96d4), owner: retrieveCharacteristics"
D NimBLERemoteService: "<< Characteristic Discovered. status: 0"
D FreeRTOS: "<< wait: Semaphore released: name: GetCharEvt (0x3ffc96d4), owner: <N/A>"
[... 470 lines cut ...]
D NimBLEClient: "Got Client event BLE_GAP_EVENT_NOTIFY_RX"
D NimBLEClient: "Notify Recieved for handle: 54"
D NimBLEClient: "checking service 00000100-0065-6c62-2e74-6f696d2e696d for handle: 54"
D NimBLEClient: "checking service 0xfe95 for handle: 54"
D NimBLEClient: "checking service ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6 for handle: 54"
D NimBLEClient: "Got Notification for characteristic Characteristic: uuid: ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6, handle: 54 0x0036, props:  0x12
Descriptor: uuid: 0x2901, handle: 55
Descriptor: uuid: 0x2902, handle: 56"
D NimBLEClient: "De ontvangen data (5 bytes) is '@⸮
', hum is 64, temp is 2051"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D NimBLERemoteCharacteristic: "De data hier (5 bytes) is '@⸮
'"
D FreeRTOS: "Semaphore giving: name: ReadCharEvt (0x3ffcfaa4), owner: readValue"
D FreeRTOS: "<< wait: Semaphore released: name: ReadCharEvt (0x3ffcfaa4), owner: <N/A>"
D NimBLERemoteCharacteristic: "<< readValue(): length: 5"
LYWSD03MMC temperature = 20.51 : humidity = 64
Can notify
D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
D NimBLEClient: ">> clearServices"
D NimBLEClient: "<< clearServices"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_DISCONNECT"

```

h2zero, 2020-04-28:
I'm so far unable to reproduce the issue. I've setup a peripheral device with static values similar to what you are trying to read and am able to read them on a client just as they should be. 

Maybe the peripheral is doing something and responding with 0 values?

I don't see anything wrong in your log, I'm assuming this is after you made changes?

Jeroen88, 2020-04-28:
@h2zero Thnx for helping me!

Yes the log is after the changes, but the only difference is the two Dutch sentences near the end. 

> Maybe the peripheral is doing something and responding with 0 values?

I don't think so, because the data read from the peripheral is correct. It is not passed correctly to the remote characteristic...

h2zero, 2020-04-28:
Interesting how you are recieving notifications before you have registered for them. It also seems you get one right when you happen to be reading the value. I wonder if there is something going on there.

Jeroen88, 2020-04-28:
Maybe I sent the wrong log, I'll send a new one tomorrow, it is late here already...

Jeroen88, 2020-04-29:
Digging deeper into this, I found out that the (standard UUID) battery characteristic is correct and the (proprietary UUID) temperature and humity characteristic has the right value in BLEClient, but the wrong value is set in BLERemoteCharacteristic. I provide a MVCE and the right logging below.

I also found out that in the original library another function is used to get the characteristic and another callback mechanism. I also saw that in the original library the value is returned in a similar way I checked the (correct) value in BLEClient (see above comment). Maybe I draft a PR for that, if you @h2zero  could check if this is correct. Please also check if it has consequences for write(), because I did not use that yet, although I doubt it has any impact.

Jeroen88, 2020-04-29:
Updated MVCE, also reading the battery characteristic:

```
#include <NimBLEDevice.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(230400);
  Serial.println("/n/nStarted");

  BLEDevice::init("ESP32");

  readBLEClients();
}

void readBLEClients() {
// LYWSD03MMC
  static BLEClient * pClient = BLEDevice::createClient();

  Serial.println("Connecting...");
  bool success = pClient->connect(BLEAddress("a4:c1:38:5d:ef:16"));
  if(!success) {
    BLEDevice::deleteClient(pClient);

    Serial.println("Failed to connect");
    return;
  }
  Serial.println("Connected");

  BLERemoteService * pRemoteService = pClient->getService(BLEUUID("ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6"));
  if(pRemoteService == nullptr) {
    Serial.print("Failed to find our service UUID: ");
  } else {
    Serial.println(" - Found our service");

    // Temperature and humidity (fails)
    // Obtain a reference to the characteristic in the service of the remote BLE server.
    BLERemoteCharacteristic * pRemoteCharacteristic = pRemoteService->getCharacteristic(BLEUUID("ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6"));
    if(pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
    } else {
      Serial.println(" - Found our characteristic");
      if(pRemoteCharacteristic->canRead()) {
        Serial.println("Can read");
//        Serial.printf("string value is '%s'\n", pRemoteCharacteristic->readValue().c_str());
//        Serial.printf("uint8_t value is %u\n", pRemoteCharacteristic->readUInt8());
//        Serial.printf("uint16_t value is %u\n", pRemoteCharacteristic->readUInt16());
//        const uint8_t * pData = pRemoteCharacteristic->readRawData();
        const uint8_t * pData = reinterpret_cast<const uint8_t*>(pRemoteCharacteristic->readValue().c_str());

        float temperature = (pData[0] | (pData[1] << 8)) * 0.01; //little endian 
        uint8_t humidity = pData[2];
        Serial.printf("LYWSD03MMC temperature = %.2f : humidity = %u\n", temperature, humidity);    
      }
      if(pRemoteCharacteristic->canNotify()) {
        Serial.println("Can notify");
//        pRemoteCharacteristic->registerForNotify(notifyCallback);
      }
    }

    // Battery (success)
    // Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic = pRemoteService->getCharacteristic(BLEUUID("ebe0ccc4-7a0a-4b0c-8a1a-6ff2997da3a6"));
    if(pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
    } else {
      Serial.println(" - Found our characteristic");
      if(pRemoteCharacteristic->canRead()) {
        Serial.println("Can read");
//        Serial.printf("string value is '%s'\n", pRemoteCharacteristic->readValue().c_str());
//        Serial.printf("uint8_t value is %u\n", pRemoteCharacteristic->readUInt8());
//        Serial.printf("uint16_t value is %u\n", pRemoteCharacteristic->readUInt16());
//        const uint8_t * pData = pRemoteCharacteristic->readRawData();
        const uint8_t * pData = reinterpret_cast<const uint8_t*>(pRemoteCharacteristic->readValue().c_str());

        uint8_t battery = pData[0];
        Serial.printf("LYWSD03MMC battery = %u\n", battery);    
      }
      if(pRemoteCharacteristic->canNotify()) {
        Serial.println("Can notify");
//        pRemoteCharacteristic->registerForNotify(notifyCallback);
      }
    }
  }

  pClient->disconnect();
  BLEDevice::deleteClient(pClient);
}

void loop() {
}

```

Jeroen88, 2020-04-29:
Updated logging, using this library without any changes:

```
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
Connecting...
D NimBLEClient: ">> connect(a4:c1:38:5d:ef:16)"
D FreeRTOS: "Semaphore taking: name: OpenEvt (0x3ffc8b0c), owner: <N/A> for connect"
D FreeRTOS: "Semaphore taken:  name: OpenEvt (0x3ffc8b0c), owner: connect"
D FreeRTOS: ">> wait: Semaphore waiting: name: OpenEvt (0x3ffc8b0c), owner: connect for connect"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_CONNECT "
D NimBLEClient: "Connection established"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8b0c), owner: connect"
D FreeRTOS: "<< wait: Semaphore released: name: OpenEvt (0x3ffc8b0c), owner: <N/A>"
D NimBLEClient: "Refreshing Services for: (a4:c1:38:5d:ef:16)"
D NimBLEClient: ">> clearServices"
D NimBLEClient: "<< clearServices"
D NimBLEClient: ">> retrieveServices"
D FreeRTOS: "Semaphore taking: name: SearchCmplEvt (0x3ffc8b6c), owner: <N/A> for retrieveServices"
D FreeRTOS: "Semaphore taken:  name: SearchCmplEvt (0x3ffc8b6c), owner: retrieveServices"
D FreeRTOS: ">> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffc8b6c), owner: retrieveServices for retrieveServices"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_MTU"
I NimBLEClient: "mtu update event; conn_handle=0 mtu=23"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8b0c), owner: <N/A>"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 14 handle: 0"
D FreeRTOS: "Semaphore giving: name: SearchCmplEvt (0x3ffc8b6c), owner: retrieveServices"
D NimBLEClient: "<< Service Discovered. status: 0"
D FreeRTOS: "<< wait: Semaphore released: name: SearchCmplEvt (0x3ffc8b6c), owner: <N/A>"
D NimBLERemoteService: ">> retrieveCharacteristics() for service: 00000100-0065-6c62-2e74-6f696d2e696d"
D FreeRTOS: "Semaphore taking: name: GetCharEvt (0x3ffc96d4), owner: <N/A> for retrieveCharacteristics"
D FreeRTOS: "Semaphore taken:  name: GetCharEvt (0x3ffc96d4), owner: retrieveCharacteristics"
D FreeRTOS: ">> wait: Semaphore waiting: name: GetCharEvt (0x3ffc96d4), owner: retrieveCharacteristics for retrieveCharacteristics"
D NimBLERemoteService: "Characteristic Discovered >> status: 0 handle: 0"
D NimBLERemoteService: "<< Characteristic Discovered. status: 0"
D NimBLERemoteService: "Characteristic Discovered >> status: 0 handle: 0"
D NimBLERemoteService: "<< Characteristic Discovered. status: 0"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_L2CAP_UPDATE_REQ"
D NimBLEClient: "Peer requesting to update connection parameters"
[... 480 lines cut ...]
Descriptor: uuid: 0x2902, handle: 56"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D FreeRTOS: "Semaphore giving: name: ReadCharEvt (0x3ffcfaa4), owner: readValue"
D FreeRTOS: "<< wait: Semaphore released: name: ReadCharEvt (0x3ffcfaa4), owner: <N/A>"
D NimBLERemoteCharacteristic: "<< readValue(): length: 3"
LYWSD03MMC temperature = 0.00 : humidity = 0
Can notify
 - Found our characteristic
Can read
D NimBLERemoteCharacteristic: ">> readValue(): uuid: ebe0ccc4-7a0a-4b0c-8a1a-6ff2997da3a6, handle: 58 0x3a"
D FreeRTOS: "Semaphore taking: name: ReadCharEvt (0x3ffcfd8c), owner: <N/A> for readValue"
D FreeRTOS: "Semaphore taken:  name: ReadCharEvt (0x3ffcfd8c), owner: readValue"
D FreeRTOS: ">> wait: Semaphore waiting: name: ReadCharEvt (0x3ffcfd8c), owner: readValue for readValue"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D FreeRTOS: "Semaphore giving: name: ReadCharEvt (0x3ffcfd8c), owner: readValue"
D FreeRTOS: "<< wait: Semaphore released: name: ReadCharEvt (0x3ffcfd8c), owner: <N/A>"
D NimBLERemoteCharacteristic: "<< readValue(): length: 1"
LYWSD03MMC battery = 100
D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
D NimBLEClient: ">> clearServices"
D NimBLEClient: "<< clearServices"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_DISCONNECT"

```

h2zero, 2020-04-29:
This part of the log is strange:
```
D NimBLERemoteCharacteristic: ">> readValue(): uuid: ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6, handle: 54 0x36"
D FreeRTOS: "Semaphore taking: name: ReadCharEvt (0x3ffcfaa4), owner: <N/A> for readValue"
D FreeRTOS: "Semaphore taken:  name: ReadCharEvt (0x3ffcfaa4), owner: readValue"
D FreeRTOS: ">> wait: Semaphore waiting: name: ReadCharEvt (0x3ffcfaa4), owner: readValue for readValue"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_NOTIFY_RX"
D NimBLEClient: "Notify Recieved for handle: 54"
D NimBLEClient: "checking service 00000100-0065-6c62-2e74-6f696d2e696d for handle: 54"
D NimBLEClient: "checking service 0xfe95 for handle: 54"
D NimBLEClient: "checking service ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6 for handle: 54"
D NimBLEClient: "Got Notification for characteristic Characteristic: uuid: ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6, handle: 54 0x0036, props:  0x12
Descriptor: uuid: 0x2901, handle: 55
Descriptor: uuid: 0x2902, handle: 56"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D FreeRTOS: "Semaphore giving: name: ReadCharEvt (0x3ffcfaa4), owner: readValue"
D FreeRTOS: "<< wait: Semaphore released: name: ReadCharEvt (0x3ffcfaa4), owner: <N/A>"
D NimBLERemoteCharacteristic: "<< readValue(): length: 3"
LYWSD03MMC temperature = 0.00 : humidity = 0
Can notify
 - Found our characteristic
Can read
D NimBLERemoteCharacteristic: ">> readValue(): uuid: ebe0ccc4-7a0a-4b0c-8a1a-6ff2997da3a6, handle: 58 0x3a"
D FreeRTOS: "Semaphore taking: name: ReadCharEvt (0x3ffcfd8c), owner: <N/A> for readValue"
D FreeRTOS: "Semaphore taken:  name: ReadCharEvt (0x3ffcfd8c), owner: readValue"
D FreeRTOS: ">> wait: Semaphore waiting: name: ReadCharEvt (0x3ffcfd8c), owner: readValue for readValue"
I NimBLERemoteCharacteristic: "Read complete; status=0 conn_handle=0"
D FreeRTOS: "Semaphore giving: name: ReadCharEvt (0x3ffcfd8c), owner: readValue"
D FreeRTOS: "<< wait: Semaphore released: name: ReadCharEvt (0x3ffcfd8c), owner: <N/A>"
D NimBLERemoteCharacteristic: "<< readValue(): length: 1"
LYWSD03MMC battery = 100
D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
D NimBLEClient: ">> clearServices"
D NimBLEClient: "<< clearServices"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_DISCONNECT"
```

Is this your own custom peripheral device?

It looks like when you try to read the temp/humidity characteristic the peripheral sends a notification instead of returning a read response with the data. However the battery characteristic does not, so you get the correct reading. 

Could you add this:
```
for(int i=0; i<attr->om->om_len; i++) {
	NIMBLE_LOGD(LOG_TAG, "byte%d = 0x%02x", i, attr->om->om_data[i]);
}
```
[here](https://github.com/h2zero/NimBLE-Arduino/blob/846d690255859d0998ee4ac5dff007ff889ca884/src/NimBLERemoteCharacteristic.cpp#L395) to see what data NimBLE returns?

Jeroen88, 2020-04-29:
No this is not a custom device, it is a standard (square, new type) Xiaomi temp/humidity sensor. I also saw that the same call for the read of a characteristic leads to different paths through the program. For the battery this leads to the correct result, for the temp / humidity it does not.

I wil try your suggestion later, probably tomorrow. For now, from my own tests I found out that for the humidity and temperature, the right values are here (in NimBLEClient::handleGapEvent)

```
            for(auto &sPair : client->m_servicesMap){
                // Dont waste cycles searching services without this handle in their range
                if(sPair.second->getEndHandle() < event->notify_rx.attr_handle) {
                    continue;
                }
                auto cMap = sPair.second->getCharacteristicsByHandle();
                NIMBLE_LOGD(LOG_TAG, "checking service %s for handle: %d", sPair.second->getUUID().toString().c_str(),event->notify_rx.attr_handle);
                auto characteristic = cMap->find(event->notify_rx.attr_handle);
                if(characteristic != cMap->end()) {
                    NIMBLE_LOGD(LOG_TAG, "Got Notification for characteristic %s", characteristic->second->toString().c_str());
char buf[256];
memcpy(buf, event->notify_rx.om->om_data, event->notify_rx.om->om_len);
*(event->notify_rx.om->om_data + event->notify_rx.om->om_len) = '\0';
characteristic->second->m_value = std::string(buf, event->notify_rx.om->om_len);
//event->notify_rx.om->om_data -= 2;
//event->notify_rx.om->om_len += 2;

NIMBLE_LOGD(LOG_TAG, "HERE length is %u and value is'%s'\n", event->notify_rx.om->om_len, (char*) event->notify_rx.om->om_data);

```
The assignment to characteristic->second->m_value works to get the right value to BLERemoteCharacteristic::readValue(), but I did not find out yet how to differentiate between the two cases, because this messes up the battery value...

h2zero, 2020-04-29:
Well my thoughts on this are that the device you're connecting to does not support reading properly and was only intended to send notifications of those values. Which makes sense in a way but it should not have it's readable flag set. Would not be the first time an off the shelf device didn't follow the core specification 😄.

So for an explanation of what your code change has done is it has populated the remote characteristic value with the data sent in a notification, but if the battery characteristic doesn't send those you won't get the value updated there and you will have to read it. The `readValue()` method here is quite simple, makes a call to the NimBLE stack, and waits for the returned response in the `onReadCB()`. The fact you are getting the correct data when reading the battery characteristic shows that this is working. The temp/humidity characteristic however does not respond with the data when read but sends notifications instead, which also seems to happen when you have not subscribed to them yet lol, again not following the spec. 

Unless I'm missing something here, it seems to me to be an incorrectly coded peripheral. For your puropses I would suggest waiting for a notification to obtain the value, then read the battery then disconnect/delete client.

Staars, 2020-04-29:
Hi,

the LYWSD03MMC definitely provides hum/temp via notifications and it is working without any problem with your library. You can check this behavior with every BLE-APP on your phone (i.e. BLE Scanner). In contrast the battery is a readValue and BTW it always gives you 99% as it seems to be hard coded ;)

Jeroen88, 2020-04-30:
@Staars Hi Christiaan, it certainly does, but it makes my use case less elegant. I just need the values at regular intervals, and reading the values at those times is more logical than registrating a call back and waiting for the result. But maybe it is the only option...

@h2zero Thanks for your extensive analysis. Would it be an option to have the library support such a scheme, reading a characteristic and pick it up from the notification?

chegewara, 2020-04-30:
Sorry, i ddint read all posts in this topic, but after reading this one:
https://github.com/h2zero/NimBLE-Arduino/issues/19
i may know what problem you have.

You are trying to read raw data with function that is using std::string. You need to use getData() instead (its function from old library, dont know if this library got it). 
Now, when you know what data are sent you can use casting to type you need, ie:
```
uint8_t* data = pChar->getData(); // 5 bytes
int integer = *data; // int is sizeof 4 bytes
uint8_t short = data[4]; // last one remaining byte
```

Jeroen88, 2020-04-30:
@chegewara the function getData() does not exist indeed, neither on the NimBLECharacteristic nor on the BLECharacteristic. I would love to have such a function, do you have example code?

The issue is that the peripheral responds in a, according to @h2zero, off specs way. Does the getData() that you have in mind solve that problem?

chegewara, 2020-04-30:
Yes, i think it is in BLERemoteCharacteristic.

Jeroen88, 2020-04-30:
@chegewara I meant, it is not (Nim-)BLERemoteCharacteristic. I made a PR #27 for this, but if the functionality is already available, that would be better.

h2zero, 2020-04-30:
I think @chegewara was referring to the `getRawData()` method, which does exist.

I am unsure what you are trying to accomplish in your app, just trying to avoid using the callbacks? I don't think that would be realistic considering the nature of the peripheral. BLE only has 3 ways to get data from a server, 1 is a read operation, 2 is notifications, 3 is in advertisement packets.

The trouble with this peripheral is it does not respond to a read request with a read response containing the data and instead sends a notification with it. The good news is it sends the notification immediately after the read request, so it seems from your logs, at least you don't have to wait long for it.  

From what I've seen on other posts about this device is you may be able to extract the data you want from the advertisement packet, but you need to decrypt it.

Jeroen88, 2020-05-01:
getRawData() uses readValue() under the hood, so that leads to the same problem.

I try to accomplish simple and easy to maintain code. Reading a characteristic is far more simple than reading it, registering a callback, monitoring if the callback has completed, stopping the callback, using the value. I realize that reading a characteristic is also async, but this is abstracted away from the sketch by the library, which is a good thing!

Jeroen88, 2020-05-31:
Solved by PR #49

closed 2020-05-31 by Jeroen88