h2zero/NimBLE-Arduino #22: Behavior differs from default BLE library during BLE scan when passive mode is set

thorrak, 2020-04-29:
I had an issue earlier where a subset of BLE iBeacon devices that were showing up in the original ESP32 were not being picked up by my callback when using NimBLE. 

From digging through the code, it appears that behavior differs between the two libraries when `BLEScan->setActiveScan()` is set `true` thereby disabling passive mode. In the default library the callback still gets triggered regardless of the state of `passive` mode, whereas in NimBLE the callback does not get triggered when `passive` mode is set.

This seems like this behavior changed in NimBLE-Arduino after some of the initial commits - if this behavior change was intentional and expected to remain part of the library, it might be helpful to document the change somewhere.

Passive check in the NimBLE Library: https://github.com/h2zero/NimBLE-Arduino/blob/15448515d4bda4a042929f7b6546917d8df6e75a/src/NimBLEScan.cpp#L144

Line /w behavior in the Non-NimBLE Library: https://github.com/espressif/arduino-esp32/blob/1efcd21ba91c08e702a0c6e47e00a2d0a8c06b84/libraries/BLE/src/BLEScan.cpp#L127

h2zero, 2020-04-29:
Yes I was never very happy with this code, but seemed necessary at the time because NimBLE will trigger the callback for the advertisement and the scan response, so the callback would be called twice when active scanning. This was a workaround to wait until all the data was available before calling the callback.

thorrak, 2020-04-29:
Thinking about this - of the three devices I'm trying to pick up, the two that aren't being picked up with this filter are the two that have advType of `Non-Connectable - No Scan Response` which I _think_ aligns with what is happening.

That said - would it make sense to check the advType and return if it is Non-Connectable if the scan is active?

```
Advertised iBeacon Device: Name: , Address: ca:62:7a:c9:f9:da, manufacturer data: 4c000215a495bb80c5b14b44b5121370f02d74de004a045cfc, advType: Non-Connectable - No Scan Response 
Advertised iBeacon Device: Name: , Address: f8:31:ac:ae:09:ed, manufacturer data: 4c000215a495bb20c5b14b44b5121370f02d74de004b04a9c5, advType: Non-Connectable - No Scan Response 
Advertised iBeacon Device: Name: , Address: 88:c2:55:ac:26:81, manufacturer data: 4c000215a495bb40c5b14b44b5121370f02d74de004b044fc5, advType: Undirected - Connectable / Scannable 
```

The first two devices in the log above were the two that weren't being picked up when the scan was set as active, the last (the connectable one) is the one that was being picked up.

h2zero, 2020-04-29:
Yes this should be changed for sure, I had thought about this before but didn't get around to working on it.

Should work if [this line](https://github.com/h2zero/NimBLE-Arduino/blob/846d690255859d0998ee4ac5dff007ff889ca884/src/NimBLEScan.cpp#L144) was changed to 
```
if(pScan->m_scan_params.passive || event->disc.event_type == BLE_HCI_ADV_TYPE_ADV_NONCONN_IND) {
```

If you'd like to try that and let me know I'd appreciate it.

thorrak, 2020-04-29:
That worked! PR incoming. :)

thorrak, 2020-04-29:
Closed with the merge of #24

closed 2020-04-29 by thorrak