h2zero/NimBLE-Arduino #25: Upcoming major update.

h2zero, 2020-04-30:
I'd like to thank all of you who have taken an interest in this library and I hope it's working well for you, and if not, please report an issue and/or PR, they are always welcome!

I have been working on "finalizing" a v1.0 release and been involved with sorting out a few issues upstream. Some of you may have noticed I have changed my stance slightly by modifying the main NimBLE files when necessary to fix an issue. These changes have mostly been merged upstream in the mynewt/nimble repo and there is still one, possibly a second in the works in the espressif/esp-nimble repo. 

The nature of this library, essentially, is it is a CPP wrapper on a fork of a fork and originally I didn't want to change too much from the official esp-nimble fork. However the development of this library has exposed some issues in the upstream repos and your posted issues have helped make significant improvements!

That said I would like to encourage anyone who is interested to checkout the `resync-to-esp-nimble-master+patches` and or `client-long-read-write` branches and post any issues you encounter.

Minor feature requests are also encouraged as the goal is to make this library as useful and stable as possible. Major features, such as BLE mesh will be on the todo list for future updates, hopefully not too far in the future.

Again, thank you all for your contributions!

thorrak, 2020-04-30:
Thanks for your hard work on this library! 

Using this I was able to shave a massive chunk (~9%!) off my flash size, and regain over 100k of heap which means that my application suddenly had enough available memory to send data over HTTPS without having to unload the BLE scanner first. It's a massive improvement over the default BLE scanner that comes with the ESP32 Arduino framework.

Jeroen88, 2020-04-30:
Thank you from me too. I have the same case as @thorrak and this works now with the default partition. This implementation is a big memory saver!

h2zero, 2020-05-02:
I'm glad to hear those results, I didn't know what to expect when I started this project but once I had some base code started I knew it was a significant improvement. I have identified a few more pain points in the ram usage, especially on the client, I expect free heap to increase ~2k/connection depending on multiple factors.

wakwak-koba, 2020-05-05:
I think the following issues have been completely improved.
https://github.com/h2zero/NimBLE-Arduino/issues/7

Thank you for your great achievement !

dendavidseh, 2020-05-14:
Hi
just found this library! is there a place i can get some help with developing ESP/BLE software with this library? like a discord or something? dont want to pollute the issues with general questions...

h2zero, 2020-05-14:
Hi @dendavidseh, there isn’t any discord etc. setup for this as I haven’t really seen the need. 

If you are already familiar with the original library then you shouldn’t have much trouble with this one. This is very much backward compatible with a few minor differences. You can see those differences highlighted in the `refactored original examples` files as well as in `API_DIFFERENCES.md`.  If you can’t find the answer to your questions in the examples or api_differences just post an issue.

dendavidseh, 2020-05-14:
Hi thanks for the quick answer! Wasn't looking for a discord for this library specifically but maybe somewhere people have experience with esp32 ble.
I'll keep trying to get it to do what I want :)

h2zero, 2020-05-14:
@dendavidseh Careful when replying here by email. I edited your post to remove your email info.

Can you be more specific with what you're looking to do? Do you have experience with BLE on other platforms?

dendavidseh, 2020-05-15:
> @dendavidseh Careful when replying here by email. I edited your post to remove your email info.
> 
> Can you be more specific with what you're looking to do? Do you have experience with BLE on other platforms?

Hi, ok i will.
No experience with BLE, and last time i dabbled in C was 10y ago so its all quite far away.

I want to have multiple devices all with same setup
ESP is both server and client.
In a loop absolute minimum i want to do is:
scan for other ESPs, identified by my specific service
save those ESPs mac to persistent memory if its a new one
get the RSSI
do something when the average RSSI of 5 scans is x

later functionality would be to 
- add actions based on those saved macs, like white or blacklisting.
- connect to the device and get some char. 

I had the connect/disconnect working with the original BLE sample but it gave me a big memory leak allowing it to run for +- 15m and also disconnect() would crash the esp.
i'm trying to get it to work now by modifying your client example.

i've found this discord but not much activity (or knowledge?) about ble there for my queries.
https://www.reddit.com/r/hackerspaces/comments/bcc88z/digital_hackerspace_discord_server_for/

h2zero, 2020-05-15:
That should be do-able, although I have not tested, I would imagine that you would need to scan for say 2 seconds, then advertise for 2 seconds in a loop. 

With the appropriate scan result handling in place  you could store the addresses of the devices you want and average the rssi at that time. Then if you need to connect you could just create a client instance and connect to it->read->disconnect->delete client->continue loop.

dendavidseh, 2020-05-15:
> That should be do-able, although I have not tested, I would imagine that you would need to scan for say 2 seconds, then advertise for 2 seconds in a loop.
>
> With the appropriate scan result handling [...]

from your response i gather its not possible to run the server at the same time as the scan/client, ofc with the scan ignoring results of its own server(mac) ?

h2zero, 2020-05-15:
Yes you can run both (or at least it should, have not tested), that was just a simple scenario for testing. The thing you might have trouble with is scanning and advertising at the same time, you need to manage the scan interval/window and advertising interval/window well so they both have radio time often enough. Client/server simultaneous should be quite do-able as well, again if the parameters are managed. It's a juggle since there is only 1 antenna and you need to share it with 4 processes.

closed 2020-07-04 by h2zero