h2zero/NimBLE-Arduino #28: Can't connect anymore...

Jeroen88, 2020-05-01:
I try to connect to a MJ-HT-V1. I connected to this device before, but now I do don't succeed. This is the logging:

```
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
Connecting...
D NimBLEClient: ">> connect(58:2d:34:39:22:58)"
D FreeRTOS: "Semaphore taking: name: OpenEvt (0x3ffc8a58), owner: <N/A> for connect"
D FreeRTOS: "Semaphore taken:  name: OpenEvt (0x3ffc8a58), owner: connect"
D FreeRTOS: ">> wait: Semaphore waiting: name: OpenEvt (0x3ffc8a58), owner: connect for connect"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_CONNECT "
D NimBLEClient: "Connection established"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8a58), owner: connect"
D FreeRTOS: "<< wait: Semaphore released: name: OpenEvt (0x3ffc8a58), owner: <N/A>"
D NimBLEClient: "Refreshing Services for: (58:2d:34:39:22:58)"
D NimBLEClient: ">> clearServices"
D NimBLEClient: "<< clearServices"
D NimBLEClient: ">> retrieveServices"
D FreeRTOS: "Semaphore taking: name: SearchCmplEvt (0x3ffc8ab8), owner: <N/A> for retrieveServices"
D FreeRTOS: "Semaphore taken:  name: SearchCmplEvt (0x3ffc8ab8), owner: retrieveServices"
D FreeRTOS: ">> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffc8ab8), owner: retrieveServices for retrieveServices"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_MTU"
I NimBLEClient: "mtu update event; conn_handle=0 mtu=23"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8a58), owner: <N/A>"
D NimBLEClient: "Service Discovered >> status: 7 handle: 0"
D FreeRTOS: "Semaphore giving: name: SearchCmplEvt (0x3ffc8ab8), owner: retrieveServices"
D NimBLEClient: "<< Service Discovered. status: 7"
D FreeRTOS: "<< wait: Semaphore released: name: SearchCmplEvt (0x3ffc8ab8), owner: <N/A>"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_DISCONNECT"
E NimBLEClient: "Could not retrieve services"
I NimBLEClient: "disconnect; reason=531, Remote User Terminated Connection"
D NimBLEClient: ">> disconnect()"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8a58), owner: <N/A>"
D NimBLEClient: "<< disconnect()"
D FreeRTOS: "Semaphore giving: name: SearchCmplEvt (0x3ffc8ab8), owner: <N/A>"
D NimBLEClient: ">> clearServices"
D FreeRTOS: "Semaphore giving: name: Security (0x3ffc8b18), owner: <N/A>"
D NimBLEClient: "<< clearServices"
FDa NimBLEClientCallbacks: "onDisconnect: default"
```

Maybe the cause is that I did not disconnect before I ran a new sketch? I unplugged the ESP32 and took the battery out the peripheral without success...

Jeroen88, 2020-05-01:
Not sure if I could connect before. Closing for now...

Jeroen88, 2020-05-01:
I think it has something to do with the peripheral assuming that the client is already connected. If I keep trying to connect without a disconnect, sometimes after a while connecting succeeds again. If I do not disconnect and try to connect again it fails in the same way as above.

Is it possible to **force** a disconnect, eg when the ESP32 power supply is interrupted and it restarts again, there might be a dangling connection?

This is the logging just before a connection succeeds and just afterwards:

```
Connecting...
D NimBLEClient: ">> connect(58:2d:34:39:22:58)"
D FreeRTOS: "Semaphore taking: name: OpenEvt (0x3ffc8a58), owner: <N/A> for connect"
D FreeRTOS: "Semaphore taken:  name: OpenEvt (0x3ffc8a58), owner: connect"
D FreeRTOS: ">> wait: Semaphore waiting: name: OpenEvt (0x3ffc8a58), owner: connect for connect"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_CONNECT "
D NimBLEClient: "Connection established"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8a58), owner: connect"
D FreeRTOS: "<< wait: Semaphore released: name: OpenEvt (0x3ffc8a58), owner: <N/A>"
D NimBLEClient: "Refreshing Services for: (58:2d:34:39:22:58)"
D NimBLEClient: ">> clearServices"
D NimBLEClient: "<< clearServices"
D NimBLEClient: ">> retrieveServices"
D FreeRTOS: "Semaphore taking: name: SearchCmplEvt (0x3ffc8ab8), owner: <N/A> for retrieveServices"
D FreeRTOS: "Semaphore taken:  name: SearchCmplEvt (0x3ffc8ab8), owner: retrieveServices"
D FreeRTOS: ">> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffc8ab8), owner: retrieveServices for retrieveServices"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_MTU"
I NimBLEClient: "mtu update event; conn_handle=0 mtu=23"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8a58), owner: <N/A>"
D NimBLEClient: "Service Discovered >> status: 7 handle: 0"
D FreeRTOS: "Semaphore giving: name: SearchCmplEvt (0x3ffc8ab8), owner: retrieveServices"
D NimBLEClient: "<< Service Discovered. status: 7"
D FreeRTOS: "<< wait: Semaphore released: name: SearchCmplEvt (0x3ffc8ab8), owner: <N/A>"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_DISCONNECT"
E NimBLEClient: "Could not retrieve services"
I NimBLEClient: "disconnect; reason=531, Remote User Terminated Connection"
D NimBLEClient: ">> disconnect()"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8a58), owner: <N/A>"
D NimBLEClient: "<< disconnect()"
D FreeRTOS: "Semaphore giving: name: SearchCmplEvt (0x3ffc8ab8), owner: <N/A>"
D NimBLEClient: ">> clearServices"
D FreeRTOS: "Semaphore giving: name: Security (0x3ffc8b18), owner: <N/A>"
D NimBLEClient: "<< clearServices"
DF NimBLEClientCallbacks: "onDisconnect: default"
...................
Connecting...
D NimBLEClient: ">> connect(58:2d:34:39:22:58)"
D FreeRTOS: "Semaphore taking: name: OpenEvt (0x3ffc8a58), owner: <N/A> for connect"
D FreeRTOS: "Semaphore taken:  name: OpenEvt (0x3ffc8a58), owner: connect"
D FreeRTOS: ">> wait: Semaphore waiting: name: OpenEvt (0x3ffc8a58), owner: connect for connect"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_CONNECT "
D NimBLEClient: "Connection established"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8a58), owner: connect"
D FreeRTOS: "<< wait: Semaphore released: name: OpenEvt (0x3ffc8a58), owner: <N/A>"
D NimBLEClient: "Refreshing Services for: (58:2d:34:39:22:58)"
D NimBLEClient: ">> clearServices"
D NimBLEClient: "<< clearServices"
D NimBLEClient: ">> retrieveServices"
D FreeRTOS: "Semaphore taking: name: SearchCmplEvt (0x3ffc8ab8), owner: <N/A> for retrieveServices"
D FreeRTOS: "Semaphore taken:  name: SearchCmplEvt (0x3ffc8ab8), owner: retrieveServices"
D FreeRTOS: ">> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffc8ab8), owner: retrieveServices for retrieveServices"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_MTU"
I NimBLEClient: "mtu update event; conn_handle=0 mtu=23"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8a58), owner: <N/A>"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
D NimBLERemoteService: ">> BLERemoteService()"
D NimBLERemoteService: "<< BLERemoteService()"
D NimBLEClient: "<< Service Discovered. status: 0"
D NimBLEClient: "Service Discovered >> status: 0 handle: 0"
[... 306 lines cut ...]
D NimBLEClient: ">> clearServices"
D NimBLEClient: "<< clearServices"
D NimBLEClient: ">> retrieveServices"
D FreeRTOS: "Semaphore taking: name: SearchCmplEvt (0x3ffc8ab8), owner: <N/A> for retrieveServices"
D FreeRTOS: "Semaphore taken:  name: SearchCmplEvt (0x3ffc8ab8), owner: retrieveServices"
D FreeRTOS: ">> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffc8ab8), owner: retrieveServices for retrieveServices"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_MTU"
I NimBLEClient: "mtu update event; conn_handle=0 mtu=23"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8a58), owner: <N/A>"
D NimBLEClient: "Service Discovered >> status: 7 handle: 0"
D FreeRTOS: "Semaphore giving: name: SearchCmplEvt (0x3ffc8ab8), owner: retrieveServices"
D NimBLEClient: "<< Service Discovered. status: 7"
D FreeRTOS: "<< wait: Semaphore released: name: SearchCmplEvt (0x3ffc8ab8), owner: <N/A>"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_DISCONNECT"
E NimBLEClient: "Could not retrieve services"
I NimBLEClient: "disconnect; reason=531, Remote User Terminated Connection"
D NimBLEClient: ">> disconnect()"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8a58), owner: <N/A>"
D NimBLEClient: "<< disconnect()"
D FreeRTOS: "Semaphore giving: name: SearchCmplEvt (0x3ffc8ab8), owner: <N/A>"
D NimBLEClient: ">> clearServices"
D FreeRTOS: "Semaphore giving: name: Security (0x3ffc8b18), owner: <N/A>"
D NimBLEClient: "<< clearServices"
FDa NimBLEClientCallbacks: "onDisconnect: default" 
```

h2zero, 2020-05-01:
This comes down to the connection parameters. The default parameters set the connection timeout to 2560ms, this is how long the peripheral will wait before it recognizes you have disconnected, assuming it follows the spec. If you want a shorter time you would need to specify it either before connecting with `setConnectionParams()` or change the parameters after connection with `updateConnParams()`.

Jeroen88, 2020-05-01:
@h2zero Thnx again for your answer! Then I think it must be something else. I get a `BLE_GAP_EVENT_DISCONNECT` soon after the connection. Do you know why?

chegewara, 2020-05-01:
If i would have guessing i would say this is the reason:
`Semaphore giving: name: Security`

One of 2 devices is requiring secured connection, probably with pass code or something.

chegewara, 2020-05-01:
Im not sure it is about your device, spent only 1-2 minutes to read few posts, but maybe it can help you:
https://www.domoticz.com/forum/viewtopic.php?t=20925&start=20

h2zero, 2020-05-01:
I don't think there is any security going on here, that semaphore is released in the disconnect event whether it was taken or not as a precaution if it were to get stuck. In fact the NimBLE stack is pretty smart about informing the app via callbacks and I may remove that bit of code.

Jeroen88, 2020-05-02:
But have you got any idea _why_ I get the disconnect?

Jeroen88, 2020-05-02:
If I try to connect using a phone app it gets disconnected right away too. I think it must be something in the peripheral... I bought it on Ali, so maybe fake :/

Staars, 2020-05-02:
No, no fake. This is the typical behavior of the MJ_HT_V1. A short google search will find you multiple reports about it.
This sensor is not designed for connections but shall be read read out via BLE advertisements which already works totally flawless with this library. If you want to read out via connection -> be quick. Even this works (with a callback).
I only connect to the LYWSD03 (because of the encrypted advertisements). All the rest of my Mijia sensors do work fine with this library through the use of advertised data (MJ_HT_V1, Flower Care, LYWSD02, CGD01).

Jeroen88, 2020-05-02:
@Staars yes I already got it working on advertisements, but I wanted it to be read out with a connection. Giving that up, if the MJ_HT_V1 is not designed for that. Thanks for the advise!

closed 2020-05-02 by Jeroen88

chegewara, 2020-05-03:
Like i said, it may be security problem, because i saw few articles about retrieving token from that device. If you have token, you have security. Or maybe token is only used to register it in third party services like online dashboard.

h2zero, 2020-05-03:
Interesting, good thoughts @chegewara. I wonder what would happen if they tried calling `NimBLEDevice::startSecurity()` in the `onConnect()` handler. I should get some of these devices to play with.

Jeroen88, 2020-05-03:
@h2zero They are very cheap, I bought 2 round and 2 square pieces [here](https://www.aliexpress.com/item/4000783319372.html)

h2zero, 2020-05-03:
@Jeroen88 Thanks, I might grab a couple to mess with.
I'm interested to see if you could sustain a connection by calling
```
NimBLEDevice::startSecurity(pClient->getConnId());
```
after you've connected. If it works it might help others too.

Jeroen88, 2020-05-04:
Hi @h2zero, that won't work,the connection is closed before the client has finished discovering the services, see above log: `D NimBLEClient: ">> retrieveServices"` for retrieving the services is reported, but the finishing is never reported, because the peripheral closes the connection before that event takes place.

chegewara, 2020-05-05:
Just for info. Before you start discovering services there is a lot going on.

Jeroen88, 2020-05-05:
I saw in the logging that discovering services is a "lot going on", but before that there is nothing in the logging. A lot going on on the lower layers?

h2zero, 2020-05-06:
@Jeroen88 Yes there is quite a bit of back and forth on connection, I guess the only place this test would work for you is in the `handleGapEvent()` connect event section, just before the MTU exchange call. If it did work I would need to consider making modifications here.

Jeroen88, 2020-05-06:
@h2zero Thnx, maybe I give that a try later!