h2zero/NimBLE-Arduino #32: Undefined reference to 'ble_uuid_cmp'

elliottkember, 2020-05-03:
Very excited about this port!

I've hit a weird snag trying to compile this. I'm running ESP-IDF 3.2 and trying to replace an existing BLE implementation. Compiling works, then I hit this during the GNU linker:

```
LD build/soulmate-example.elf
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLECharacteristic.o):(.literal._ZN20NimBLECharacteristic14handleGapEventEttP20ble_gatt_access_ctxtPv+0x0): undefined reference to `ble_uuid_cmp'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLECharacteristic.o):(.literal._ZN20NimBLECharacteristic14handleGapEventEttP20ble_gatt_access_ctxtPv+0x4): undefined reference to `os_mbuf_append'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLECharacteristic.o):(.literal._ZN20NimBLECharacteristic6notifyEb+0x14): undefined reference to `ble_hs_mbuf_from_flat'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLECharacteristic.o):(.literal._ZN20NimBLECharacteristic6notifyEb+0x18): undefined reference to `ble_gattc_indicate_custom'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLECharacteristic.o):(.literal._ZN20NimBLECharacteristic6notifyEb+0x1c): undefined reference to `ble_gattc_notify_custom'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLECharacteristic.o): In function `NimBLECharacteristic::handleGapEvent(unsigned short, unsigned short, ble_gatt_access_ctxt*, void*)':
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLECharacteristic.cpp:144: undefined reference to `ble_uuid_cmp'
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLECharacteristic.cpp:144: undefined reference to `os_mbuf_append'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLECharacteristic.o): In function `std::map<unsigned short, unsigned short, std::less<unsigned short>, std::allocator<std::pair<unsigned short const, unsigned short> > >::erase(unsigned short const&)':
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLECharacteristic.cpp:144: undefined reference to `ble_hs_mbuf_from_flat'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLECharacteristic.o): In function `NimBLECharacteristic::notify(bool)':
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLECharacteristic.cpp:144: undefined reference to `ble_gattc_indicate_custom'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLECharacteristic.o): In function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()':
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLECharacteristic.cpp:144: undefined reference to `ble_gattc_notify_custom'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDescriptor.o): In function `NimBLEDescriptor::handleGapEvent(unsigned short, unsigned short, ble_gatt_access_ctxt*, void*)':
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLEDescriptor.cpp:134: undefined reference to `ble_uuid_cmp'
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLEDescriptor.cpp:138: undefined reference to `os_mbuf_append'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o):(.literal._ZN12NimBLEDevice9host_taskEPv+0x0): undefined reference to `nimble_port_run'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o):(.literal._ZN12NimBLEDevice9host_taskEPv+0x4): undefined reference to `nimble_port_freertos_deinit'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o):(.literal._ZN12NimBLEDevice6onSyncEv+0x10): undefined reference to `ble_hs_util_ensure_addr'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o):(.literal._ZN12NimBLEDevice12createServerEv+0x4): undefined reference to `ble_gatts_reset'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o):(.literal._ZN12NimBLEDevice12createServerEv+0x8): undefined reference to `ble_svc_gap_init'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o):(.literal._ZN12NimBLEDevice12createServerEv+0xc): undefined reference to `ble_svc_gatt_init'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o):(.literal._ZN12NimBLEDevice4initENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x18): undefined reference to `ble_hs_cfg'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o):(.literal._ZN12NimBLEDevice4initENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x24): undefined reference to `ble_store_util_status_rr'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o):(.literal._ZN12NimBLEDevice4initENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x30): undefined reference to `esp_nimble_hci_and_controller_init'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o):(.literal._ZN12NimBLEDevice4initENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x34): undefined reference to `nimble_port_init'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o):(.literal._ZN12NimBLEDevice4initENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x38): undefined reference to `ble_svc_gap_device_name_set'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o):(.literal._ZN12NimBLEDevice4initENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x3c): undefined reference to `ble_store_config_init'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o):(.literal._ZN12NimBLEDevice4initENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x40): undefined reference to `nimble_port_freertos_init'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o): In function `NimBLEDevice::host_task(void*)':
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLEDevice.cpp:659: undefined reference to `nimble_port_run'
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLEDevice.cpp:659: undefined reference to `nimble_port_freertos_deinit'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o): In function `NimBLEDevice::onSync()':
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLEDevice.cpp:659: undefined reference to `ble_hs_util_ensure_addr'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o): In function `NimBLEDevice::createServer()':
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLEDevice.cpp:659: undefined reference to `ble_gatts_reset'
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLEDevice.cpp:659: undefined reference to `ble_svc_gap_init'
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLEDevice.cpp:659: undefined reference to `ble_svc_gatt_init'
/Users/elliott/Sites/soulmate/soulmate-example/build/nimble/libnimble.a(NimBLEDevice.o): In function `NimBLEDevice::init(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLEDevice.cpp:659: undefined reference to `esp_nimble_hci_and_controller_init'
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLEDevice.cpp:659: undefined reference to `nimble_port_init'
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLEDevice.cpp:659: undefined reference to `ble_svc_gap_device_name_set'
/Users/elliott/soulmate/soulmate/components/nimble/src/NimBLEDevice.cpp:659: undefined reference to `ble_store_config_init'
```

h2zero, 2020-05-03:
Yep, using this directly in esp-idf could be troublesome. I would need to create a makefile for the port and since nimble is available in idf I haven't done that. 

You could use my other repo [here](https://github.com/h2zero/esp-nimble-cpp). Just need to put that in your project `components` folder.

But then you have some work to do as I haven't gotten around to making it easy yet. 
 
You'll need to get [nimble folder from idf v3.3](https://github.com/espressif/esp-idf/tree/release/v3.3/components/nimble) in the components folder and copy it over to your 3.2 `esp-idf/components`. At that point you might encounter a compile issue, or maybe not, it's been a while since I did that operation. If you have an issue compiling this way let me know, I've seen them all  😄.

elliottkember, 2020-05-03:
Thank you! I made a `component.mk` and it was encouraging that it compiled, but I was confused by the linking part.

What I should probably do is upgrade to ESP v3.3, I believe esp32-arduino works with that version now in a branch. I was hitting the issue described in https://github.com/espressif/esp32-arduino-lib-builder/issues/10

h2zero, 2020-05-03:
Yeah I had that issue too with 3.3, had to use the [esp32-arduino v3.3 latest branch](https://github.com/espressif/arduino-esp32/tree/latest-3.3)

elliottkember, 2020-05-04:
Ah, now I get this: 

`components/nimble-esp32/src/FreeRTOS.h:65:31: error: 'RingbufferType_t' has not been declared`

Looks like that got renamed from ringbuf_type_t between v3.3 and v4.0

h2zero, 2020-05-04:
Are you using an older commit of this library? That issue was fixed (at least it was) a while ago.

h2zero, 2020-05-04:
I had a second to try this again in 3.2 I forgot there's a few things that need to be done to make it work. I think I will make a branch on the esp-nimble-cpp repo for that backport so it can just be added to the project/components folder. Easy enough to do.

elliottkember, 2020-05-04:
That'd be great! I'd love to get my project upgraded to v3.3, but so far it's been tricky - it also seems like v4.0 support could drop any day now so I'm looking forward to that. v3.2 is working well, just trying to free up a bit more memory for HTTPS and OTA things. Bluedroid uses so much.

h2zero, 2020-05-04:
Sorry you're having these issues, I just realized I had updated idf to v3.3.2, that is where the `ESP_IDF_VERSION` was first defined but that came after the change to `RingbufferType_t` I believe. So there was a gap between those in 3.3, not 100% sure of that but I'll check.

The undefined references are odd, not sure about those. Are you using the `NimBLE-Arduino`(not working in IDF yet) library or the `esp-nimble-cpp` library?

I did make the proper IDF 3.2 backport last night, compiles and runs but was getting strange errors on test, I think it was something in my sdkconfig, but it was late and didn't have the energy to fix it. I'll try to put it up tonight.

h2zero, 2020-05-05:
As promised [here](https://github.com/h2zero/esp-nimble-component/tree/idf-v3.2) is a quick idf 3.2 NimBLE backport. You just need to add it to your `project/components` folder. Then add [esp-nimble-cpp](https://github.com/h2zero/esp-nimble-cpp) as another component in `project/components` along side the arduino component, if you wish, and include `NimBLEDevice.h` in `main.cpp`. 

One thing I should mention is while testing I was getting the occasional assert from rwble.c which is part of the BLE controller binary, nothing I can do about it here sadly, hopefully for your use it doesn't happen. If you do have this issue, the only solution is to update IDF, I'll keep working on the 3.3 and 4.0 issues in the meantime.

elliottkember, 2020-05-05:
Thank you! I’ll try this out this evening when I get some time. I was using the Arduino version which would explain it.

h2zero, 2020-05-06:
@elliottkember No problem, you may have to mess with menuconfig a little (probably not, but that depends). Worked for me with defaults mostly. Let me know how it works out for you.

elliottkember, 2020-05-17:
@h2zero I finally got around to setting this up - and it works great! Thank you so much!

elliottkember, 2020-05-17:
Looks like I saved 32kb of heap, and heap fragmentation went down a ton after switching from Bluedroid. For reference, I'm using Bluetooth LE and WiFi simultaneously for an LED project, plus a homekit stack that needs a lot of allocation heap space to pair. Now an ESP32-Pico with no external SPI ram can run all these things perfectly!

h2zero, 2020-05-17:
Glad to hear it 😄

h2zero, 2020-05-18:
@elliottkember What BLE roles are you using in your project (client, server, scan, advertise)?

elliottkember, 2020-06-11:
> @elliottkember What BLE roles are you using in your project (client, server, scan, advertise)?

Server / advertise. Controlling an ESP32 from a mobile app

closed 2020-06-11 by elliottkember

ntwallace, 2020-09-28:
@h2zero I may be doing something wrong here, but after migrating from arduino-esp32 to IDF v3.3 with Arduino as a component I'm getting the same linking error after compiling. I added [this](https://github.com/h2zero/esp-nimble-component) as a component and configured in menuconfig. Is there another step I'm missing?

I am able to get it to link if I enable the default nimBLE in IDF 3.3 and gut the Arduino Bluedroid libs out. However, calling `NimBLEDevice::init(DEVICE_NAME);` results in a panic boot loop due to an assertion failing in timers.c internally: 

```
/Users/nick/esp-idf/components/freertos/timers.c:922 (prvCheckForValidListAndQueue)- assert failed!
abort() was called at PC 0x4009337b on core 0
```

```0x400900db: invoke_abort at /Users/nick/esp-idf/components/esp32/panic.c line 156
0x400903b1: abort at /Users/nick/esp-idf/components/esp32/panic.c line 171
0x4009337b: prvCheckForValidListAndQueue at /Users/nick/esp-idf/components/freertos/timers.c line 922
0x4009338b: xTimerCreateTimerTask at /Users/nick/esp-idf/components/freertos/timers.c line 270
0x400921ac: vTaskStartScheduler at /Users/nick/esp-idf/components/freertos/tasks.c line 2057
0x40081227: start_cpu0_default at /Users/nick/esp-idf/components/esp32/cpu_start.c line 445
0x400813fd: call_start_cpu0 at /Users/nick/esp-idf/components/esp32/cpu_start.c line 256```

h2zero, 2020-09-29:
Hi @ntwallace are you trying to use the NimBLE-Arduino library? When using IDF you will need to use the component library you linked as well as esp-nimble-cpp instead of this one. After that you shouldn't need to disable any libs etc in menuconfig.

ntwallace, 2020-09-29:
> Hi @ntwallace are you trying to use the NimBLE-Arduino library? When using IDF you will need to use the component library you linked as well as esp-nimble-cpp instead of this one. After that you shouldn't need to disable any libs etc in menuconfig.

🤦‍♂️ thanks. 

It looks like esp-nimble-cpp will only compile for v4.0+ because of the changes to CMAKE for `idf_component_register()` vs the older `register_component()`. I attempted to modify the cmakelists file to use the older format, but am getting build errors telling me it can't find header files, e.g.:

```
../components/esp-nimble-cpp/src/NimBLEUUID.h:20:27: fatal error: host/ble_uuid.h: No such file or directory
../components/esp-nimble-cpp/src/NimBLELog.h:13:27: fatal error: syscfg/syscfg.h: No such file or directory
```

My modified CMakeLists.txt file looks like this, do you see any obvious errors?
```
set(SUPPORTED_TARGETS esp32)

#idf_component_register(SRCS "src/FreeRTOS.cpp"
#                            "src/NimBLE2904.cpp"
#                            "src/NimBLEAddress.cpp"
#                            "src/NimBLEAdvertisedDevice.cpp"
#                            "src/NimBLEAdvertising.cpp"
#                            "src/NimBLEBeacon.cpp"
#                             "src/NimBLECharacteristic.cpp"
#                             "src/NimBLEClient.cpp"
#                             "src/NimBLEDescriptor.cpp"
#                             "src/NimBLEDevice.cpp"
#                             "src/NimBLEEddystoneTLM.cpp"
#                             "src/NimBLEEddystoneURL.cpp"
#                            "src/NimBLERemoteCharacteristic.cpp"
#                             "src/NimBLERemoteDescriptor.cpp"
#                             "src/NimBLERemoteService.cpp"
#                             "src/NimBLEScan.cpp"
#                             "src/NimBLESecurity.cpp"
#                             "src/NimBLEServer.cpp"
#                             "src/NimBLEService.cpp"
#                             "src/NimBLEUtils.cpp"
#                             "src/NimBLEUUID.cpp"
#                   INCLUDE_DIRS "src"
#                   REQUIRES bt)

# idf_component_register(SRCS "${MAIN_SRCS}" INCLUDE_DIRS ".")

set(COMPONENT_ADD_INCLUDEDIRS
            src
            .
)

set(COMPONENT_SRCS
    "src/FreeRTOS.cpp"
    "src/NimBLE2904.cpp"
    "src/NimBLEAddress.cpp"
    "src/NimBLEAdvertisedDevice.cpp"
    "src/NimBLEAdvertising.cpp"
    "src/NimBLEBeacon.cpp"
    "src/NimBLECharacteristic.cpp"
    "src/NimBLEClient.cpp"
    "src/NimBLEDescriptor.cpp"
    "src/NimBLEDevice.cpp"
    "src/NimBLEEddystoneTLM.cpp"
    "src/NimBLEEddystoneURL.cpp"
    "src/NimBLERemoteCharacteristic.cpp"
    "src/NimBLERemoteDescriptor.cpp"
    "src/NimBLERemoteService.cpp"
    "src/NimBLEScan.cpp"
    "src/NimBLESecurity.cpp"
    "src/NimBLEServer.cpp"
    "src/NimBLEService.cpp"
    "src/NimBLEUtils.cpp"
)

set(COMPONENT_PRIV_REQUIRES bt)

register_component()
```

h2zero, 2020-09-29:
I'm not sure what errors you're seeing when compiling with IDF 3.2/3.3, I have no issues with those versions with esp-nimble-cpp as is. Should be no need to alter make files. I'll have a closer look when I get a chance though.

ntwallace, 2020-09-29:
Thanks. Idf_component_register() is only supported on IDF 4.0 and up, so compiling under 3.3 as is fails with 

```
[0/1] Re-running CMake...
-- Building for target esp32
CMake Error at /Users/nick/Documents/ESPLED/firmware/components/esp-nimble-cpp/CMakeLists.txt:7 (idf_component_register):
  Unknown CMake command "idf_component_register".
Call Stack (most recent call first):
  /Users/nick/esp-idf/tools/cmake/scripts/expand_requirements.cmake:107 (include)
```

FWIW it seems that under 3.3 it's unable to find the underlying header files included in the esp-nimble-cpp files that are located in the esp-nimble-component source. For example, NimBLEAddress.h has an include on line 20 for `"nimble/ble.h"`, but the compiler isn't correctly getting to esp-nimble-component/nimble/nimble/include/ble.h

h2zero, 2020-09-29:
Sorry I didn't notice earlier (hard to read on phone sometimes), you are using cmake. I didn't test against cmake for IDF 3.2/3.3 as that was a v4.0 change. 

My apologies, it should work fine with `make`. I'm not sure what would need to be done to get cmake to compile it in 3.3, I'll have to play with this.

ntwallace, 2020-09-29:
No worries at all, appreciate your responsiveness. I'll keep working on it tomorrow as well.

ntwallace, 2020-09-29:
@h2zero fixed the compile issue by explicitly calling esp-nimble-component in the requires list for esp-nimble-cpp in cmakelists. For future reference, this is the working file for compiling with cmake on 3.3:

```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.15.4)

set(SUPPORTED_TARGETS esp32)

set(COMPONENT_SRCS
    "src/FreeRTOS.cpp"
    "src/NimBLE2904.cpp"
    "src/NimBLEAddress.cpp"
    "src/NimBLEAdvertisedDevice.cpp"
    "src/NimBLEAdvertising.cpp"
    "src/NimBLEBeacon.cpp"
    "src/NimBLECharacteristic.cpp"
    "src/NimBLEClient.cpp"
    "src/NimBLEDescriptor.cpp"
    "src/NimBLEDevice.cpp"
    "src/NimBLEEddystoneTLM.cpp"
    "src/NimBLEEddystoneURL.cpp"
    "src/NimBLERemoteCharacteristic.cpp"
    "src/NimBLERemoteDescriptor.cpp"
    "src/NimBLERemoteService.cpp"
    "src/NimBLEScan.cpp"
    "src/NimBLESecurity.cpp"
    "src/NimBLEServer.cpp"
    "src/NimBLEService.cpp"
    "src/NimBLEUtils.cpp"
)

set(COMPONENT_ADD_INCLUDEDIRS
    src
)

set(COMPONENT_PRIV_REQUIRES 
    esp-nimble-component
    nvs_flash
    bt
)

register_component()
```

Unfortunately, I'm still getting some linking errors at the end. Any thoughts on where to track this, seems to be specifically with NimBLEUUID:
```
esp-idf/esp-nimble-cpp/libesp-nimble-cpp.a(NimBLEAdvertising.cpp.obj):(.literal._ZN17NimBLEAdvertising5startEjPFvPS_E+0x28): undefined reference to `NimBLEUUID::getNative() const'
esp-idf/esp-nimble-cpp/libesp-nimble-cpp.a(NimBLEAdvertising.cpp.obj): In function `NimBLEAdvertising::start(unsigned int, void (*)(NimBLEAdvertising*))':
/Users/nick/Documents/ESPLED/firmware/build/../components/esp-nimble-cpp/src/NimBLEAdvertising.cpp:456: undefined reference to `NimBLEUUID::getNative() const'
/Users/nick/Documents/ESPLED/firmware/build/../components/esp-nimble-cpp/src/NimBLEAdvertising.cpp:456: undefined reference to `NimBLEUUID::getNative() const'
/Users/nick/Documents/ESPLED/firmware/build/../components/esp-nimble-cpp/src/NimBLEAdvertising.cpp:456: undefined reference to `NimBLEUUID::getNative() const'
/Users/nick/Documents/ESPLED/firmware/build/../components/esp-nimble-cpp/src/NimBLEAdvertising.cpp:456: undefined reference to `NimBLEUUID::getNative() const'
esp-idf/esp-nimble-cpp/libesp-nimble-cpp.a(NimBLEAdvertising.cpp.obj):/Users/nick/Documents/ESPLED/firmware/build/../components/esp-nimble-cpp/src/NimBLEAdvertising.cpp:456: more undefined references to `NimBLEUUID::getNative() const' follow
esp-idf/esp-nimble-cpp/libesp-nimble-cpp.a(NimBLEAdvertisedDevice.cpp.obj):(.literal._ZN22NimBLEAdvertisedDevice14setServiceUUIDE10NimBLEUUID+0x0): undefined reference to `NimBLEUUID::operator==(NimBLEUUID const&) const'
esp-idf/esp-nimble-cpp/libesp-nimble-cpp.a(NimBLEAdvertisedDevice.cpp.obj):(.literal._ZN22NimBLEAdvertisedDevice18parseAdvertisementEPhh+0xc): undefined reference to `NimBLEUUID::NimBLEUUID(unsigned short)'
esp-idf/esp-nimble-cpp/libesp-nimble-cpp.a(NimBLEAdvertisedDevice.cpp.obj):(.literal._ZN22NimBLEAdvertisedDevice18parseAdvertisementEPhh+0x10): undefined reference to `NimBLEUUID::NimBLEUUID(unsigned int)'
esp-idf/esp-nimble-cpp/libesp-nimble-cpp.a(NimBLEAdvertisedDevice.cpp.obj):(.literal._ZN22NimBLEAdvertisedDevice18parseAdvertisementEPhh+0x14): undefined reference to `NimBLEUUID::NimBLEUUID(ble_uuid128_t const*)'
esp-idf/esp-nimble-cpp/libesp-nimble-cpp.a(NimBLEAdvertisedDevice.cpp.obj):(.literal._ZN22NimBLEAdvertisedDevice18parseAdvertisementEPhh+0x18): undefined reference to `NimBLEUUID::NimBLEUUID(unsigned char const*, unsigned int, bool)'
esp-idf/esp-nimble-cpp/libesp-nimble-cpp.a(NimBLEAdvertisedDevice.cpp.obj): In function `NimBLEAdvertisedDevice::setServiceUUID(NimBLEUUID)':
/Users/nick/.espressif/tools/xtensa-esp32-elf/1.22.0-96-g2852398-5.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:544: undefined reference to `NimBLEUUID::operator==(NimBLEUUID const&) const'
esp-idf/esp-nimble-cpp/libesp-nimble-cpp.a(NimBLEAdvertisedDevice.cpp.obj): In function `NimBLEAdvertisedDevice::setServiceData(NimBLEUUID, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/Users/nick/Documents/ESPLED/firmware/build/../components/esp-nimble-cpp/src/NimBLEAdvertisedDevice.cpp:512: undefined reference to `NimBLEUUID::operator==(NimBLEUUID const&) const'
esp-idf/esp-nimble-cpp/libesp-nimble-cpp.a(NimBLEAdvertisedDevice.cpp.obj): In function `NimBLEAdvertisedDevice::parseAdvertisement(unsigned char*, unsigned char)':
/Users/nick/Documents/ESPLED/firmware/build/../components/esp-nimble-cpp/src/NimBLEAdvertisedDevice.cpp:316: undefined reference to `NimBLEUUID::NimBLEUUID(unsigned short)'
/Users/nick/Documents/ESPLED/firmware/build/../components/esp-nimble-cpp/src/NimBLEAdvertisedDevice.cpp:322: undefined reference to `NimBLEUUID::NimBLEUUID(unsigned int)'
/Users/nick/Documents/ESPLED/firmware/build/../components/esp-nimble-cpp/src/NimBLEAdvertisedDevice.cpp:328: undefined reference to `NimBLEUUID::NimBLEUUID(ble_uuid128_t const*)'
/Users/nick/Documents/ESPLED/firmware/build/../components/esp-nimble-cpp/src/NimBLEAdvertisedDevice.cpp:356: undefined reference to `NimBLEUUID::NimBLEUUID(unsigned short)'
/Users/nick/Documents/ESPLED/firmware/build/../components/esp-nimble-cpp/src/NimBLEAdvertisedDevice.cpp:364: undefined reference to `NimBLEUUID::NimBLEUUID(unsigned int)'
/Users/nick/Documents/ESPLED/firmware/build/../components/esp-nimble-cpp/src/NimBLEAdvertisedDevice.cpp:365: undefined reference to `NimBLEUUID::NimBLEUUID(unsigned char const*, unsigned int, bool)'
```

h2zero, 2020-09-29:
Glad you got it compiled. How did you setup sdkconfig? Normally when using Arduino as a component I copy the sdkconfig from the arduino core repo to main folder then use menuconfig to adjust settings. Give that a try if you haven't already.

ntwallace, 2020-09-29:
Ok thanks. My sdkconfig was originally based on the default Arduino sdkconfig, but I have made some changes. I'll go back and check and try to match.

One other note, when I enable NimBLE debugging (CONFIG_BT_NIMBLE_DEBUG=y), I can no longer compile. Does that work for you? I get issues where it's trying to use sizeof() on a bit in C.
```
In file included from /Users/nick/esp-idf/components/newlib/platform_include/assert.h:23:0,
                 from ../components/esp-nimble-component/nimble/porting/nimble/include/os/os.h:23,
                 from ../components/esp-nimble-component/nimble/nimble/include/nimble/ble.h:26,
                 from ../components/esp-nimble-component/nimble/nimble/host/src/ble_gatts.c:23:
../components/esp-nimble-component/nimble/nimble/host/src/ble_gatts.c: In function 'ble_gatts_bonding_established':
/Users/nick/esp-idf/components/newlib/include/assert.h:14:36: error: 'sizeof' applied to a bit-field
 # define assert(__e) ((void) sizeof(__e))
                                    ^
../components/esp-nimble-component/nimble/nimble/host/src/ble_hs_priv.h:171:34: note: in expansion of macro 'assert'
     #define BLE_HS_DBG_ASSERT(x) assert(x)
                                  ^
../components/esp-nimble-component/nimble/nimble/host/src/ble_gatts.c:1692:5: note: in expansion of macro 'BLE_HS_DBG_ASSERT'
     BLE_HS_DBG_ASSERT(conn->bhc_sec_state.bonded);
     ^
../components/esp-nimble-component/nimble/nimble/host/src/ble_gatts.c: In function 'ble_gatts_bonding_restored':
/Users/nick/esp-idf/components/newlib/include/assert.h:14:36: error: 'sizeof' applied to a bit-field
 # define assert(__e) ((void) sizeof(__e))
                                    ^
../components/esp-nimble-component/nimble/nimble/host/src/ble_hs_priv.h:171:34: note: in expansion of macro 'assert'
     #define BLE_HS_DBG_ASSERT(x) assert(x)
                                  ^
../components/esp-nimble-component/nimble/nimble/host/src/ble_gatts.c:1743:5: note: in expansion of macro 'BLE_HS_DBG_ASSERT'
     BLE_HS_DBG_ASSERT(conn->bhc_sec_state.bonded);
```

h2zero, 2020-09-29:
That's an interesting one, I have never seen this error before, must be something to do with using cmake. 

Do you get any errors when using make?

ntwallace, 2020-09-29:
Same exact error using make

```
In file included from /Users/nick/esp-idf/components/newlib/platform_include/assert.h:23:0,
                 from /Users/nick/Documents/ESPLED/firmware/components/esp-nimble-component/nimble/porting/nimble/include/os/os.h:23,
                 from /Users/nick/Documents/ESPLED/firmware/components/esp-nimble-component/nimble/nimble/include/nimble/ble.h:26,
                 from /Users/nick/Documents/ESPLED/firmware/components/esp-nimble-component/nimble/nimble/host/src/ble_gatts.c:23:
/Users/nick/Documents/ESPLED/firmware/components/esp-nimble-component/nimble/nimble/host/src/ble_gatts.c: In function 'ble_gatts_bonding_established':
/Users/nick/esp-idf/components/newlib/include/assert.h:14:36: error: 'sizeof' applied to a bit-field
 # define assert(__e) ((void) sizeof(__e))
                                    ^
/Users/nick/Documents/ESPLED/firmware/components/esp-nimble-component/nimble/nimble/host/src/ble_hs_priv.h:171:34: note: in expansion of macro 'assert'
     #define BLE_HS_DBG_ASSERT(x) assert(x)
                                  ^
/Users/nick/Documents/ESPLED/firmware/components/esp-nimble-component/nimble/nimble/host/src/ble_gatts.c:1692:5: note: in expansion of macro 'BLE_HS_DBG_ASSERT'
     BLE_HS_DBG_ASSERT(conn->bhc_sec_state.bonded);
     ^
/Users/nick/Documents/ESPLED/firmware/components/esp-nimble-component/nimble/nimble/host/src/ble_gatts.c: In function 'ble_gatts_bonding_restored':
/Users/nick/esp-idf/components/newlib/include/assert.h:14:36: error: 'sizeof' applied to a bit-field
 # define assert(__e) ((void) sizeof(__e))
                                    ^
/Users/nick/Documents/ESPLED/firmware/components/esp-nimble-component/nimble/nimble/host/src/ble_hs_priv.h:171:34: note: in expansion of macro 'assert'
     #define BLE_HS_DBG_ASSERT(x) assert(x)
                                  ^
/Users/nick/Documents/ESPLED/firmware/components/esp-nimble-component/nimble/nimble/host/src/ble_gatts.c:1743:5: note: in expansion of macro 'BLE_HS_DBG_ASSERT'
     BLE_HS_DBG_ASSERT(conn->bhc_sec_state.bonded);
```

h2zero, 2020-09-29:
I just had a chance to test this issue using cmake. After correcting the cmake files it builds and runs as expected (without arduino component). I will try with Arduino next, Can you post your sdkconfig?

ntwallace, 2020-09-29:
Interesting ok. Sure, here it is. FYI this is setup for a WROVER board with 16MB flash and a partition offset at 0x10000:

```
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
CONFIG_APP_EXCLUDE_PROJECT_VER_VAR=
CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR=
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16

#
# Arduino Configuration
#
CONFIG_ENABLE_ARDUINO_DEPENDS=y
CONFIG_AUTOSTART_ARDUINO=
CONFIG_ARDUINO_RUN_CORE0=
CONFIG_ARDUINO_RUN_CORE1=y
CONFIG_ARDUINO_RUN_NO_AFFINITY=
CONFIG_ARDUINO_RUNNING_CORE=1
CONFIG_ARDUINO_EVENT_RUN_CORE0=
CONFIG_ARDUINO_EVENT_RUN_CORE1=y
CONFIG_ARDUINO_EVENT_RUN_NO_AFFINITY=
CONFIG_ARDUINO_EVENT_RUNNING_CORE=1
CONFIG_ARDUINO_UDP_RUN_CORE0=
CONFIG_ARDUINO_UDP_RUN_CORE1=y
CONFIG_ARDUINO_UDP_RUN_NO_AFFINITY=
CONFIG_ARDUINO_UDP_RUNNING_CORE=1
CONFIG_DISABLE_HAL_LOCKS=

#
# Debug Log Configuration
#
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_NONE=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_ERROR=y
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_WARN=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_INFO=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_DEBUG=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_VERBOSE=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL=1
CONFIG_ARDUHAL_LOG_COLORS=
CONFIG_ARDUHAL_ESP_LOG=y
CONFIG_ARDUHAL_PARTITION_SCHEME_DEFAULT=y
CONFIG_ARDUHAL_PARTITION_SCHEME_MINIMAL=
CONFIG_ARDUHAL_PARTITION_SCHEME_NO_OTA=
CONFIG_ARDUHAL_PARTITION_SCHEME_HUGE_APP=
CONFIG_ARDUHAL_PARTITION_SCHEME_MIN_SPIFFS=
CONFIG_ARDUHAL_PARTITION_SCHEME="default"
CONFIG_ARDUINO_SELECTIVE_COMPILATION=y
CONFIG_ARDUINO_SELECTIVE_ArduinoOTA=y
CONFIG_ARDUINO_SELECTIVE_AsyncUDP=y
CONFIG_ARDUINO_SELECTIVE_AzureIoT=
CONFIG_ARDUINO_SELECTIVE_BLE=
CONFIG_ARDUINO_SELECTIVE_BluetoothSerial=
CONFIG_ARDUINO_SELECTIVE_DNSServer=y
CONFIG_ARDUINO_SELECTIVE_EEPROM=y
[... 851 lines cut ...]
CONFIG_UNITY_ENABLE_FIXTURE=

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096

#
# Wi-Fi Provisioning Manager
#
CONFIG_WIFI_PROV_SCAN_MAX_ENTRIES=16

#
# Supplicant
#
CONFIG_WPA_WPS_WARS=
```

ntwallace, 2020-09-29:
Ok I copied the arduino sdkconfig and changed only the options I need to run and I'm still getting the boot loop with NimBLE::init() in the code. Built using make.
```
CONFIG_AUTOSTART_ARDUINO=

CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y
CONFIG_ESPTOOLPY_FLASHSIZE="16MB"

CONFIG_PARTITION_TABLE_SINGLE_APP=
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_OFFSET=0x10000
```

make doesn't indicate that I'm running out of DRAM or IRAM:
```
Total sizes:
 DRAM .data size:   16772 bytes
 DRAM .bss  size:   42824 bytes
Used static DRAM:   59596 bytes (  64984 available, 47.8% used)
Used static IRAM:  105320 bytes (  25752 available, 80.4% used)
      Flash code:  905560 bytes
    Flash rodata:  261304 bytes
Total image size:~1288956 bytes (.bin may be padded larger)
```

I also bumped the app partitions to 3Mb each just to make sure there wasn't a flash issue 
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x11000, 0x4000,
otadata,  data, ota,     ,        0x2000,
phy_init, data, phy,     ,        0x1000,
factory,  app,  factory, 0x20000, 3M,
ota_0,    app,  ota_0,   ,        3M,
ota_1,    app,  ota_1,   ,        3M,
```

ntwallace, 2020-09-29:
@h2zero One final comment, it does seem after all of that this is likely an IRAM/DRAM issue. I disabled all tasks in my app_main() and was able to get it to run normally. After renabling them one by one, I found that one specific task that is very IRAM and DRAM heavy was causing this. I'm going to keep debugging until I know what exactly is causing it and will update with a final solution for future reference.

Thanks for your patience and responsiveness.

h2zero, 2020-09-30:
@ntwallace Glad you found it, was confused as to how that error occurred. 

Thanks for keeping me up to date with the progress. I will update the cmake files for future users (component was missing stuff as well). 

Feedback is always appreciated, please let me know how you make out with this.

ntwallace, 2020-10-03:
@h2zero turns out it was some kind of issue with dynamic DRAM allocation. Commenting out my OTA code stopped the abort loop, and I worked backwards from there. Changing the menuconfig setting `CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY` to load large system libraries like lwip into PSRAM instead of DRAM .bss allows me to run OTA and NimBLE without issue. 

Having said that, I'm not sure that NimBLE is actually getting loaded in PSRAM after I enabled `CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL`. Running make size-components shows the NimBLE component still taking up a good chunk of DRAM. Is that what yours looks like too?
```
            Archive File DRAM .data & .bss  IRAM  Flash code & rodata  Total
               libmain.a         54   6579    208     140994    13188  161023
           libnet80211.a        938      0  10459      91592    21856  124845
           libbtdm_app.a        660   2386  23415      66548     4158   97167
               liblwip.a         21      0   8616      69485     5066   83188
libesp-nimble-component.        256   5107     48      68375     4227   78013
```

h2zero, 2020-10-04:
@ntwallace Thanks for the update, I'm glad you found the source of that issue.

As for the PSRAM I'm not sure if what you're seeing there is normal. I have not personally tested this config but a few others around here have reported success with that. If you suspect it is not working I would encourage creating a new issue so it's more visible to others who may have used this. I will try to test it on my end also.