h2zero/NimBLE-Arduino #34: Should we pass UUIDs (and maybe other objects) as const reference?

Jeroen88, 2020-05-06:
I did a small experiment, changing in BLEClient this function `BLERemoteService *                         getService(const BLEUUID &uuid)`, so I passed a const reference instead of the object. This lead to having to change in BLEUUID the function `std::string toString() const`, because the compiler complaint about passing a non-const this parameter.

In a very small program this saved 12 bytes of code, not a lot. **BUT** this should have a great **positive impact on performance**, because now not the full object is copied onto the stack, but just a reference (pointer) to it.

I think it is a lot of work, to change all relevant class member functions, but I think it is worth the effort, **if there are no negative side effects**. What do you think?

h2zero, 2020-05-06:
Yes there are quite a few optimization’s that can be done. 
That is something on my list to look at in the future. 
For the moment though we should make sure what exists already is working as best it can and add any missing pieces. 

Premature optimization is the root of all evil as is said.

Jeroen88, 2020-05-07:
> Premature optimization is the root of all evil as is said.

Sure, but I think this is not an optimization, adding `const` merely is a modifier that better reflects what the parameter is used for: an input parameter that is left unchanged by the function. Changing the parameter passing from value to reference doesn't change the (high level cpp) code, but does change the machine code, making it more efficient (and a bit smaler).

Closing for now.

closed 2020-05-07 by Jeroen88