h2zero/NimBLE-Arduino #43: Shouldn't NimBLEScan::getDevice() be protected by semaphore?

Jeroen88, 2020-05-14:
In NimBLEScan::getDevice() [here]( https://github.com/h2zero/NimBLE-Arduino/blob/9c011aa37402b60d6d06acc9581fe2e10ac00550/src/NimBLEScan.cpp#L399 ) the device to return to the caller is assigned to a temporary variable. Is this unary? In other words, could the background scan update the device halfway copying the device? Or is this impossible? If possible, could / should it be protected by semaphore?

h2zero, 2020-05-14:
Yes it seems possible that the data could be modified while copying, I don't think that method wasn't meant to be used while scanning is active. The idea of those related methods seems to be that you would use the blocking scan call, then when complete, you would access the results via this.

In the scenario you describe, yes a semaphore would make sense to stay on the safe side.

h2zero, 2021-02-01:
Going to close this, as the scenario that may cause this issue is extremely unlikely. 
Also, now that I've said it, I'm sure an issue involving this situation will be posted shortly 😆

closed 2021-02-01 by h2zero