h2zero/NimBLE-Arduino #51: Just want to thank you all!

mitchjs, 2020-05-19:
i just want to thank you guys for giving life to ESP32 BLE

@h2zero  for pickup up the pieces on the old stale repository
@Jeroen88  for all the performance/memory improvements

(i'm using the non Arduino version, but the core the same)

thanks,
Mitch

h2zero, 2020-05-19:
@mitchjs You are most welcome 😄, NimBLE has been the best motivator to continue the great work @nkolban and  @chegewara started. I'm very pleased that others are taking interest and seeing improvements. 

@Jeroen88 has been an amazing help to optimize things and greatly accelerated the progress.

I could make a big list of thank you's, but I'll just say thank you to all you use this repo and/or the original, your use and feedback are incredibly helpful.

Jeroen88, 2020-05-19:
@mitchjs Thank you very much for your kind words!!

@h2zero Thank you too. It was a pleasure working together getting the optimizations done so quickly. I am looking forward to the improved semaphores and the non-automatic discovery, two other big steps forward!

@nkolban and @chegewara created great libary. Without it, I would never have started to even think about using Bluetooth. You greatly reduced the complexity to communicate using BLE!

I like the open source community very much. Complete strangers create beautiful things that other people may use. Why? Just because... I am one of those lucky users, e.g. of the Arduino IDE, PlatformIO, Eclipse, the ESP32 core libraries, ESP8266 core libraries, NodeJS, git, github, etc. I am glad that I could make a contribution this time, I hope for the benefit of a lot of people!

chegewara, 2020-05-19:
BLE library was my very first contact with mcu programming and im happy i found it. I have come a long way since then and i am very happy i found it in first place.
Now @h2zero is doing great job with NimBLE library, giving users hope to build projects that were not possible because of size of bludroid library or because of bugs that still are in arduino BLE library.
Lets hope we can also have ble mesh with nimble very soon.

Recently ive hear rumors that we will have esp32/esp BLE chip (not current esp32), i cant wait to get it in my hands.

h2zero, 2020-05-19:
@chegewara mesh coming soon, once I get done with some of my time constraints and a couple more tweaks to this. 

Do you know what the new chip is called? I’d love to be able to use long range BLE.

bluelu, 2020-07-17:
I was running out of memory when I was putting together my different esphome components and added the bluetooth component. Your arduino port saved me! Thank you! :)

chegewara, 2020-07-17:
> Do you know what the new chip is called? I’d love to be able to use long range BLE.

esp32 S3

1technophile, 2020-08-02:
Thanks a lot for this library, we use to say "1 stone 2 birds" but in my case it's more 3 or 4 birds since I'm getting my hands on your work!
It's magic :-)
And thanks for the [OpenMQTTGateway ](https://github.com/1technophile/OpenMQTTGateway) users who will benefit from this.

ntwallace, 2020-08-13:
Big thanks for this library, it significantly reduced flash usage, and program complexity! So great to have the BLE stack running in PSRAM.

h2zero, 2020-08-14:
> Big thanks for this library, it significantly reduced flash usage, and program complexity! So great to have the BLE stack running in PSRAM.

I think this is the best part of the library. The ability to now tweak the settings rather than rely on the pre-compiled archive from the built-in bluedroid stack makes personalizing applications much more comprehensive. Now we can load the stack into PSRAM and connect to 9 devices, eliminate code and save memory all by adjusting settings in nimconfig. Many may not use this but it's a nice tool to have. Makes me happy to see this being used 😄.

Torxgewinde, 2021-01-10:
Connecting to two devices was a pain with the Bluedroid-Stack and just by switching to Nimble the very same sourcecode (except changing the header to Nimble) works without stalling and triggering the watchdog. You cannot imagine how happy I was to find that you made switching so easy! **Thanks a lot!**

h2zero, 2021-01-10:
Welcome @Torxgewinde, that was the goal. I had multi-client purpose in mind from the start, very glad to hear it's working for you.

pix3lize, 2021-01-24:
@bluelu Could you share how you add the nimble to esphome custom component, would greatly appreciate :). I'm stuck on adding Wifi and Bluetooth on esphome and end up disable the display (st7789v).

teajohnson90, 2021-05-18:
Thanks for doing this, really appreciated. I was switching between ESP-IDF and arduino code because the original size was too large.

h2zero, 2021-05-19:
@teajohnson90 you're welcome :smiley: good to hear it helped you!