h2zero/NimBLE-Arduino #55: Advertised device payload not copied?

Jeroen88, 2020-05-20:
Is there any good reason why payload is not copied into the advertised device member variable std::string m_payload? Now only the pointer returned from the stack is saved, see [here](https://github.com/h2zero/NimBLE-Arduino/blob/9c011aa37402b60d6d06acc9581fe2e10ac00550/src/NimBLEAdvertisedDevice.cpp#L549).

h2zero, 2020-05-20:
I think the simple answer would be to save resources. 31 bytes + std::string overhead per device found. It’s also what was in the original library so...

Jeroen88, 2020-05-20:
But manufacturer data _is_ copied, that is inconsistent, isn't it?

Can I make a PR to copy it? Or will the payload of _all_ scanned devices be accessible through the scan object? I doubt it...

chegewara, 2020-05-20:
Maybe this time would be better to left user option to copy payload and other advertising data, this wont use memory in case user is not interested some found devices, especially useful in dense environment.

I think low level stack is keeping only few most recent seen devices, at least in bluedroid.

Jeroen88, 2020-05-20:
@chegewara I see your point. @chegewara @h2zero this could also be set dynamically, like the auto pre discovery. I would opt for all or nothing, so store all data in the advertised device including the payload, or nothing at all. Also I would opt for changing payload into a `std::string` instead of a `uint8_t *`, because all other advertised device data are `std::string`'s. For consistency.

chegewara, 2020-05-20:
> Also I would opt for changing payload into a `std::string` instead of a `uint8_t *`, because all other advertised device data are `std::string`'s.

I dont agree. Payload is type of uint8* and cant be treated as string. I know you may have not printable chars in string, even 0x0 in middle of it, and maybe even it is C++ acceptable, but this is not good idea. If you want to go that way you can replace all std:string with arduino String type.

Jeroen88, 2020-05-20:
std::string is (mis-) used throughout this library and the original library

chegewara, 2020-05-20:
Yes, i know. Maybe its time to fix it and use variable types that should be used? Just saying, if we know that something was wrong that does not mean it should be replicated.

h2zero, 2020-05-20:
I agree with @chegewara on this. We already parse the data from the payload and store it anyway, no need for another copy. If the user wants to save it they can copy it to whatever container they want with the pointer. 

Not sure why you’re using this anyway? Is there some data in the payload that isn’t available from the advertisedDevice object?

h2zero, 2020-05-20:
> Yes, i know. Maybe its time to fix it and use variable types that should be used? Just saying, if we know that something was wrong that does not mean it should be replicated. Lets try to make this library better than the other one.

I fear the PR flood gates just got smashed with this comment lol.

Jeroen88, 2020-05-20:
> I fear the PR flood gates just got smashed with this comment lol.

Lol, time for a revolution and a contra revolution. Finally! :D 

In #48 the pre revolution started already, but then it was a one day war only... ;)

> We already parse the data from the payload and store it anyway

Nope, the fields are parsed [here](https://github.com/h2zero/NimBLE-Arduino/blob/a8f6e244e13559df33df6ae44c637ca43b231e51/src/NimBLEScan.cpp#L143) and the payload is set (pointer copied) just [below](https://github.com/h2zero/NimBLE-Arduino/blob/a8f6e244e13559df33df6ae44c637ca43b231e51/src/NimBLEScan.cpp#L145).

> Not sure why you’re using this anyway? Is there some data in the payload that isn’t available from the advertisedDevice object?

Short answer: yes, e.g.:
```
Address is a4:c1:38:5d:ef:16
Service data UUID is 0xfe95
0x30, 0x58, 0x5b, 0x05, 0x01, 0x16, 0xef, 0x5d, 0x38, 0xc1, 0xa4, 0x28, 0x01, 0x00, 
Payload is:
0x02, 0x01, 0x06, 0x15, 0x16, 0x95, 0xfe, 0x50, 0x20, 0xaa, 0x01, 0xb3, 0x58, 0x22, 0x39, 0x34, 0x2d, 0x58, 0x0d, 0x10, 0x04, 

```
I do not know what I want with it exactly. I am looking for a way to do a scan and filter out the devices that I support. One way is with getName(), but 2 of my 3 devices don't have a name. I could read the characteristic, but then I have to connect. I would like to determine one way or another what from the advertised data which devices I have support for. It must be generic enough to recognize all same devices, but specific enough to differentiate between e.g. two different devices of the same manufacturer. Any suggestions?

Staars, 2020-05-20:
>Any suggestions?

You simply have to parse a MiBeacon and make the decision based on the PID (the value after the frame_ctrl). In your example `0x5b 0x05` stands for the LYWSD03 and here you will have to connect and read the data (or you have the beacon key to decrypt the ADV).
Most of the other Xiaomi/Mijia-sensors should provide their data in the ADV, so you can parse it directly from the scan data. 
This approach works for me.

Look here for further info: https://github.com/Magalex2x14/LYWSD03MMC-info

Jeroen88, 2020-05-20:
@Staars Thnx Christian! Is that generic? In other words, has every unique device a unique code at bytes 2 and 3 of the service data? Or is this Xiaomi specific?

> Look here for further info

I found that link before, it was already a great help!

I am still looking for the meaning of all of the service data, do you happen to know the full meaning of the service data?

The MJ_HT_V1 has service data:
`0x50, 0x20, 0xaa, 0x01, 0xb3, 0x58, 0x22, 0x39, 0x34, 0x2d, 0x58, 0x0d, 0x10, 0x04, 0xdd, 0x00, 0x5e, 0x02
`
I know byte 11 discriminates if the data from byte 14 is temp / humidity / battery or a combination of temp and humidity. But what is byte 12 and 13? Now I think the first two bytes are a frame counter? Is this generic? Bytes 2 and 3 some kind of device ID, as you explained, is this also true for the MJ_HT_V1?Is this always a device ID? What is byte 4? I found out bytes 5-10 is the address.

The LYWSD03 has service data:
`0x30, 0x58, 0x5b, 0x05, 0x01, 0x16, 0xef, 0x5d, 0x38, 0xc1, 0xa4, 0x28, 0x01, 0x00
`
Again in order: 2 bytes frame counter, 2 bytes device ID, one (for me) unknown byte, 6 bytes address, andthe last 3?

Jeroen88, 2020-05-20:
@Staars Thank you Christian for directing me to that site again. I now almost know the structure of the service data:

```
struct ServiceData {
  uint16_t frameCtrl;
  uint16_t deviceType;
  uint8_t frameCnt;
  uint8_t MACAddress[6];
  uint8_t variantType;
  uint16_t unknown;
  union {
    uint16_t temperature;
    uint16_t humidity;
    uint8_t battery;
    struct {
      uint16_t temperature;
      uint16_t humidity;
    } both;
  } variants;
};

```

Just 2 bytes to discover: do you know what the 2 bytes at position 12 and 13 mean?

Also still looking what the last bytes mean if byte 11 equals 0xdb.

Is this structure generic for every BLE peripheral or for Xiaomi only?

Staars, 2020-05-22:
@Jeroen88 (a bit late ;)

> Just 2 bytes to discover: do you know what the 2 bytes at position 12 and 13 mean?
You mean the last two bytes before the data? It is this enum:
https://github.com/MiEcosystem/mijia_ble_common/blob/847ee73ca9ba496a91483d300a8bd6a7666a8c0c/mible_beacon.h#L28

> Also still looking what the last bytes mean if byte 11 equals 0xdc.
I am not sure, but my best guess is: RSSI

> Is this structure generic for every BLE peripheral or for Xiaomi only?
No, this is Xiaomi only, but BTW I am not aware of any other sensor families (temp/hum-stuff for homes) out there. 

> Do you happen to know if the LYWSD03 can be read without connection?
Yes, it can be done, but with many "if's" and everything is described in the link from my older post.
Basically:
1. A fresh LYWSD03 will only send advertisements without sensor data ("unbinded" frame):
`30 58  5b 05   01   xx xx xx xx xx xx   28 01 00`
2. After pairing with your phone with the MiHome-App this will change to an "encrypted" frame (only about every 10 minutes!!).
3. When you have sniffed the "bind_key", you can decrypt the data without a connection purely from the advertisement.

My personal opinion is, that I find it more convenient to just to connect to the LYWSD03 from time to time. I prefer to avoid the relatively complicated procedure to obtain the "bind_key". Another advantage of the connection-metod is, that you can read the real battery voltage instead of the questionable battery-percentage-level.

Jeroen88, 2020-05-23:
Thx @Staars for the extensive answers! 

> It is this enum:

Going  to check that link later!

> must be changed to:

Are you sure that it is not the other way around, `uint8_t type` and `uint16_t len`?

>  I find it more convenient to connect to the LYWSD03

Yes I thin you are right, that is the easiest option.

Staars, 2020-05-23:
> Are you sure that it is not the other way around, uint8_t type and uint16_t len?

Pretty sure.
Maybe it is a bit confusing to take a look at the payload seeing (and then thinking) in single bytes, but in reality the type is a word and then the byte order is swapped. So for the MJ_HT_V1 the type is always `10xx` which becomes `xx 10` in the byte stream, followed by 1,2 or 4 for the length in bytes for the actual data (temp, hum, bat or temp+hum).
At least I got confused regularly 😁

Jeroen88, 2020-05-23:
:laughing: thnx!

h2zero, 2021-02-01:
Closing this as recent commits have addressed some of this issue. Version 2.0.0 will address the rest.

closed 2021-02-01 by h2zero