h2zero/NimBLE-Arduino #58: Problems connecting during an infinite scan

Jeroen88, 2020-05-22:
If I start an infinite scan and next try to connect to an available address or a non existing address, it seems that the stack is busy scanning, but the connection process does not time-out, nor connects.

MVCE:

```
#include <Arduino.h>
#include <NimBLEDevice.h>

BLEClient *pClient = nullptr;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(230400);
  Serial.println("/n/nStarted");

  BLEDevice::init("ESP32");

  Serial.println("Start background scan forever...");
  static BLEScan *pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setActiveScan(false);
  pBLEScan->setInterval(256);
  pBLEScan->setWindow(255);

  pBLEScan->start(0, nullptr, false);
  Serial.println("Background scan started");

  pClient = BLEDevice::createClient();

//  BLEAddress address("a4:c1:38:5d:ef:16");
  BLEAddress address("01:02:03:04:05:06");
  Serial.println("Connecting...");
  pClient = BLEDevice::createClient();
  bool success = pClient->connect(address);
  if(!success) {
    Serial.println("Failed to connect");
  } else {
    Serial.println("Connected!");
    pClient->disconnect();
  }
  BLEDevice::deleteClient(pClient);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

If I limit the scan for e.g. 10s and do the same test _after_ the scan everything works as expected. Below is about 1 minute logging for the invalid address. No timeouts occur. The logging for the existing address is similar. 

```
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
Start background scan forever...
D NimBLEScan: ">> start(duration=0)"
D FreeRTOS: "Semaphore taking: name: ScanEnd (0x3ffc88cc), owner: <N/A> for start"
D FreeRTOS: "Semaphore taken:  name: ScanEnd (0x3ffc88cc), owner: start"
D NimBLEScan: "<< start()"
Background scan started
Connecting...
D NimBLEClient: ">> connect(01:02:03:04:05:06)"
D FreeRTOS: "Semaphore taking: name: OpenEvt (0x3ffc8c4c), owner: <N/A> for connect"
D FreeRTOS: "Semaphore taken:  name: OpenEvt (0x3ffc8c4c), owner: connect"
I NimBLEScan: "NEW DEVICE FOUND: 70:56:81:df:dd:15"
D NimBLEAdvertisedDevice: "- setRSSI(): rssi: -80"
D NimBLEAdvertisedDevice: "- manufacturer data: 4c0009060202c0a8b2dc"
I NimBLEScan: "NEW DEVICE FOUND: a4:c1:38:5d:ef:16"
D NimBLEAdvertisedDevice: "- setRSSI(): rssi: -59"
I NimBLEScan: "NEW DEVICE FOUND: 58:2d:34:39:22:58"
D NimBLEAdvertisedDevice: "- setRSSI(): rssi: -72"
I NimBLEScan: "NEW DEVICE FOUND: 9c:20:7b:de:4c:ad"
D NimBLEAdvertisedDevice: "- setRSSI(): rssi: -94"
D NimBLEAdvertisedDevice: "- manufacturer data: 4c0009060201c0a8b20a"
I NimBLEScan: "NEW DEVICE FOUND: ac:9a:22:cd:12:91"
D NimBLEAdvertisedDevice: "- setRSSI(): rssi: -75"
D NimBLEAdvertisedDevice: "- setName(): name: iSensor "
D NimBLEAdvertisedDevice: "- manufacturer data: 00db974642027e7a"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: a4:c1:38:5d:ef:16"
D NimBLEAdvertisedDevice: "- setRSSI(): rssi: -59"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 70:56:81:df:dd:15"
D NimBLEAdvertisedDevice: "- setRSSI(): rssi: -83"
D NimBLEAdvertisedDevice: "- manufacturer data: 4c0009060202c0a8b2dc"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 9c:20:7b:de:4c:ad"
D NimBLEAdvertisedDevice: "- setRSSI(): rssi: -96"
D NimBLEAdvertisedDevice: "- manufacturer data: 4c0009060201c0a8b20a"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 58:2d:34:39:22:58"
D NimBLEAdvertisedDevice: "- setRSSI(): rssi: -77"
I NimBLEScan: "NEW DEVICE FOUND: 69:e3:79:98:b6:9f"
D NimBLEAdvertisedDevice: "- setRSSI(): rssi: -96"
D NimBLEAdvertisedDevice: "- manufacturer data: 4c000c0e005ff8cdad20b24d1254087e38461005471c99c674"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: ac:9a:22:cd:12:91"
D NimBLEAdvertisedDevice: "- setRSSI(): rssi: -76"
D NimBLEAdvertisedDevice: "- setName(): name: iSensor "
D NimBLEAdvertisedDevice: "- manufacturer data: 00db97464202807c"
```

h2zero, 2020-05-22:
Scan has to be stopped to connect and started again after. Not my rules but NimBLEs. The stack has to use the scanner to connect.

h2zero, 2020-05-22:
Re-organized your code:
```
#include <Arduino.h>
#include <NimBLEDevice.h>

BLEClient *pClient = nullptr;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(230400);
  Serial.println("/n/nStarted");

  BLEDevice::init("ESP32");

  pClient = BLEDevice::createClient();

//  BLEAddress address("a4:c1:38:5d:ef:16");
  BLEAddress address("01:02:03:04:05:06");
  Serial.println("Connecting...");
  pClient = BLEDevice::createClient();
  bool success = pClient->connect(address);
  if(!success) {
    Serial.println("Failed to connect");
  } else {
    Serial.println("Connected!");
    //pClient->disconnect(); <- not needed (done for you in deleteClient())
  }
  BLEDevice::deleteClient(pClient);

  Serial.println("Start background scan forever...");
  static BLEScan *pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setActiveScan(false);
  pBLEScan->setInterval(256);
  pBLEScan->setWindow(255);

  pBLEScan->start(0, nullptr, false);
  Serial.println("Background scan started");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

That should work.

Jeroen88, 2020-05-22:
@h2zero Thnx, yes that is what I understood from your advise. But my use case is: I have an infinite scan running that updates the service data of the devices it scans. Every now and then I want to connect, e.g. to get other data than the service data.

As I understand now, I have to stop the scan, connect, read, disconnect and restart the scan.

Would it be possible to automate this? I could imagine connect() to pause a scan and restart it as soon as possible. Orr is it not worth the effort?

Also, I think that I see that scan data is [thrown away](https://github.com/h2zero/NimBLE-Arduino/blob/a8f6e244e13559df33df6ae44c637ca43b231e51/src/NimBLEScan.cpp#L91) on a stopped scan. Why?

h2zero, 2020-05-22:
Yes it would be possible to automate in the connect method but problematic for other users. 
Not everyone wants to keep scanning after connecting, many (maybe most) uses are scan > connect > get notifications. The onconnect() callback is the best place to handle this use case.

That code will be deleted soon as it was more of an issue in bluedroid. It would still send some results after calling scan stop and mess with the callback. NimBLE doesn’t do this so that code isn’t really doing anything now. I plan to do more work on that class soon.

Jeroen88, 2020-05-31:
If I interrupt the scan in the program, connect, disconnect and restart the scan everything works. So: closing for now.

closed 2020-05-31 by Jeroen88

michaelboeding, 2024-04-01:
What if I'm running an application that can connect to multiple clients? Basically I would like to scan forever and connect to multiple clients (create a new task to handle each client). But have the scanning find and then transfer them at all times to the handling task? Do I have to pause scanning while connecting and then restart it after a successful connection or failed connection? 

Note I'm also using this for the esp-IDF not Arduino but thought this thread was relevant.

h2zero, 2024-04-01:
Hello @michaelboeding. Scanning will stop automatically when attempting a connection, this is required for connection establishment. After connecting you'll need to restart scanning.

thorrak, 2024-04-01:
Out of curiosity @h2zero — do you have any thoughts about how that might interplay with active scanning?

I’m wondering if this might relate to the long-standing issue where active scans begin to silently fail.

h2zero, 2024-04-06:
@thorrak I don't think this is related, that's a controller bug or something else.

@michaelboeding You're welcome :)