h2zero/NimBLE-Arduino #65: Can not connect a second time to a re-created client

Jeroen88, 2020-05-28:
MVCE

```
#include <NimBLEDevice.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\n\nStarted");

  BLEDevice::init("ESP32");

  Serial.println("Create");
  NimBLEClient *pClient = BLEDevice::createClient();
  Serial.println("Connecting...");
  bool success = pClient->connect(NimBLEAddress("a4:c1:38:5d:ef:16"));
  if(success) {
    Serial.println("Connected");
    pClient->disconnect();
    Serial.println("Disconnected");

    BLEDevice::deleteClient(pClient);
    Serial.println("Create");
    pClient = BLEDevice::createClient();
    
    Serial.println("Connecting...");
    success = pClient->connect(NimBLEAddress("a4:c1:38:5d:ef:16"));
    if(success) {
      Serial.println("Connected");
      pClient->disconnect();
      Serial.println("Disconnected");
    }
  }
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Logging:
```
Started
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
Create
Connecting...
D NimBLEClient: ">> connect(a4:c1:38:5d:ef:16)"
D FreeRTOS: "Semaphore taking: name: OpenEvt (0x3ffc8870), owner: <N/A> for connect"
D FreeRTOS: "Semaphore taken:  name: OpenEvt (0x3ffc8870), owner: connect"
D FreeRTOS: ">> wait: Semaphore waiting: name: OpenEvt (0x3ffc8870), owner: connect for connect"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc8870), owner: connect"
D FreeRTOS: "<< wait: Semaphore released: name: OpenEvt (0x3ffc8870), owner: <N/A>"
D NimBLEClient: "Refreshing Services for: (a4:c1:38:5d:ef:16)"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
D NimBLEClientCallbacks: "onConnect: default"
D NimBLEClient: "<< connect()"
Connected
D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
Disconnected
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
Create
Connecting...
D NimBLEClient: ">> connect(a4:c1:38:5d:ef:16)"
D FreeRTOS: "Semaphore taking: name: OpenEvt (0x3ffc8870), owner: <N/A> for connect"
D FreeRTOS: "Semaphore taken:  name: OpenEvt (0x3ffc8870), owner: connect"
D FreeRTOS: ">> wait: Semaphore waiting: name: OpenEvt (0x3ffc8870), owner: connect for connect"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Got Client event "

```

... and then just "hangs"

Staars, 2020-05-28:
This reminds of my last weekend, where I finally was "sure" that my ESP32 was broken.

For me the solution was to erase the flash , which seemed to have solved it. In the end I erased it basically before every upload.

What IDE are you using? If you are using some kind of caching mechanism, you might need to delete the compiler cache too.

Jeroen88, 2020-05-28:
I had a "broken" ESP32 too a few days ago, put it apart, I have not thrown it away yet. I'll try to erase the flash to see if that brings it back to life.

But this is another issue. I can reset to run the sketch again. I can re-flash and start the program over again. But is won't get passed the second `connect()`. As you can see in the logging, while connecting the first time it logs:

```
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"

```
and the second time it logs:

```
D NimBLEClient: "Got Client event "
D NimBLEClient: "Got Client event "

```
and then waits forever...

Most of the time, also this time, I use the Arduino IDE, 1.8.10 (that is the one but last I believe). Every now and then I use pio.

h2zero, 2020-05-28:
Turn on event code strings to see what
```
D NimBLEClient: "Got Client event "
D NimBLEClient: "Got Client event "
```
actually are. 

Looks like you're hanging on a semaphore, or the stack is getting stuck because the peripheral has not disconnected yet. Try putting a delay of 100ms or so between deleting and creating the client.

Jeroen88, 2020-05-28:
Logging now:

```
Started
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
Create
Connecting...
D NimBLEClient: ">> connect(a4:c1:38:5d:ef:16)"
D FreeRTOS: "Semaphore taking: name: OpenEvt (0x3ffc88fc), owner: <N/A> for connect"
D FreeRTOS: "Semaphore taken:  name: OpenEvt (0x3ffc88fc), owner: connect"
D FreeRTOS: ">> wait: Semaphore waiting: name: OpenEvt (0x3ffc88fc), owner: connect for connect"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_CONNECT "
D NimBLEClient: "Connection established"
D FreeRTOS: "Semaphore giving: name: OpenEvt (0x3ffc88fc), owner: connect"
D FreeRTOS: "<< wait: Semaphore released: name: OpenEvt (0x3ffc88fc), owner: <N/A>"
D NimBLEClient: "Refreshing Services for: (a4:c1:38:5d:ef:16)"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
D NimBLEClientCallbacks: "onConnect: default"
D NimBLEClient: "<< connect()"
Connected
D NimBLEClient: ">> disconnect()"
D NimBLEClient: "<< disconnect()"
Disconnected
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
Create
Connecting...
D NimBLEClient: ">> connect(a4:c1:38:5d:ef:16)"
D FreeRTOS: "Semaphore taking: name: OpenEvt (0x3ffc88fc), owner: <N/A> for connect"
D FreeRTOS: "Semaphore taken:  name: OpenEvt (0x3ffc88fc), owner: connect"
D FreeRTOS: ">> wait: Semaphore waiting: name: OpenEvt (0x3ffc88fc), owner: connect for connect"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_MTU"
D NimBLEClient: "Got Client event BLE_GAP_EVENT_DISCONNECT"

```

I think the issue is indeed that `BLE_GAP_EVENT_DISCONNECT` arrives AFTER calling `connect()` again. 

I think `disconnect()` should wait for this event...!

Jeroen88, 2020-05-28:
Adding a **200 ms** delay between `createClient()` and `connect()` solves the problem, 100 ms is too short

Staars, 2020-05-29:
> Adding a **200 ms** delay between `createClient()` and `connect()` solves the problem, 100 ms is too short. Still I think `disconnect()` should wait for the event.

Interesting observation, I will test this too.

Jeroen88, 2020-05-29:
In the integration I am creating several devices are read out independently. Most don't need a connection, some do. I just want to use the functionality that is there. Also, I think it should always work

Could you give me some directions how to check if the client has disconnected? The function isDisconnected() does not do this job

chegewara, 2020-05-29:
> For me the solution was to erase the flash , which seemed to have solved it. In the end I erased it basically before every upload.

This is must when you are building bluetooth app.

chegewara, 2020-05-29:
> D FreeRTOS: ">> wait: Semaphore waiting: name: OpenEvt (0x3ffc88fc), owner: connect for connect"
> D NimBLEClient: "Got Client event BLE_GAP_EVENT_MTU"
> D NimBLEClient: "Got Client event BLE_GAP_EVENT_DISCONNECT"

This may have happen when there is no successful connection, maybe except MTU exchange, because this requires full connection established, unless MTU is called too early (in wrong place).

Staars, 2020-05-29:
I have learned this   ….   after several hours :)
(It was not obvious to me before)

h2zero, 2020-05-30:
This issue was just resolved in commit 4c798fe.

Jeroen88, 2020-05-31:
> This issue was just resolved in commit 4c798fe.

I can confirm that. Thank you @h2zero!

closed 2020-05-31 by Jeroen88