h2zero/NimBLE-Arduino #66: Any idea why it crashes?

Jeroen88, 2020-05-29:
I have a too complicated sketch to publish, but I create clients, connect, read, disconnect and delete clients in a loop with 3 peripherals.

I get every now and then crashes like:

```
0x400961ed: multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 214
0x400837b6: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 268
0x4008307d: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 42
0x4015a0f9: operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc line 46
0x400d6d26: NimBLEDevice::removeIgnored(NimBLEAddress const&) at /home/jeroen/Downloads/arduino-1.8.10-linux64/arduino-1.8.10/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/ext/new_allocator.h line 110
0x400d5e44: NimBLEClient::handleGapEvent(ble_gap_event*, void*) at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/NimBLEClient.cpp line 641
0x400dbdc9: ble_gap_call_event_cb at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_gap.c line 628
0x400dcb8a: ble_gap_conn_broken at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_gap.c line 1086
0x400dcbb8: ble_gap_rx_disconn_complete at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_gap.c line 1113
0x400e17ea: ble_hs_hci_evt_disconn_complete at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c line 157
0x400e1b58: ble_hs_hci_evt_process at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c line 889
0x400df939: ble_hs_event_rx_hci_ev at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_hs.c line 520
0x400e772a: nimble_port_run at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/nimble_npl_os.h line 121
0x400d696f: NimBLEDevice::host_task(void*) at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/NimBLEDevice.cpp line 429
0x40093535: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

and like

```
0x40092ea9: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 720
0x400d4a15: FreeRTOS::Semaphore::give() at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/FreeRTOS.cpp line 153
0x400d4a35: FreeRTOS::Semaphore::give(unsigned int) at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/FreeRTOS.cpp line 169
0x400d5e6c: NimBLEClient::handleGapEvent(ble_gap_event*, void*) at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/NimBLEClient.cpp line 666
0x400dbdc9: ble_gap_call_event_cb at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_gap.c line 628
0x400dcb8a: ble_gap_conn_broken at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_gap.c line 1086
0x400dcbb8: ble_gap_rx_disconn_complete at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_gap.c line 1113
0x400e17ea: ble_hs_hci_evt_disconn_complete at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c line 157
0x400e1b58: ble_hs_hci_evt_process at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c line 889
0x400df939: ble_hs_event_rx_hci_ev at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_hs.c line 520
0x400e772a: nimble_port_run at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/nimble_npl_os.h line 121
0x400d696f: NimBLEDevice::host_task(void*) at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/NimBLEDevice.cpp line 429
0x40093535: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

Any idea why?

chegewara, 2020-05-29:
> 0x400961ed: multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 214
> 0x400837b6: heap_caps_free at [...]

Memory corruption. Most likely you are trying to delete object that does not exist anymore or didnt exists at all.

> 0x40092ea9: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 720

Problem with semaphore in library. Most likely trying to give semaphore that does not exist or has not been taken.

Jeroen88, 2020-05-29:
Thnx @chegewara, so library problems?

Another one:
```
0x4008e300: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008e57d: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x4008313d: lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 143
0x40083261: _lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 171
0x400f6943: _vfiprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 860
0x400f0ed1: fiprintf at ../../../.././newlib/libc/stdio/fiprintf.c line 50
0x400eaec8: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 59
0x400961ed: multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 214
0x400837b6: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 268
0x4008307d: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 42
0x4015a0f9: operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc line 46
0x400d6d26: NimBLEDevice::removeIgnored(NimBLEAddress const&) at /home/jeroen/Downloads/arduino-1.8.10-linux64/arduino-1.8.10/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/ext/new_allocator.h line 110
0x400d5e44: NimBLEClient::handleGapEvent(ble_gap_event*, void*) at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/NimBLEClient.cpp line 641
0x400dbdc9: ble_gap_call_event_cb at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_gap.c line 628
0x400dcb8a: ble_gap_conn_broken at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_gap.c line 1086
0x400dcbb8: ble_gap_rx_disconn_complete at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_gap.c line 1113
0x400e17ea: ble_hs_hci_evt_disconn_complete at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c line 157
0x400e1b58: ble_hs_hci_evt_process at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_hs_hci_evt.c line 889
0x400df939: ble_hs_event_rx_hci_ev at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/host/src/ble_hs.c line 520
0x400e772a: nimble_port_run at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/nimble/nimble_npl_os.h line 121
0x400d696f: NimBLEDevice::host_task(void*) at /home/jeroen/Arduino/libraries/NimBLE-Arduino/src/NimBLEDevice.cpp line 429
0x40093535: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

```

chegewara, 2020-05-29:
> 0x400961ed: multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 214
> 0x400837b6: heap_caps_free at [...]

Still the same error, memory corruption due to delete.

h2zero, 2020-05-29:
These are all caused by the stack receiving an event and trying to call the registered callback on a client object that was deleted.

h2zero, 2020-05-29:
There is also an issue in the disconnect() method that needs to be rectified as well.

h2zero, 2020-05-29:
Try deleting [this line](https://github.com/h2zero/NimBLE-Arduino/blob/4197a1c333b10f1d8cdff4c812a47d6ac4cdc0e2/src/NimBLEClient.cpp#L223) and in your application don’t call disconnect() before deleteClient(), just call deleteClient(), it will wait for disconnection for you.

Jeroen88, 2020-05-29:
I won't have time to try things until Sunday :/

h2zero, 2020-05-30:
This issue was just resolved in commit 4c798fe

Jeroen88, 2020-05-30:
@h2zero Thank you!!! I'll test it tomorrow.

Jeroen88, 2020-05-31:
> This issue was just resolved in commit 4c798fe

@h2zero I can confirm that it is resolved. Nice!

closed 2020-05-31 by Jeroen88