h2zero/NimBLE-Arduino #67: Question: meaning advertisedDevice::getAddressType()

Jeroen88, 2020-05-30:
How should I interpret the value returned by `getAddressType()`? I found [this](https://devzone.nordicsemi.com/f/nordic-q-a/27012/how-to-distinguish-between-random-and-public-gap-addresses) post. That post explains that the address type is coded into bits, but I am not sure if this is indeed the value the function returns. If so then the return value is as follows:
- Public device address: 0
- Static device address: 111, so 7
- Non-resolvable device address: 001, so 1
- Resolvable device address: 101, so 5

Is this actually the value returned?

h2zero, 2020-05-30:
From ble.h:
```
#define BLE_ADDR_PUBLIC      (0x00)
#define BLE_ADDR_RANDOM      (0x01)
#define BLE_ADDR_PUBLIC_ID   (0x02)
#define BLE_ADDR_RANDOM_ID   (0x03)
```
Mainly you will only see the first 2.

Jeroen88, 2020-05-31:
Thnx @h2zero!

closed 2020-05-31 by Jeroen88