h2zero/NimBLE-Arduino #68: Why is NIMBLE_MAX_CONNECTIONS defined?

Jeroen88, 2020-06-01:
In NimBLEDevice the connections are stored in a dynamic container that can store any number of connections. Why limit it? If the programmer needs more connections than the limit, it can be adapted, so why don't limit it in the first place? I think there is no advantage in this.

h2zero, 2020-06-01:
The BLE controller on the esp32 has a max limit of 9 connections, 3 is the default max in IDF, so it was left at that. You could change it if you need to.

Jeroen88, 2020-06-01:
> The BLE controller on the esp32 has a max limit of 9 connections

Wouldn't it be more logical to set it to 9 then, the "real" maximum? And maybe add a note [here](https://github.com/h2zero/NimBLE-Arduino/blob/ec5b37cb98e9ae983d481fe2940f8f31732fbc35/src/NimBLEDevice.h#L79) that the controller has a max of 9 connections

>  I believe setting the max connections higher also reserves more ram for the controller

If you search for NIMBLE_MAX_CONNECTIONS here on github, you get only tests for the number of connections.

h2zero, 2020-06-01:
Sorry, I just woke up when I answered that so it wasn't a very good explanation. 

NIMBLE_MAX_CONNECTIONS was necessary due to changes between IDF versions which changed the macro name. So for this library I made this universal and checked against either IDF macro. 
The one you need to change is CONFIG_BT_NIMBLE_MAX_CONNECTIONS (default 3) in nimconfig.h or using IDF menuconfig options. 

Also if you do change this value it does reserve more memory for the stack/controller based on the number, as seen [here](https://github.com/h2zero/NimBLE-Arduino/blob/69163088075067b23a7fee3783ce47af3d6f14ff/src/nimble/host/src/ble_hs_conn.c#L35).

Jeroen88, 2020-06-01:
You are getting better during the day...! :smile: 

Thnx again, since it reserves memory it is indeed better to kee[p it this way.

closed 2020-06-01 by Jeroen88