h2zero/NimBLE-Arduino #75: (Passive) BLE Scan missing a lot of advertisments

TheNitek, 2020-06-26:
First of all: Thx for this awesome project!
I just migrated my project from the standard lib to NimBLE-Arduino and in general it works fine, but unfortunatelly I noticed that a lot of advertisments are missing.

I'm using the following code to do a endless passive scan and gather data from Xiaomi BLE sensors:

```
pBLEScan = BLEDevice::getScan();
pBLEScan->setAdvertisedDeviceCallbacks(new MiDeviceCallbacks());
pBLEScan->setActiveScan(false);
pBLEScan->start(0);
```

Using the standard lib one minute results in this:

> Friday, June 26 2020 08:34:51: Wohnzimmer Humid: 57 
> Friday, June 26 2020 08:34:52: Guave Light: 375 
> Friday, June 26 2020 08:34:56: Guave Moisture: 41% 
> Friday, June 26 2020 08:34:56: Wohnzimmer Temp: 24.6 
> Friday, June 26 2020 08:34:58: Wohnzimmer Temp: 24.6 
> Friday, June 26 2020 08:35:04: Guave Conductivity: 41 
> Friday, June 26 2020 08:35:16: Guave Temp: 23.2 
> Friday, June 26 2020 08:35:16: Wohnzimmer Temp: 24.6 
> Friday, June 26 2020 08:35:21: Wohnzimmer Humid: 57 
> Friday, June 26 2020 08:35:22: Wohnzimmer Temp: 24.5 
> Friday, June 26 2020 08:35:23: Wohnzimmer Humid: 57 
> Friday, June 26 2020 08:35:24: Guave Light: 356 
> Friday, June 26 2020 08:35:29: Wohnzimmer Humid: 57 
> Friday, June 26 2020 08:35:32: Wohnzimmer Temp: 24.6 
> Friday, June 26 2020 08:35:34: Guave Moisture: 41% 
> Friday, June 26 2020 08:35:44: Guave Conductivity: 41 
> Friday, June 26 2020 08:35:44: Wohnzimmer Temp: 24.6 
> Friday, June 26 2020 08:35:54: Guave Temp: 23.2 

while one minute using NimBLE results in:

> Friday, June 26 2020 08:38:54: Guave Moisture: 41% 
> Friday, June 26 2020 08:38:58: Wohnzimmer Temp: 24.5 
> Friday, June 26 2020 08:39:38: Wohnzimmer Humid: 57 
> Friday, June 26 2020 08:39:41: Guave Moisture: 41% 

Hardware is the same and did not move, so "traffic" should be about the same in both cases.

About 80% of the advertisments seem not to be reported to my callback.

Full source of my Program is available here ( https://github.com/TheNitek/MicroHome/blob/master/MicroHome.ino ).

Is that a problem with NimBLE or did I miss any NimBLE specific parameters that I have to set/change?

chegewara, 2020-06-26:
Im not saying its the case here, but some ble sensors increase delay between advertising data for better power usage, so it may be it is more often after it is powered then less often, usually after 30 sec.

TheNitek, 2020-06-26:
The sensors are broadcasting quite regularly. I have been monitoring the for a while now and the frequency of advertisement does not change significantly over time.

h2zero, 2020-06-26:
Hi @TheNitek thank you for reporting this issue.

I think it may be caused by [this line](https://github.com/h2zero/NimBLE-Arduino/blob/e45ee669e94ed7c3d1bf6d6e360a1d1f8d22719c/src/NimBLEScan.cpp#L68), if you could test by setting that value to 0 and report back it would be greatly appreciated.

TheNitek, 2020-06-26:
Looks much better, thx! Funny thing is: If this flag is the reason, it does not work properly, because I am still getting multiple advertisment packages from the same device (and source address)

h2zero, 2020-06-26:
Great, I’ll make the change in the library as well. 

Yes it certainly seems that it doesn’t do what the description says, might be a  timer that resets it or could be bugged.

h2zero, 2020-06-26:
I found the reason that filter_duplicates appears to not work. From ESP IDF Controller manual:
>Note that scan duplicate list will be automatically cleared when the maximum amount of device in the filter is reached the amount of device in the filter can be configured in menuconfig.

Just thought I'd share that for anyone watching.

Also I just pushed the change in commit aaf42f8 .

TheNitek, 2020-06-27:
I don't think this explaination fits. In my logs "Wohnzimmer" und "Guava" are two sensors (delivering multiple data points) and they are still reported multiple times.

Regarding the fix: Wouldn't it make sense to change `setAdvertisedDeviceCallbacks` to take the second parameter again? Currently wantDuplicates is just commented out there

h2zero, 2020-06-28:
> I don't think this explaination fits. In my logs "Wohnzimmer" und "Guava" are two sensors (delivering multiple data points) and they are still reported multiple times.

You're correct, it seems it also resets the filter every ~45 seconds, but I have not seen any documentation to indicate such.

> Regarding the fix: Wouldn't it make sense to change setAdvertisedDeviceCallbacks to take the second parameter again? Currently wantDuplicates is just commented out there

Yes, that is still incomplete, a decision has not been made on the implementation of this as we don't duplicate the devices anymore, simply update the advertised data instead. The original library made a new addition to the vector (map) for the same device if found more than once while scanning with that value set to true, We try to avoid that here and just update the data we have of it. Perhaps, as your code shows, it may be advantageous to save the device and data more than once if that is set to true so the user can get the data from multiple advertisements if required.

The reason for `filter_duplicates = 1` previously is that was a something I changed for testing at one point and accidentally pushed it into the master code.

TheNitek, 2020-06-28:
In my specific case I don't even care about "other" devices/advertisments. I just need the callback to process the data and that's it. I even had to add a clearResults() at the end of my callback to avoid memory issues.
In a perfect world (and for what I am doing right now) the library would neither care about duplicates nor about storing device it has previously found. (I'm not saying it has to be implemented like this, just want to make my use case as clear as possible)

h2zero, 2020-06-29:
@TheNitek I understand your use case, it just highlighted an issue that may come up for others. Such as gathering advertisement data in a specified duration so they may need multiple advertisements from a device if not using the callback.

closed 2020-07-01 by h2zero