h2zero/NimBLE-Arduino #80: Pending v1.0 release and potential sub-moduling.

h2zero, 2020-07-04:
Most of the todo list has been crossed off and a v1.0 release is coming very soon. 

As no news is good news I'm taking the lack of recent issues and PR's as a nod of stability, but I'd like to reach out here to ask if there are any lingering bugs or nags that should be dealt with?

I also would like to ask for feedback regarding using the esp-nimble-cpp repo as a submodule in this repo. Currently I have been keeping the code in both synced together manually but I would prefer to have one repo with the actual C++ library code that would be linked into this for easier maintenance. 

In this proposal I would simply have a cpp folder in this repo that links to the  esp-nimble-cpp repo, however that would mean all pull requests would need to be made there instead of here and would require the use of  `git submodule` operations when cloning/updating. Probably not the best thing to do for an Arduino library,

The second option is a subtree, which makes the `git submodule` commands unnecessary but code modifications and PR's more troublesome (and I would have to learn to use them 😄 )  but would effectively work "as is".

The last option is keep doing what I've been doing. Not my favorite due to potential human error and having to recreate the updates in one repo or the other. 

If anyone has any thoughts about this I would love to hear them.

lknop, 2020-07-14:
I don't know if it fits here but is there a plan to publish this library to the Arduino library manager? I think this would tremendously expand user base and as a result - improve the library thanks to more testing.

h2zero, 2020-07-14:
Yes this is planned. If I’m not mistakenas soon as I slap a release tag on the library manager should pick it up automatically.

lknop, 2020-07-14:
You will also need to open an issue in arduino github: https://github.com/arduino/Arduino/issues pointing to this repo. But yes, that's about it.

h2zero, 2020-07-21:
Looks like arduino library manager doesn’t work with Submodules so I’ll test with subtrees and see if that works.

tobozo, 2020-07-23:
@h2zero a github action or a travis job can pull those submodules, perform some code quality checks and unit/compilation tests then create the release tarballs for you

on push/pull request = perform check + tests

on tag = perform check + tests + create tarball

h2zero, 2020-07-23:
@tobozo that’s a great idea, thanks! I’ll look into that. I’m not very familiar with Travis yet, good opportunity to learn more about it.

tobozo, 2020-07-24:
it's probably the worst implementation ever but here's a [travis example](https://github.com/tobozo/ESP32-Chimera-Core/blob/master/.travis.yml) I'm using to get arduino-ci to compile a few test sketches (see `before_deploy` and `deploy`) and create a tarball, a bit useless since this project doesn't have submodules, but still a valid example since travis automatically performs a submodule update.

if you plan to use a github action instead, the [upload-releas-asset](https://github.com/actions/upload-release-asset) action is a good start

h2zero, 2020-07-26:
Thanks @tobozo, I'll give that a try when I get some time.

everslick, 2020-08-14:
I just came here to express my deepest gratitude for making this library. BLE resource consumption is currently one of the biggest pain points IMHO.

One thing I'd consider an improvement still would be the switch from std::string to String throughout the API. Is this somethimg worth discussing?

lknop, 2020-08-14:
@everslick This would not be an improvement :-) I personally try to never use arduino Strings, ever. Here is an interesting article why that is the case: https://majenko.co.uk/blog/evils-arduino-strings

everslick, 2020-08-14:
@lknop All that is said about `String` in this article, is also (and even more so) true for `std::string`. `String` has some optimizations for MCU platforms and I'd expect some improvements in memory consumption by replacing `std::string` with `String`. Besides it would improve Arduino compatibility, because one would not need to convert between types so often. `String` is great when used wisely - most important: preallocate enough mem with `resize()` before concatenation and pass `String` objects around via reference or `const` reference whenever possible. So avoiding `String` _just because_ will very likely increase your codesize. Look at the implementation of `String` and you will see that it is a very well implemented `class`. Just my 2 cents. ;-)

lknop, 2020-08-14:
@everslick point taken, I just thought that it's in the same ballpark so that would give little advantage. I never got to analysing the String implementation, that's a good idea for a long evening, maybe I'll find time for it this weekend, thanks.

Nevertheless I think I read that @h2zero 's main interest is the esp-idf implementation and the Arduino port is secondary, so I assume it would not be welcome to let those two drift apart, but I may be wrong. I wonder if you are right that it would really be less heap-hungry to use String rather than std::string.

h2zero, 2020-08-14:
Hi @everslick, I agree that using `String` would be better in Arduino. The problem with is it's not usable outside of Arduino. 
This library is a mirror of the esp-idf library and changing this to use `String` here would break that relationship and would need to be maintained separately. That is something I'm not too keen to do. I think it's best to stick to the STL as much as possible.