h2zero/NimBLE-Arduino #81: Why does scanEndedCB get called twice in client connection example?

elguiri, 2020-07-12:
Hi

Firstly I would like to say thanks for the library; I got stuck very quickly using the "built in" library on an esp32 arduino project needing multi client connect, and then discovered this which has given me confidence to proceed.

I have a query relating to the NimBLE_Client.ino example...

When the preferred service is found by scanning, getScan()->stop() is called which triggers the scanEndedCB callback.

That callback is then unexpectedly triggered again after the client is connected (even though the scan has not been restarted at that point).

Is this behaviour by design?

Thanks

h2zero, 2020-07-12:
Hi @elguiri, there were some recent changes in that code and they may have introduced this bug. I will look into this and get it fixed.

Thanks for reporting!

h2zero, 2020-07-12:
I just made PR #82 to address this.

elguiri, 2020-07-12:
Hi @h2zero

Thanks for the quick response!

I've tested with the revised code and the callback is no longer called twice.

Cheers

h2zero, 2020-07-13:
Great, thanks for the feedback!

closed 2020-07-13 by h2zero