h2zero/NimBLE-Arduino #88: NimBLE implementation of Server-Client scheduling

kaiwen98, 2020-07-19:
Hi, I will like to first thank you for your amazing work on the NimBLE libraries for Arduino IDE.
I want to ask if this library supports Server-Client scheduling, such that I can run a NimBLEDevice::init one time and then switch between NimBLEScan and NimBLEAdvertising, or if the same end is met with a different implementation.

Cheers, KW

h2zero, 2020-07-19:
Hi @kaiwen98, you are most welcome. 

This functionality is supported, you should also be able to do both at the same time if you manage your scan and advertise interval and window times. 

Pseudo code example:
```
    NimBLEDevice::init();
    NimBLEScan *pScan = NimBLEDevice::getScan();
    /* set scan parameters */

    NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
    /* set advertising parameters */

    pAdvertising->start();
    pScan->start();
```

I have not tested this, however it should work. If you do, please let us know the results 😄,

kaiwen98, 2020-07-19:
Previously I tried something like this:
setup:
   device::init
   getscan()
   getadv()

task1client:
  startscan
  getdatafromindicate
  stopscan
  mutex give

task2 -> server
  mutex take
  startadv
  .
  .
  .
<rc = 15, host busy>

The esp32 crashes with the error message being Bluetooth host error code 15: host busy. Can you speculate what might have caused the error?

I will try out your pseudocode asap. Thank you!

h2zero, 2020-07-19:
That error just means the controller isn’t finished stopping the scan or advertising. You should wait in a loop until it returns success before starting the next operation.

elguiri, 2020-07-20:
@kaiwen98 

I am experimenting (and having success so far) with having a client and server running at the same time. The idea is that there will be multiple client connections being aggregated into a single server. I'm using the non-blocking scan so the server can do its thing, and have values that update as and when the relevant client connections receive updates. Do you have something similar in mind? The key for you might be to look at using the non-blocking scan.

h2zero, 2020-12-29:
Closing this as stale, feel free to re-open if your issue is not yet resolved.

closed 2020-12-29 by h2zero