h2zero/NimBLE-Arduino #90: Private m_pCharacteristic in NimBLEDescriptor class

elguiri, 2020-07-20:
Hello @h2zero  again

I have a question about the NimBLEDescriptor class.

Is there a particular reason why m_pCharacteristic is private?

In NimBLEDescriptorCallbacks, there is no way to check what characteristic the descriptor is associated with. So you would know that notifications have been enabled for a descriptor (for example), but the server could have many descriptors (because of many characteristics). If m_pCharacteristic had a public property, it would avoid needing to keep a separate lookup of the descriptors that had been added to characteristics during server initialisation.

To look at it another way; I have a server with a handful of 2902 descriptors, so the callback is not that useful at the moment in determining what clients are subscribing to. I just know they are subscribing to something with a 2902 descriptor.

Thanks for your consideration of this!

h2zero, 2020-07-20:
Hi @elguiri I can certainly see your point on this. I'm trying to get a picture of how you are using this in your application. 

From what you described it seems you just want to know which characteristic the client subscribed to? 

Perhaps a better way would be to add an onSubscribe callback to `NimBLECharacteristicCallbacks` and provide the connection ID of the subscribing client and the notifiy/indicate status?

elguiri, 2020-07-20:
@h2zero 

Yes thanks, the onSubscribe callback on the characteristic would be better (less obtuse than going via the 2902), as the client is subscribing to the characteristic not the descriptor.

kaiwen98, 2020-07-20:
@h2zero @elguiri Thank you for the engaging discussion and tips. I will proceed to try them out, and will get back to you all if I have any more difficulties.

elguiri, 2020-07-20:
@h2zero Thanks for the quick fix. I am using platformio and my library referencing is now broken after getting the latest code - NimBLE-Arduino is colliding with framework-arduinoespressif32. I'll have to unravel that before I can do some testing.

Cheers

elguiri, 2020-07-21:
@h2zero Sorry for the delay, I sorted out my sprawling code and all is good and tested. The PR changes are perfect, thanks very much! I can now see what a connecting client is subscribing to on the server.

h2zero, 2020-07-21:
Great, thanks! If you get a chance to try the latest changes please let me know your thoughts.

elguiri, 2020-07-22:
For anybody following this, I've posted the results of testing on https://github.com/h2zero/NimBLE-Arduino/pull/91

closed 2020-08-30 by elguiri