h2zero/NimBLE-Arduino #92: More examples?

h2zero, 2020-07-20:
I’m thinking this library could use a couple more examples. Does anyone have suggestions on what would be informative? 

I have some of my own ideas but I’d like to hear from users as to what they would like to see.

Also if anyone would like to submit a PR for any examples (or anything else) please feel free to do so.

alvarowolfx, 2020-07-21:
I'm creating a demo on how to provision a device (ESP32) to Google IoT Core over BLE. Basically I'm starting a BLE Uart interface when the user clicks a button and them the device can accept things like Wifi creds, IoT Core settings and private key.

https://github.com/alvarowolfx/gcloud-iot-core-device-provisioning

I can try to submit an example with just the Wifi provisioning part, I think that would be really valuable for users of this library.

BTW, awesome work that you're doing, saw major memory improvements here and is working flawlessly.

h2zero, 2020-07-21:
@alvarowolfx That's a nice project! 

> I can try to submit an example with just the Wifi provisioning part, I think that would be really valuable for users of this library.

That would be a great addition, if you have time to do it I would certainly include it.

txf-, 2020-10-06:
> I’m thinking this library could use a couple more examples. Does anyone have suggestions on what would be informative?

A bonding example would be useful. I'm currently having issues implementing automatic bonding on connection, such as what happens with the original ESP32 stack.

h2zero, 2020-10-07:
Hi @txf-, good suggestion. I am going to be improving that aspect of the library shortly to add more functionality and will create an example to show the use of it. 

For your purposes right now all you should need to do is call:
```
NimBLEDevice::setSecurityAuth(true, false, true);
```
The first parameter set to `true` enables bonding.

After that to replicate the original library functionality (for a server) you would need to call:
```
NimBLEDevice::startSecurity(uint16_t conn_id); 
```
From the `NimBLEServerCallbacks::onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc)` callback.
The `conn_id` is available from `desc-> conn_handle` 

For a client app you just need to call `NimBLEClient::secureConnection()` after connecting.

Sorry this is not described well, still a work in progress.

hannahvsawiuk, 2020-11-02:
Hi @h2zero, I am also trying to implement automatic bonding upon connection. The security settings were configured as you mentioned and custom server callbacks class was defined as such

```cpp
class MyServerCallbacks: public NimBLEServerCallbacks {
    void onConnect(NimBLEServer* pServer) {
      ESP_LOGD(LOG_TAG, "Device connected. No connection descriptor found.");
      deviceConnected = true;
    };
    void onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc) {
      deviceConnected = true;
      ESP_LOGD(LOG_TAG, 
      "Device connected. Client address: %s", NimBLEAddress(desc->peer_ota_addr).toString().c_str()
      );
      // Update the connection parameters
      pServer->updateConnParams(desc->conn_handle, 24, 48, 0, 60);
      // Start secure connection
      NimBLEDevice::startSecurity(desc->conn_handle); 
    };
```

My confusion is with the `NimBLEDevice::startSecurity` method being private. I am unsure of where to call the `startSecurity` method as it is not allowed within the callbacks class.

h2zero, 2020-11-02:
Sorry about that, your code looks fine. `StartSecurity` should not be private, my mistake.

hannahvsawiuk, 2020-11-02:
Thank you for the rapid response!

hannahvsawiuk, 2020-11-03:
Update: the automatic bonding works! I moved the `startSecurity` method to be defined as a public method instead and called it as I did in the server callbacks class I posted above.

h2zero, 2020-11-03:
Great, if you want to PR that change I will merge it. Otherwise I will do it next time I get a chance.

hannahvsawiuk, 2020-11-03:
I will do that now

h2zero, 2020-12-29:
Closing this, I will create more examples when planned updates are finished.

closed 2020-12-29 by h2zero