h2zero/NimBLE-Arduino #93: only BLE server liberary

darshan-mirajkar, 2020-07-21:
For some application need only BLE server library. Can free more heap. Please guide me to remove other functionalities apart from the basic ble server

h2zero, 2020-07-21:
Hi @darshan-mirajkar do you need server capability or just advertising?

You can configure the settings for minimal resources in the [nimconfig.h file](https://github.com/h2zero/NimBLE-Arduino/blob/e0e940ab3e8d7fdd52679a3a54074594a3078ef5/src/nimconfig.h#L39)

darshan-mirajkar, 2020-07-21:
I tried the same thank you for reply. By removing any functionality can we reduse more heap??

h2zero, 2020-07-21:
Yes, if you comment out:
```
#define CONFIG_BT_NIMBLE_ROLE_CENTRAL
#define CONFIG_BT_NIMBLE_ROLE_OBSERVER
```
It will remove client and scanning functionality, saving flash space and a bit of heap.

Other than disabling those roles you can free up some heap space by reducing these values in nimconfig:
CONFIG_BT_NIMBLE_TASK_STACK_SIZE
CONFIG_BT_NIMBLE_MAX_CONNECTION

Some of the others values can be adjusted as well but I do not recommend it as they may create other issues.
Feel free to experiment though.

darshan-mirajkar, 2020-07-21:
from this some 5-6 kb of heap become free. Thank you for replay.

closed 2020-07-27 by darshan-mirajkar