h2zero/NimBLE-Arduino #95: Unable to add binary data with '\0' in the advertisement. [patch code inside]

straccio, 2020-07-22:
Good morning,
i had to add an overload to the function addData of NinBLEAdvertising class in order to bypass the problem caused by the using of std::string as raw binary buffer.

Here the overload:
```
void NimBLEAdvertisementData::addData(char * data, size_t length){
    if ((m_payload.length() + length) > BLE_HS_ADV_MAX_SZ) {
        return;
    }
    m_payload.append(data,length);
    printf("Payload length:%d\n",m_payload.length());
}
```

You can test this case with the eddystone beacon example by changing the url with one that start with "http://www.", because this match eddystone_url_prefix_subs at 0 index.

Best regards and good job

h2zero, 2020-07-22:
Thanks! I’ll add this shortly.

straccio, 2020-07-22:
Thanks to you! :P

h2zero, 2020-07-22:
Just a quick note on this, you can create a std::string beginning with \0, it just needs to be done by specifying the length.
```
std::string dataStr((char*)data, length);
pAdvData->addData(dataStr);
```

straccio, 2020-07-22:
I have tried but it doesn't work.
Before calling addData the string size is correctly; inside the adData the size of the string is truncated at \0 position.

I think that happen because passing the string to the function call the copy of the object ad that cause the issue.

The correct way must be use the variable as std::basic_string<unsigned char/uint8_t>

h2zero, 2020-07-22:
That's odd, I can't reproduce that here. If I use the url beacon example with a http://www. address and change the eddyStoneData construction to `std::string eddyStoneData(ret_data, ret_data[7]+8);` it works just fine. 

I'll still add this anyway so a std::string doesn't need to be used.

h2zero, 2020-07-28:
Sorry for the delay, I pushed this patch just now in commit 3976074.

h2zero, 2020-12-29:
resolved

closed 2020-12-29 by h2zero