h2zero/NimBLE-Arduino #96: Scan with duration 0 in background

straccio, 2020-07-22:
I have modified the function start of NimBLEScan in order to return immediately if is called with duration = 0.
That can permit to leave the scan in background and release the task to do something else.

```
/**
 * @brief Start scanning and block until scanning has been completed.
 * @param [in] duration The duration in seconds for which to scan.
 * @param [in] is_continue Set to true to save previous scan results, false to clear them.
 * @return The NimBLEScanResults.
 */
NimBLEScanResults NimBLEScan::start(uint32_t duration, bool is_continue) {
    if(duration == 0) {
        NIMBLE_LOGW(LOG_TAG, "Blocking scan called with duration = forever");
    }

    ble_task_data_t taskData = {nullptr, xTaskGetCurrentTaskHandle(),0, nullptr};
    m_pTaskData = &taskData;

    if(start(duration, nullptr, is_continue)) {
        if(duration==0) return m_scanResults;
        ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
    }
    m_pTaskData = nullptr;

    return m_scanResults;
} // start
```

I'm not an expert, than i don't know if m_pTaskData must be cleared before the return.
This patch is intended as a draft

h2zero, 2020-07-22:
That method is the blocking scan, meant to be used with a scan time (that's why the warning is there).

If you want to scan continuously you should use the overloaded start method:
`pScan->start(0, nullptr, false);`

That is what the code you posted does.

straccio, 2020-07-22:
Right, my fault, tnx!

h2zero, 2020-12-29:
Closing this as it's resolved.

closed 2020-12-29 by h2zero