h2zero/NimBLE-Arduino #97: Scan state to be available

elguiri, 2020-07-22:
@h2zero I've just seen the other question on non-blocking scan so I thought I would ask mine! Is it possible to add a property to access the state of the scan when used in non-blocking fashion? At the moment, I am setting my own flag at the start of the scan, and when scan complete is called, but that's a bit clunky when I really want the internal state (and it looks like m_stopped or the inverse might do). Thanks!

h2zero, 2020-07-22:
I could add a method to NimBLEScan (::isActive()?) for this, you can also use  `ble_gap_disc_active()` to ask NimBLE for the state.

elguiri, 2020-07-22:
Thanks, I'll take a look at ble_gap_disc_active

elguiri, 2020-07-22:
@h2zero 

ble_gap_disc_active() does what I need. I think `bool isActive()` would be a good idea as this would be consistent with the programming standard that the state of background tasks is accessible (and would avoid me going "down a layer" into NimBLE).

h2zero, 2020-07-23:
Agreed, I will add it.

h2zero, 2020-07-24:
Decided to change it to isScanning() since isActive() would be ambiguous due to setActiveScan(). Semantics.. but clarity is important.
Just pushed it in commit bad29ee.

elguiri, 2020-07-24:
@h2zero Agreed, using isActive would cause confusion. As you have committed the change and I have tested it, I've closed this. Thanks once again!

closed 2020-07-24 by elguiri