h2zero/NimBLE-Arduino #99: [Question] Connect to more of 3 peripheral device

GianCann, 2020-07-26:
Good morning,
I need to connect (from a ESP32 BLE with a client role) to 6-8 peripheral device subscribing notification from these device.

It's possible to do with this library?

h2zero, 2020-07-26:
Hi @GianCann, yes up to 9 connections are supported (esp controller limit). You will need to change the setting in nimconfig.h on [this line](https://github.com/h2zero/NimBLE-Arduino/blob/bad29eecb055abf2ec4e6d328d03fd0e40901594/src/nimconfig.h#L89).

h2zero, 2020-07-30:
As an update to my previous answer; in Arduino this is not possible without rebuilding the libbtdm_app.a with the bluetooth controller max connections setting to more than 3 using idf and replacing it in your Arduino core. I completely forgot about this.

corneliusmunz, 2020-07-30:
@h2zero thanks for that hint. Up to now (without rebuilding the libbtdm_app.a ) the connection to more than 3 peripheral devices was not possible. Do you have a hint how i could rebuild the libbtdm_app.a library?

h2zero, 2020-07-30:
@corneliusmunz Actually I mentioned the wrong lib file, that is the controller binary file, I can put a rebuilt libbt.a file here for you to try, that's the only other one I can think of that may affect this. 

This is for esp32-arduino version 1.0.4
[libbt.a.txt](https://github.com/h2zero/NimBLE-Arduino/files/5004011/libbt.a.txt)

h2zero, 2020-07-30:
Could you post a log of the error you get when trying to connect to more than 3 devices?

h2zero, 2020-07-31:
@corneliusmunz @GianCann  I was wrong, no files need to be replaced, sorry for the bad info. I was playing with controller versions to test some things and I thought that was what made it work for me, it was not. 

The solution is actually a lot simpler. Just change the defines in the sdkconfig.h file at Arduino/hardware/espressif/esp32/tools/sdk/include/config to:
```
#define CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN 9
#define CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN_EFF 9
```
That will allow the controller the max connections it can handle and you can set the NimBLE max connections to what you need.

corneliusmunz, 2020-07-31:
@h2zero First of all many thanks for your support! I have tested it with the changed settings in sdkconfig.h. and in the nimconfig.h file. Unfortunately it does not connect to the 4th device. The devices are Lego Hubs which are connected via the legoino arduino library: https://github.com/corneliusmunz/legoino
 
The following error message occurs which tells me, that the max connections in sdkconfig.h. and nimconfig.h are not really taken into account:
`22:44:59.737 -> E NimBLEClient: "Error: Failed to connect to device; addr_type=0 " "addr=00:81:f9:43:f2:06, rc=521; Connection Limit Exceeded"`
[Here you can find the complete log with nimble logging enabled and esp logging set to verbose](https://github.com/h2zero/NimBLE-Arduino/files/5009005/nimble_error_log_4_connections.log)

The follwing settings were used:
[nimconfig.txt](https://github.com/h2zero/NimBLE-Arduino/files/5009008/nimconfig.txt)
[sdkconfig.txt](https://github.com/h2zero/NimBLE-Arduino/files/5009009/sdkconfig.txt)

For the connection procedure i used this sketch from my Lego Bluetooth library:
https://github.com/corneliusmunz/legoino/tree/feature/check_NimBle_Library/examples/MultipleTrainHubs

h2zero, 2020-07-31:
That’s odd, I connected to 5 devices today just with the config changes. I will try again with my other pc as I’m not quite sure about the status of the binary I used.

GianCann, 2020-08-01:
Thank you @h2zero : I look forward to a definitive clarification on the use of the ESP32 module with more than 3 BLE devices

h2zero, 2020-08-02:
I was just able to do some more testing, with the above changes to the config files I am able to connect to 4 devices with no issue currently. This is using arduino-esp32 1.0.4 and arduino IDE. I am wondering if you are using platformio, this may be why the config changes do not have effect?

corneliusmunz, 2020-08-03:
@h2zero 👍 👏 💪 🤘 ❤️ I switch to another PC (former try was on a MAC) and have done all your proposed changes (so only 3 changed settings in 2 header files) and fortunately it works like a charm!! I think on my OSX environment the formerly build object-files are used without applying the changes in the max connection values. I will try to evaluate a littlebit deeper why it have not worked out on my OSX system. 

So again: Thank you very much for you help and your superb work on that library. I will have a look on your proceeding and will definitely think about porting my Lego library (https://github.com/corneliusmunz/legoino) from the standard ble library to yours. Do you plan to "officially" release the libray for the library manager? 

@GianCann Can you try it with your 8 Train Hub setup? So for me it worked out and i can connect to 4 hubs at the same time

corneliusmunz, 2020-08-03:
@GianCann I could connect at the same time to 2 train hubs, 2 train remote controls, 1 boost hub and 1 control+ hub 😀[image]

h2zero, 2020-08-04:
@corneliusmunz 💯 I'm glad you found the issue and all is working as expected, I have not connected to more than 6 devices before so if you have any issues connecting to this many them please make an issue about it. NimBLE actually supports up to 32 connections, sadly the esp32 controller is not equipped for this, but I'd be curious to see what happens if attempted.

> Do you plan to "officially" release the libray for the library manager?

This is in the works, a few minor things to deal with like mirroring the code between Arduino and IDF repos and some documentation/minor cleanup, other than that the current code is v1.0.0 I believe and ready to release. I just haven't had time to finish up these things.

GianCann, 2020-08-04:
@corneliusmunz I have seen your test. Today I try to replicate on Windows PC.

corneliusmunz, 2020-08-04:
@h2zero, @GianCann I have tested it today on my MAC with the following procedure and it worked like a charm! I think i have changed maybe the wrong header files or the Arduino IDE does not recompile the hardware libs after the header was changed.

- Delete Arduino out of my Application folder
- Delete the `/Users/<username>/Documents/Arduino` folder
- Delete the `/Users/<username>/Library/Arduino15` folder
- Install Arduino again via download from https://www.arduino.cc
- Download mimble library zip file and extract it to `/Users/<username>/Documents/Arduino/libraries`
- Install esp32 Boards via the Boardmanager
- Changed the setting in the `/Users/<username>/Documents/Arduino/libraries/NimBLE-Arduino/src/nimconfig.h` file to `#define CONFIG_BT_NIMBLE_MAX_CONNECTIONS 8 `
- Changed the setting in the `/Users/<username>/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config/sdkconfig.h` file to `#define CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN_EFF 8` and `#define CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN 8`
- Install the Lego Library in the `/Users/<username>/Documents/Arduino/libraries` directory via downloading the Files of the specific nimble feature branch https://github.com/corneliusmunz/legoino/tree/feature/check_NimBle_Library
- Open the following sketch in the Arduino IDE and load it to the esp32 board https://github.com/corneliusmunz/legoino/blob/feature/check_NimBle_Library/examples/MultipleTrainHubs/MultipleTrainHubs.ino
- Connect the hubs (BLE peripheral devices)
- Be happy 😄

h2zero, 2020-08-05:
@corneliusmunz Thank you for providing the details of your solution. I will include this in the library documentation. 

How many simultaneous connections have you been able to maintain?

GianCann, 2020-08-05:
@corneliusmunz I following you instruction but I get this error:

C:\Users\remoteadmin\Documents\Arduino\libraries\legoino-feature-check_NimBle_Library\src\Lpf2Hub.cpp: In member function 'bool Lpf2Hub::connectHub()':

C:\Users\remoteadmin\Documents\Arduino\libraries\legoino-feature-check_NimBle_Library\src\Lpf2Hub.cpp:885:66: error: no matching function for call to 'NimBLEClient::connect(NimBLEAddress&, int, bool)'

             if(!pClient->connect(pAddress, BLE_ADDR_PUBLIC, false)) {

                                                                  ^
In file included from C:\Users\remoteadmin\Documents\Arduino\libraries\NimBLE-Arduino-master\src/NimBLEDevice.h:31:0,

                 from C:\Users\remoteadmin\Documents\Arduino\libraries\legoino-feature-check_NimBle_Library\src\Lpf2Hub.h:13,

                 from C:\Users\remoteadmin\Documents\Arduino\libraries\legoino-feature-check_NimBle_Library\src\Lpf2Hub.cpp:9:

C:\Users\remoteadmin\Documents\Arduino\libraries\NimBLE-Arduino-master\src/NimBLEClient.h:41:49: note: candidate: bool NimBLEClient::connect(NimBLEAdvertisedDevice*, bool)

     bool                                        connect(NimBLEAdvertisedDevice* device, bool deleteAttibutes = true);

                                                 ^

C:\Users\remoteadmin\Documents\Arduino\libraries\NimBLE-Arduino-master\src/NimBLEClient.h:41:49: note:   candidate expects 2 arguments, 3 provided

C:\Users\remoteadmin\Documents\Arduino\libraries\NimBLE-Arduino-master\src/NimBLEClient.h:42:49: note: candidate: bool NimBLEClient::connect(const NimBLEAddress&, bool)

     bool                                        connect(const NimBLEAddress &address, bool deleteAttibutes = true);

                                                 ^

C:\Users\remoteadmin\Documents\Arduino\libraries\NimBLE-Arduino-master\src/NimBLEClient.h:42:49: note:   candidate expects 2 arguments, 3 provided

C:\Users\remoteadmin\Documents\Arduino\libraries\NimBLE-Arduino-master\src/NimBLEClient.h:43:49: note: candidate: bool NimBLEClient::connect(bool)

     bool                                        connect(bool deleteAttibutes = true);

                                                 ^

C:\Users\remoteadmin\Documents\Arduino\libraries\NimBLE-Arduino-master\src/NimBLEClient.h:43:49: note:   candidate expects 1 argument, 3 provided

exit status 1
Errore durante la compilazione per la scheda M5Stack-Core-ESP32.

GianCann, 2020-08-05:
Ok, I have changed this part of code in C:\Users\remoteadmin\Documents\Arduino\libraries\legoino-feature-check_NimBle_Library\src\Lpf2Hub.cpp

        /** Check if we have a client we should reuse first **/
    if(NimBLEDevice::getClientListSize()) {
        /** Special case when we already know this device, we send false as the 
         *  second argument in connect() to prevent refreshing the service database.
         *  This saves considerable time and power.
         */
        pClient = NimBLEDevice::getClientByPeerAddress(pAddress);
        if(pClient){
			/* if(!pClient->connect(pAddress, BLE_ADDR_PUBLIC, false)) { */
            if(!pClient->connect(pAddress, false)) {
                Serial.println("Reconnect failed");
                return false;
            }
            Serial.println("Reconnected client");
        }

GianCann, 2020-08-05:
Yeah!

09:51:41.275 -> Number of Clients: 0
09:51:41.275 -> New client created
09:51:41.377 -> Connected to: 90:84:2b:0c:06:41
09:51:41.377 -> RSSI: -63
09:51:42.022 -> Connected to HUB1
09:51:44.803 -> Number of Clients: 1
09:51:44.803 -> New client created
09:51:44.905 -> Connected to: 90:84:2b:0c:09:16
09:51:44.939 -> RSSI: -71
09:51:45.516 -> Connected to HUB2
09:51:48.502 -> Number of Clients: 2
09:51:48.502 -> New client created
09:51:48.638 -> Connected to: a4:34:f1:9b:07:9e
09:51:48.638 -> RSSI: -60
09:51:49.112 -> Connected to Remote1
09:51:53.653 -> Number of Clients: 3
09:51:53.653 -> New client created
09:51:53.789 -> Connected to: e0:7d:ea:0d:6d:94
09:51:53.789 -> RSSI: -73
09:51:54.262 -> Connected to Remote2

Now... go with others hub!

corneliusmunz, 2020-08-05:
> Yeah!
>
> 09:51:41.275 -> Number of Clients: 0
> 09:51:41.275 -> New client created
> 09:51:41.377 -> Connected to: 90:84:2b:0c:06:41
> 09:51:41.377 -> RSSI: -63
> 09:51:42.022 -> Connected to HUB1 [...]

Cool! What do you have changed after the error?

corneliusmunz, 2020-08-05:
> Yeah!
>
> 09:51:41.275 -> Number of Clients: 0
> 09:51:41.275 -> New client created
> 09:51:41.377 -> Connected to: 90:84:2b:0c:06:41
> 09:51:41.377 -> RSSI: -63
> 09:51:42.022 -> Connected to HUB1 [...]

Cool! I am really looking forward to your test with more that 4 hubs :-) I only have 6 hubs so that was my maximum

corneliusmunz, 2020-08-05:
@GianCann I see now why this was failing on your side. The NimBLE lib has changed the interface yesterday and i was downloading it before the change. I will update the code on my feature branch according to your changes. Thanks for evaluating this and fixing the error

GianCann, 2020-08-05:
I successful get 8 device connected to my ESP32 
Now the fun begins :smile:

corneliusmunz, 2020-08-05:
> I successful get 8 device connected to my ESP32
> Now the fun begins 😄

Very cool! Thanks for your patience and your super hint to use the nimBLE Lib! I have pushed the code changes with your fix on the feature branch. Can you check it with 10 Devices? maybe it works :-) As far as i remember you have had 10 TrainHubs in your test setup picture

GianCann, 2020-08-05:
> As far as i remember you have had 10 TrainHubs in your test setup picture

I have 12 Train Hub, 6 Control+ Hub, 2 Move Hub and 4 Remote control :smile:

GianCann, 2020-08-05:
A simple example:
https://www.youtube.com/watch?v=vOMIabLJhak

GianCann, 2020-08-05:
@h2zero I suggest you add a note on the description page of the library, useful for those who need to connect to more than 3 devices

h2zero, 2020-08-05:
> @h2zero I suggest you add a note on the description page of the library, useful for those who need to connect to more than 3 devices

I will do that. Thanks for confirming the required changes.

I'm curious also to know what happens if you connect to more than 9. If you try it please post your results.

GianCann, 2020-08-05:
> I'm curious also to know what happens if you connect to more than 9. If you try it please post your results

Yesz I will try later

GianCann, 2020-08-06:
I can confirm: max 9 device!

[image]

```
11:34:40.116 -> Wait for connection
11:35:16.708 -> Number of Clients: 0
11:35:16.708 -> New client created
11:35:16.843 -> Connected to: 90:84:2b:49:63:64
11:35:16.843 -> RSSI: -41
11:35:17.450 -> Connected to HUB1
11:35:18.835 -> Number of Clients: 1
11:35:18.835 -> New client created
11:35:18.937 -> Connected to: 90:84:2b:49:63:9d
11:35:18.937 -> RSSI: -30
11:35:19.546 -> Connected to HUB2
11:35:23.603 -> Number of Clients: 2
11:35:23.603 -> New client created
11:35:23.739 -> Connected to: 90:84:2b:4a:b0:aa
11:35:23.739 -> RSSI: -22
11:35:24.347 -> Connected to HUB3
11:35:27.055 -> Number of Clients: 3
11:35:27.055 -> New client created
11:35:27.123 -> Connected to: 90:84:2b:4b:17:a2
11:35:27.123 -> RSSI: -37
11:35:27.765 -> Connected to HUB4
11:35:29.452 -> Number of Clients: 4
11:35:29.452 -> New client created
11:35:29.588 -> Connected to: 90:84:2b:4c:7c:f9
11:35:29.588 -> RSSI: -22
11:35:30.195 -> Connected to HUB5
11:35:31.517 -> Number of Clients: 5
11:35:31.517 -> New client created
11:35:31.653 -> Connected to: 90:84:2b:4c:88:23
11:35:31.653 -> RSSI: -25
11:35:32.262 -> Connected to HUB6
11:35:33.618 -> Number of Clients: 6
11:35:33.618 -> New client created
11:35:33.753 -> Connected to: 90:84:2b:0c:0c:12
11:35:33.753 -> RSSI: -45
11:35:34.396 -> Connected to HUB7
11:35:35.347 -> Number of Clients: 7
11:35:35.347 -> New client created
11:35:35.449 -> Connected to: 90:84:2b:0d:a7:ea
11:35:35.449 -> RSSI: -37
11:35:36.059 -> Connected to HUB8
11:35:37.075 -> Number of Clients: 8
11:35:37.075 -> New client created
11:35:37.210 -> Connected to: 90:84:2b:0c:06:41
11:35:37.210 -> RSSI: -43
11:35:37.820 -> Connected to HUB9
```

If I try to modify these line with 10 value:
#define CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN 10
#define CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN_EFF 10
(and also the nimconfig.h file)
I get ESP32 crashing in loop at boot time...

So, maximum 9 BLE device.
Now, the next step is to use the WiFi network simultaneously

(I want to make a MQTT/BLE Gateway)

h2zero, 2020-08-06:
Thanks for testing that, interesting that it crashes on boot with those values set to 10. 

WiFi + 9 BLE connections might need some connection interval management to maintain.

GianCann, 2020-08-06:
> WiFi + 9 BLE connections might need some connection interval management to maintain

Can you explain better?

h2zero, 2020-08-06:
I’m not sure if you will need to do anything until you try it but I will explain.

The BLE connection interval sets how often the devices should send a packet to each other. There is also connection supervision timeout that will disconnect if a packet is not received in that time. 

Depending on the device you’re connected to and the parameters set you may get disconnected from one or more devices frequently. Also if the connection interval is short and you’re also using WiFi you may get disconnected from WiFi often as the BLE connections may not give it enough time to use the antenna.

I would say don’t worry about it until you try it. If you have issues we can make the adjustments then.

GianCann, 2020-08-06:
Ok, it's all clear.
I will do real tests and see what happens.
In the worst case, I will use an ESP8266 as a WiFi / MQTT gateway connected via serial to the ESP32

h2zero, 2020-08-06:
> In the worst case, I will use an ESP8266 as a WiFi / MQTT gateway connected via serial to the ESP32

That shouldn't need to be done at all. The connection parameters are easy to change, just a matter of testing with different values if you need to. Connection parameter documentation is [here] (https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_client.html#a17718339f76eb621db0d7919c73b9267) and [here](https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e_client.html#aff7d389ec48567286ea732c54d320526).

h2zero, 2020-09-06:
As an update to this I have created a patch that should allow maximum connection change without the need to modify `sdkconfig`, only need to change the value in `nimconfig` now. I need someone to help test it though as I do not have enough BT devices to test more than 3 currently.

@GianCann if you're interested, perhaps you could test this patch with your setup, just apply the patch and revert the changes you made in sdkconfig.

[max_conn_patch.txt](https://github.com/h2zero/NimBLE-Arduino/files/5179743/max_conn_patch.txt)

corneliusmunz, 2020-09-10:
@h2zero I have applied your patch and used the original settings in the `sdkconfig.h` (MAX_CONN = 3) With your patch i could connect to more than 3 devices with only change the settings in the `nimconfig.h` So in my point of view, your patch works and i think it could be merged into the master branch. Very cool, that you only have to change one setting in one header file!

h2zero, 2020-09-10:
@corneliusmunz Thank you for testing and I'm glad it works as intended. Will will commit this to master shortly.

h2zero, 2020-12-29:
Closing this as resolved.

closed 2020-12-29 by h2zero

goddade, 2025-01-18:
@GianCann Hello, how's your gateway project coming along?