h2zero/NimBLE-Arduino #105: PSRAM for BLE stack

darshan-mirajkar, 2020-07-30:
Can we use PSRAM for BLE stack?? If yes what are the changes required?

h2zero, 2020-07-30:
I think this is possible, I have not tested. 

You would need to change [this line](https://github.com/h2zero/NimBLE-Arduino/blob/05c68ed8926d1144191ce469001b6a3f3d13f715/src/nimconfig.h#L122) in nimconfig.h to `CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL`.

Let me know if that works.

darshan-mirajkar, 2020-07-31:
Sure... I am looking for module with PSRAM...I'll let you know

ntwallace, 2020-08-13:
@h2zero changing that line results in:

```lib/NimBLE/src/port/src/esp_nimble_mem.c: In function 'nimble_platform_mem_malloc':
lib/NimBLE/src/port/src/esp_nimble_mem.c:31:47: error: #elif with no expression
lib/NimBLE/src/port/src/esp_nimble_mem.c: In function 'nimble_platform_mem_calloc':
lib/NimBLE/src/port/src/esp_nimble_mem.c:44:47: error: #elif with no expression
*** [.pio/build/esp-wrover-kit/lib839/NimBLE/port/src/esp_nimble_mem.c.o] Error 1
```

It looks like in `esp_nimble_mem.c` all the `#elif`s are missing a `defined`. Adding that fixes the problem and the stack loads in PSRAM, e.g.:
```
#ifdef CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_INTERNAL
    return heap_caps_malloc(size, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
#elif defined CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL
    return heap_caps_malloc(size, MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT);
#elif defined CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_IRAM_8BIT
    return heap_caps_malloc_prefer(size, 2, MALLOC_CAP_INTERNAL|MALLOC_CAP_IRAM_8BIT, MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
```

h2zero, 2020-08-14:
@ntwallace Thanks for reporting this. Please consider raising the issue in esp-idf as that file is tracked from there. In the meantime I will consider fixing it here for this library's purposes.

h2zero, 2020-08-15:
After looking at this again I think the fix is actually to define a value of 1 to the config define, such as:

`#define CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL 1`

I will add this line, commented, in nimconfig.

ntwallace, 2020-08-17:
Ah good catch, makes sense. Thanks.

darshan-mirajkar, 2020-08-17:
Nice one guys... I am also looking for hardware with PSRAM to test this.

ntwallace, 2020-08-17:
@darshan-mirajkar [Mouser](https://www.mouser.com/ProductDetail/Espressif-Systems/ESP32-DevKitC-VB?qs=w%2Fv1CP2dgqq1iFM3SpreoA==) has the official Dev Kit C for $11 USD, looks like they have stock in most countries.

darshan-mirajkar, 2020-08-17:
Yes I know... I our application hardware we designed PSRAM with the help of expressif. Gerber is ready... Waiting for PCB. So we. Continue on that.

h2zero, 2020-12-29:
Closing this as resolved.

closed 2020-12-29 by h2zero