h2zero/NimBLE-Arduino #107: BLE HEAP memory reduction

sanketa-54, 2020-07-31:
Hello ,
Please anyone can help with HEAP reduction due to NimBLE.h file in the code.
BLEDevice::deinit - It only clears BLE stack or server or service ?
How to clear out whole memory of NimBLE Stack

h2zero, 2020-07-31:
NimBLEDevice::deinit(true) will delete all BLE objects and free the memory.

sanketa-54, 2020-07-31:
NimBLEDevice::deinit(true) --- This is giving me error
NimBLEDevice::deinit() -- This is working fine but its not deleting HEAP of NimBLE.h Library Memory.

sanketa-54, 2020-07-31:
I want the complete NimBLE functionality, including server, to be removed.
How can it be done ?
deinit() ---> Its the only way to do that .

h2zero, 2020-07-31:
deinit(true) is the best way to clear it all. I suspect you are using an older commit before that was added. Please update your library to the latest commit and try again.

sanketa-54, 2020-07-31:
Ok...I'll try and update u
Thanks

h2zero, 2020-07-31:
> But Can we clear NimBLE library memory

The deinit function will clear the library memory. Do you also want to clear the NimBLE stack memory? That can be done with the esp_bt_controller_mem_release() function, if you do this though you won't be able to re-init and use BLE again until reboot.

sanketa-54, 2020-07-31:
Yess....NimBLE stack memory also 
Got it !!!
Thanks again
I'll try and update

sanketa-54, 2020-07-31:
One more thing
ESP32 device, the NimBLE consumes around 80 K of RAM. Is this OK? I feel this is large amount of RAM, although it is less than Bluedriod.

h2zero, 2020-07-31:
> NimBLE consumes around 80 K of RAM 

That seems a little high for NimBLE, but this depends on what you're doing with it, how many attributes/data/connections etc.
For example, in some simple test applications I have 306kB of free heap before initializing, and 277kB after connection, so 39kB consumed.

sanketa-54, 2020-08-03:
When I'm using BLE_UART service and the difference between the memory before and after initialisation is approx 16kB.
But When I'm using BLE and MQTT the difference is approx 110KB.
So, this diff is because of BLE used with MQTT or only MQTT

m-menon, 2020-08-03:
> NimBLEDevice::deinit(true) --- This is giving me error
> NimBLEDevice::deinit() -- This is working fine but its not deleting HEAP of NimBLE.h Library Memory.

@sanketa-54 I thought I'd leave a note for those who face the above problem. I faced a similar problem and then realized that I was using a slightly older version of the library and in that version `deinit` doesn't include the `clearAll` parameter. `::deinit(true)` should just work fine with the latest version of NimBLE-Arduino.

sanketa-54, 2020-08-03:
> @sanketa-54 I thought I'd leave a note for those who face the above problem. I faced a similar problem and then realized that I was using a slightly older version of the library and in that version [...]

@m-menon : Yes...Everyone should know latest commit.

h2zero, 2020-08-03:
> When I'm using BLE_UART service and the difference between the memory before and after initialisation is approx 16kB.
But When I'm using BLE and MQTT the difference is approx 110KB.
So, this diff is because of BLE used with MQTT or only MQTT

@sanketa-54 Yes NimBLE is pretty light on it's own. When WiFi is used as well it causes the BT+Wifi coexist drivers to load + the WiFi driver and consumes quite a bit of heap. Sadly, there isn't much we can do about that here.

I've thought about creating a light weight WiFi library for basic connections / MQTT, that's only a thought for now.

sanketa-54, 2020-08-28:
@h2zero :
It wolud be great if there's lighter WIfi library for basic functions.
One more thing If we use pSRAM for ESP32 then can we reduce NimBLE stack , library and functionalities heap ?

h2zero, 2020-08-28:
@sanketa-54 
> It wolud be great if there's lighter WIfi library for basic functions.

I'm not sure how much could be saved by doing this but it would be an interesting experiment. I won't be attempting it for a while though.

> One more thing If we use pSRAM for ESP32 then can we reduce NimBLE stack , library and functionalities heap ?

NimBLE can be loaded into PSRAM, which will reduce the main heap use. Also if you need further reduction you can disable functionality you don't need (client / server etc..). All documented in nimconfig.h and [here](https://h2zero.github.io/esp-nimble-cpp/nimconfig_8h.html)

Sanketa54, 2020-08-31:
@h2zero : Yes Thanku

Sanketa54, 2020-08-31:
@h2zero : Is there a way to see detailed layout of memory in ESP32 - arduino ?
The below stuff we can see in idf.
Total sizes:
DRAM .data size: 19104 bytes
DRAM .bss size: 49264 bytes
Used static DRAM: 68368 bytes ( 56212 available, 54.9% used)
Used static IRAM: 118037 bytes ( 13035 available, 90.1% used)
Flash code: 1079425 bytes
Flash rodata: 274040 bytes
Total image size:~1490606 bytes (.bin may be padded larger)

Please help me this...This will help me to add pSRAM efficiently

h2zero, 2020-08-31:
I don't know if there is any way for arduino to produce that data. 

Only way I know is to use IDF and arduino as a component.

sanketa-54, 2020-12-03:
@h2zero : I'm using psram with WROVER standalone module ( With Wifi , MQTT , Nimble ).
I have added this line in nimconfig.h as u suggested --- #define CONFIG_BT_NIMBLE_MEM_ALLOC_MODE_EXTERNAL 1
But only approx 10180 bytes is transferred to external RAM. ( that too after BLE init).

Please confirm the approach is correct and suggest more ways to reduce the memory.

sanketa-54, 2020-12-03:
> I don't know if there is any way for arduino to produce that data.
> 
> Only way I know is to use IDF and arduino as a component

h2zero, 2020-12-03:
Yes that looks correct. The NimBLE host is quite small.

sanketa-54, 2020-12-04:
Great !!!
Can u suggest some more ways to transfer some memory to pSRAM.

h2zero, 2020-12-04:
I can't think of much else you could do. This is not something I have tried personally.

sanketa-54, 2020-12-07:
Thanks alot for the help.
Please let me know if you got anything.

closed 2022-01-19 by h2zero