h2zero/NimBLE-Arduino #108: A quick question - Set Mac address (iBeacon)

m-menon, 2020-07-31:
A quick one; Is it possible to set the mac address when using this [iBeacon](https://github.com/h2zero/NimBLE-Arduino/tree/master/examples/Refactored_original_examples/BLE_iBeacon) example?

I have a peculiar case where I have no choice but to change the mac address. In short the other scanners should see the mac address that I have set rather than the real mac address of the iBeacon. Any pointers? Thanks.

h2zero, 2020-07-31:
I have not implemented this as a library function yet but it is possible. Do you need a random address that changes often like on phones or just a different address that you set yourself?

m-menon, 2020-07-31:
Thanks for the quick reply. I have a fixed address that I need to set.

h2zero, 2020-08-01:
For that purpose you’ll have to refer to [this document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system.html) as it involves esp function calls. 

When I can get on my pc I’ll check how it is set via NimBLE when it sets a random address.

m-menon, 2020-08-01:
OK cool thanks. I'll check out the documentation and see what needs to be done ... but I am guessing I'll need to change the base_mac ... probably something like

```
uint8_t new_mac[8] = {0x01,0x02,0x03,0x04,0x05,0x06};
esp_base_mac_addr_set(new_mac);
```

Gona try in a while. Thanks once again @h2zero

m-menon, 2020-08-01:
On the first look, it seem like `esp_base_mac_addr_set` works ... just gota remember that for the Bluetooth mac (4 universally administered) it adds `base_mac, +2` to the last octet.

> If the number of universal MAC addresses is two, only two interfaces (Wi-Fi Station and Bluetooth) receive a universally administered MAC address. These are generated sequentially by adding 0 and 1 (respectively) to the base MAC address. The remaining two interfaces (Wi-Fi SoftAP and Ethernet) receive local MAC addresses. These are derived from the universal Wi-Fi station and Bluetooth MAC addresses, respectively.
> 
> If the number of universal MAC addresses is four, all four interfaces (Wi-Fi Station, Wi-Fi SoftAP, Bluetooth and Ethernet) receive a universally administered MAC address. These are generated sequentially by adding 0, 1, 2 and 3 (respectively) to the final octet of the base MAC address.
> 
> When using the default (Espressif-assigned) base MAC address, either setting can be used. When using a custom universal MAC address range, the correct setting will depend on the allocation of MAC addresses in this range (either 2 or 4 per device.)

[Number of universally administered MAC address](url)

h2zero, 2020-08-01:
Yes that would be the case, so if you wanted to specify a MAC address exactly you would have to -2 on the last octet so if you wanted 11:22:33:44:55:66 you would have to set it to 11:22:33:44:55:64.

m-menon, 2020-08-01:
Okay cool ... now messing with one little thing; the name of the device does not appear in the scanner application. I can see the mac address and the advertisement string but the name does not appear ... it's empty

From the scanner app
`Advertised Device: Name: , Address: 62:c1:7a:74:27:69 type: 1, manufacturer data: 4c001005XXXXXXXXXXXXXXXXXX, txPower: 12`

I tried setting the iBeacon name in the `BLEDevice::init("NAMETEST");` ... as well as using `setName()` , `setShortName()` (for `BLEBeacon()`) etc.

I might be missing a tiny something here.

h2zero, 2020-08-01:
You'll need to set the name in the scan response data, the advertising packet doesn't have room.

m-menon, 2020-08-01:
Ok cool. I'll experiment with the scan response and see how it goes. A big thanks @h2zero.

closed 2020-08-01 by m-menon

m-menon, 2020-08-03:
> Okay cool ... now messing with one little thing; the name of the device does not appear in the scanner application. I can see the mac address and the advertisement string but the name does not [...]

Thought I'll mention this; the below worked for me. Earlier I might have missed something and that's why it didn't work.

Example: [https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/Refactored_original_examples/BLE_iBeacon/BLE_iBeacon.ino](https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/Refactored_original_examples/BLE_iBeacon/BLE_iBeacon.ino)

One can use either `setName()` or `setShortName()` depending on what's needed on the other end.

```
void setBeacon()
{
  const char deviceName[] = "TestDevice";
  const char deviceShortName[] = "TestDvc";
  BLEBeacon oBeacon = BLEBeacon();
  oBeacon.setManufacturerId(0x4C00); // fake Apple 0x004C LSB (ENDIAN_CHANGE_U16!)
  oBeacon.setProximityUUID(BLEUUID(BEACON_UUID));
  oBeacon.setMajor((bootCnt & 0xFFFF0000) >> 16);
  oBeacon.setMinor(bootCnt & 0xFFFF);
  BLEAdvertisementData oAdvertisementData = BLEAdvertisementData();
  BLEAdvertisementData oScanResponseData = BLEAdvertisementData();

  /* Set name */
  oAdvertisementData.setName(deviceName);
  /* Set Short name */
  oAdvertisementData.setShortName(deviceShortName);

  oAdvertisementData.setFlags(0x04); // BR_EDR_NOT_SUPPORTED 0x04

  std::string strServiceData = "";
  strServiceData += (char)26;   // Len
  strServiceData += (char)0xFF; // Type
  strServiceData += oBeacon.getData();
  oAdvertisementData.addData(strServiceData);

  pAdvertising->setAdvertisementData(oAdvertisementData);
  pAdvertising->setScanResponseData(oScanResponseData);
  /**  pAdvertising->setAdvertisementType(ADV_TYPE_NONCONN_IND);
  *    Advertising mode. Can be one of following constants:
  *  - BLE_GAP_CONN_MODE_NON (non-connectable; 3.C.9.3.2).
  *  - BLE_GAP_CONN_MODE_DIR (directed-connectable; 3.C.9.3.3).
  *  - BLE_GAP_CONN_MODE_UND (undirected-connectable; 3.C.9.3.4).
  */
  pAdvertising->setAdvertisementType(BLE_GAP_CONN_MODE_NON);
}
```