h2zero/NimBLE-Arduino #109: iBeacon advertisement - how to exclude type ( Question / help )

m-menon, 2020-08-03:
Is it possible to not include the `type` in the iBeacon advertisement. For example when using the iBeacon example, the advertisement is something like this

`Name: , Address: 69:31:32:ea:12:d7 type: 1, manufacturer data: 4c001006351a5e09505b, txPower: 7`

Is it possible to not include the `type`? I mean is there any setting/param to do that? 

I should be able to modify the source to achieve this but if there's a existing way I'd prefer that. Thanks.

m-menon, 2020-08-03:
I am closing this and should there be any problem I'll comment back. Thanks.

m-menon, 2020-08-03:
Sorry, wana do a quick check. In this example what would be the best way to exclude the type in the advertisement string?

https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/Refactored_original_examples/BLE_scan/BLE_scan.ino

```
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
  /*** Only a reference to the advertised device is passed now
    void onResult(BLEAdvertisedDevice advertisedDevice) { **/
    void onResult(BLEAdvertisedDevice* advertisedDevice) {
  /** Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str()); **/
      Serial.printf("Advertised Device: %s \n", advertisedDevice->toString().c_str());
    }
};
```

This prints something like 
`Name: , Address: 66:33:32:3f:12:d7 type: 1, manufacturer data: 4c001006351a5e09505b, txPower: 7`

How can I exclude the `type: 1` in the above advertised device data. Any flag or configurable params? The original nkolban/chegewara version library/scan does not include the type. eg
`Name: , Address: 66:33:32:3f:12:d7 , manufacturer data: 4c001006351a5e09505b, txPower: 7`

Thanks

h2zero, 2020-08-03:
Sorry, that was added for logging purposes mainly, I didn't consider that someone may be using the string representation for their data. 

I will remove this shortly.

m-menon, 2020-08-03:
Great, thanks. I was trying to refactor an older program that parses the broadcast string because that is much easier and straightforward as it contains all the required information :-D

h2zero, 2020-08-03:
Quick fix, just pushed 111ff3a which removed they type from the string.

m-menon, 2020-08-03:
Cool, thanks :100: but there's one extra line it prints ( the [refactored scan program](https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/Refactored_original_examples/BLE_scan/BLE_scan.ino) ). For example there is an `advType:` that appears. 

```
Advertised Device: Name: , Address: b0:52:a0:4c:a0:32, serviceUUID: cbbfe0e1-f7f3-4206-84e0-84cbb3d09dfc
advType:  
Advertised Device: Name: , Address: 6b:50:4a:d0:d2:4f, manufacturer data: 4c0010054118423f1a, txPower: 12
advType:  
```

h2zero, 2020-08-03:
Yes, that's been in there for a long time now. That's the advertisement packet type, not to be confused with the address type. I could remove that as well since there is a function to retrieve it now. Does it cause an issue?

m-menon, 2020-08-03:
> Yes, that's been in there for a long time now. That's the advertisement packet type, not to be confused with the address type. I could remove that as well since there is a function to retrieve it now. Does it cause an issue?

Actually I validate and check for a bunch of things before processing the string further. So it isn't a big issue but the original scan program on which it is based doesn't include the `advType:`

For example this is the result from running the same scan program from ESP32 Ble library (nkolban/chegewara)

```
Name: , Address: b0:a0:16:4c:a0:41, manufacturer data: 4c00021550765cb7d9ea4e2199a4fa879613a4925c67f1e4ce
Name: , Address: 5e:d1:a0:de:e2:12, manufacturer data: 4c00100506184d6bfc, txPower: 12
```

m-menon, 2020-08-03:
> Yes, that's been in there for a long time now. That's the advertisement packet type, not to be confused with the address type. I could remove that as well since there is a function to retrieve it now. Does it cause an issue?

@h2zero On a second thought I think it is perhaps better to remove it. The main reason being that I have never seen it get/print any value. Also it adds an unnecessary check while processing the scan callback. It isn't a big deal when there's a small bunch of beacons but feels like an unnecessary step when there is a whole bunch of beacons. Also the line break it adds instead of just one clean line.

h2zero, 2020-08-04:
> I have never seen it get/print any value

This is due to an optional setting in nimconfig.h to show this (define commented out to save flash space).

Never the less I will remove it as it doesn't provide any useful info unless enabled.

m-menon, 2020-08-04:
Great, thanks. Yup, it doesn't serve much purpose in there. Those needing advType could call `advertisedDevice->getAdvType();` I guess. Cheers.

h2zero, 2020-12-29:
Closing as resolved.

closed 2020-12-29 by h2zero