h2zero/NimBLE-Arduino #110: Custom ble advertisement [Not iBeacon standard/Data spec]

m-menon, 2020-08-06:
I want to send out some custom ble advertisement. Is something like this possible with NimBLE-Arduino library at the moment? I got the below code working (with [ESP32 BLE for Arduino](https://github.com/nkolban/ESP32_BLE_Arduino)) after getting some help from chegewara. I am yet to try anything with NimBLE-Arduino along the same lines but I thought I'd get a word from you. I am referring to this [example](https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_iBeacon/BLE_iBeacon.ino). Anything similar with NimBLE-Arduino? Thanks.

```
// Create the BLE Device
BLEDevice::init("");
// Create the BLE Server
// BLEServer *pServer = BLEDevice::createServer(); // <-- no longer required to instantiate BLEServer, less flash and ram usage
pAdvertising = BLEDevice::getAdvertising();

uint8_t Adv_DATA[] = {0x4c, 0x00, 0x02, 0x15, 0xe9, 0xd5, 0xad, 0x15};
BLEAdvertisementData oAdvertisementCustom = BLEAdvertisementData();
oAdvertisementCustom.setManufacturerData(std::string((char *)&Adv_DATA[0], 8)); // 8 is length of Adv_DATA
pAdvertising->setAdvertisementData(oAdvertisementCustom);
pAdvertising->setScanResponse(false);
//  oAdvertisementCustom.setFlags(0x04);
oAdvertisementCustom.setFlags(0x06);

// Start advertising
pAdvertising->start();
Serial.println("Advertising started...");
delay(100);
pAdvertising->stop();
```

h2zero, 2020-08-06:
Yes, that code should just work, only have to change the include header.

m-menon, 2020-08-06:
Cool. Any chance of adding a simple example ( or perhaps a few lines to the existing example )? I am not so familiar with the flags n stuff otw I would want to.

h2zero, 2020-08-06:
The code you posted makes a pretty good simple example of custom data.

Any specific things you would like to see or do? 

All I can think of at the moment would be setting the device name and custom scan response (done the same as advertised data).

m-menon, 2020-08-07:
> The code you posted makes a pretty good simple example of custom data.
> 
> Any specific things you would like to see or do?
> 
> All I can think of at the moment would be setting the device name and custom scan response (done the same as advertised data).

Ok cool. I just wanted to check if there's any important setting / flag that I may have missed. If I haven't then it's cool. Thanks a lot. I'll leave this open for a day or two - just in case there's any update to be done to the code snippet that I posted. It may help others cos I think there seems to be fair number of folks looking to just broadcast stuff without strictly following the iBeacon standard.

closed 2020-08-10 by m-menon