h2zero/NimBLE-Arduino #111: `GetService(ServiceUUID)` does not succeed with 16-bit ServiceUUID

wakwak-koba, 2020-08-09:
The BLE-HID device I'm using in my test appears to own 0x1812 when I view the ServiceUUID list after doing `getSerices(true)`.
However, if I do getServices(true) first and do getService(NimBLEUUID((uint16_t)0x1812) it fails.

Successful Code:
```c++
NimBLEUUID    uuid_service((uint16_t)0x1812);

pClient->getServices(true);
auto pService = pClient->getService(uuid_service);
```
Failing code:
```c++
NimBLEUUID    uuid_service((uint16_t)0x1812);

// pClient->getServices(true);
auto pService = pClient->getService(uuid_service);   // nullptr is returned
```
The ServiceUUID on this device I needed was appearing as 00001812-0000-1000-8000-00805f9b34fb instead of 0x1812 in previous BLE libraries.
However, specifying 0001812-0000-1000-8000-00805f9b34fb still fails.

Failing code, too:
```c++
NimBLEUUID    uuid_service("0001812-0000-1000-8000-00805f9b34fb");

// pClient->getServices(true);
auto pService = pClient->getService(uuid_service);   // nullptr is returned
```
What do you think ?

h2zero, 2020-08-09:
This is quite puzzling, it's possible there is a bug in the ble_gattc_disc_svc_by_uuid() but I would suspect not since no other issues regarding this have been reported and would likely have been by now on the upstream. 

My best guess is that the device you're connecting to does not respond to that request and only seems to respond to the "get all" request.

If you post a log with NimBLE debug enabled in nimconfig.h I could confirm/deny this.

ntwallace, 2020-08-13:
I'm wondering if this is related to a bug I'm having where calling `createService(UUID)` or `createCharacteristic(UUID)` with a uint16_t UUID causes an abort boot loop on the ESP.

For example, `NimBLEService *serviceDevice = server->createService((uint16_t)0x180a);` results in: 
```Backtrace: 0x400943fc:0x3ffda530 0x4009462d:0x3ffda550 0x400fc043:0x3ffda570 0x400faace:0x3ffda5c0 0x400d42d1:0x3ffda5f0 0x400f261f:0x3ffda630 0x400907f5:0x3ffda660
  #0  0x400943fc:0x3ffda530 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4009462d:0x3ffda550 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x400fc043:0x3ffda570 in NimBLEServer::start() at lib/NimBLE/src/NimBLEServer.cpp:224
  #3  0x400faace:0x3ffda5c0 in NimBLEAdvertising::start() at lib/NimBLE/src/NimBLEAdvertising.cpp:149
  #4  0x400f261f:0x3ffda630 in TaskDraw(void*) at src/./tasks/TaskSerial.h:47
  #5  0x400907f5:0x3ffda660 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)```
```

But, `NimBLEService *serviceDevice = server->createService("7537fbd5-d66c-4b03-b132-47b1e18285ee");` works fine.

h2zero, 2020-08-17:
I wonder about this, both of your issues involve uuid's in the 0x18xx range. I'll try to find what the issue could be and report back.

h2zero, 2020-08-17:
Just did some testing using with a server with service uuid 0x1812 and client calling getService((uint16_t)0x1812) I have not encountered any issues.

I will need more information / code to reproduce.

tonton81, 2020-12-27:
You need to call it like this:

NimBLEUUID    uuid_service(**(BLEUUID)**(uint16_t)0x1812);

Thats how mine connects using 16bit UUID on both NimBLE and BLEClient

h2zero, 2020-12-27:
>NimBLEUUID uuid_service((BLEUUID)(uint16_t)0x1812); // <-- you need to cast it to BLEUUID

That's due to different internal data structures. In bluedroid it uses `esp_bt_uuid_t` and it's first member variable is `uint16_t`. This is not the case in NimBLE. It's also not necessary to do at all, just passing `(uint16_t)0x1812` or if preferred `NimBLEUUID((uint16_t)0x1812)` will do the job.

tonton81, 2020-12-27:
I tried that as well. my UUID is 0xFFD0 for service which it won't accept, and 0xFFD1 is characteristic, which never reaches that point due to service failing

h2zero, 2020-12-27:
This is strange behavior indeed, I have never seen this personally but that doesn't mean there isn't something wrong. What devices are you connecting to?

Also, the fix for this was posted above, before getting the specific service try calling `XBT_client[controller]->getServices(true);` and let me know if that works.

tonton81, 2020-12-27:
It's a Morimoto XBT RGBW led controller

Yeah forgot to mention I added getServices(true), same issue

`void sync_XBT_controllers(int controller) {
  XBT_client[controller]->getServices(true);
  NimBLERemoteService* _service = XBT_client[controller]->getService((BLEUUID)((uint16_t)0xFFD0));
  if (_service == nullptr) {
    Serial.println("SERVICE ERROR");
    XBT_client[controller]->disconnect();
    return;
  }
`

h2zero, 2020-12-27:
If that is the code you're using currently, please try changing:
```
getService((BLEUUID)((uint16_t)0xFFD0));
```
to: `getService(BLEUUID((uint16_t)0xFFD0));` or `getService("FFD0");` and let me know if it helps.

tonton81, 2020-12-27:
Still service error on both attempts

`Connected to f0:b5:d1:8f:ba:e5
D NimBLEClient: "<< connect()"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Service Discovered >> status: 0 handle: 1"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 0 handle: 12"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 0 handle: 16"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 0 handle: 35"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 0 handle: 43"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
D NimBLEClient: "<< << Service Discovered"
D NimBLEClient: "<< retrieveServices"
I NimBLEClient: "Found 5 services"
D NimBLEClient: ">> getService: uuid: 0xffd0"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
D NimBLEClient: "<< << Service Discovered"
D NimBLEClient: "<< retrieveServices"
D NimBLEClient: "<< getService: not found"
SERVICE ERROR
D NimBLEClient: ">> connect(f0:b5:d1:8f:b6:b8)"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: "Got Client event "
D NimBLEClient: "Connection established"
D NimBLEClient: "Got Client event "
I NimBLEClient: "mtu update event; conn_handle=1 mtu=23"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
Connected to f0:b5:d1:8f:b6:b8
D NimBLEClient: "<< connect()"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Service Discovered >> status: 0 handle: 1"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 0 handle: 12"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 0 handle: 16"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 0 handle: 35"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 0 handle: 43"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService()"
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
D NimBLEClient: "<< retrieveServices"
D NimBLEClient: "<< << Service Discovered"
I NimBLEClient: "Found 5 services"
D NimBLEClient: ">> getService: uuid: 0xffd0"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
D NimBLEClient: "<< << Service Discovered"
D NimBLEClient: "<< retrieveServices"
D NimBLEClient: "<< getService: not found"
SERVICE ERROR
D NimBLEScan: ">> start(duration=1)"
D NimBLEScan: "<< start()"
Free Heap: 255704, debug refresh time: 2086`

h2zero, 2020-12-27:
Thanks for the log, I should change those debug messages to print uuids.

If you connect with your phone how many services does it find? I can see there are 5, I'm wondering if there is some missing in the results.

tonton81, 2020-12-27:
[image]

tonton81, 2020-12-27:
FFD0 is service and FFD1 is characteristic. These RGB modules don't seem to have authentication, their phone-app only links by highest RSSI value when phone is placed near the device. I don't need a close range as I connect to their pre-set mac address when it's found in the scan, making this universal accross all their units as they use static service and characteristic which is identical on all their units

h2zero, 2020-12-28:
Awesome, thanks! So, seeing that there is only 5 services on the phone and the debug traces also showed 5 services, that would imply that there is something wrong with the uuid matching. I'm not sure why that would be but I'll start investigating.

h2zero, 2020-12-28:
Just a thought, there might be an endian issue here. Could you try `getService` with the uuid `0xD0FF`? I don't know why or how that would be a problem but I would like to rule it out first.

h2zero, 2020-12-28:
If you wouldn't mind, could you change [this line](https://github.com/h2zero/NimBLE-Arduino/blob/5cb454bfce3bf753e26d00817b7bf66cc0de447a/src/NimBLERemoteService.cpp#L52) to:
`NIMBLE_LOGD(LOG_TAG, "<< NimBLERemoteService: %s", m_uuid.toString().c_str());`
and post a log of the results. That will greatly speed up the troubleshooting.

tonton81, 2020-12-28:
edited the line, tried with reversing endianess, still fail:

`Connected to f0:b5:d1:8f:b6:b8
D NimBLEClient: "<< connect()"
D NimBLEClient: ">> deleteServices"
D NimBLEClient: "<< deleteServices"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Service Discovered >> status: 0 handle: 1"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService: 0x1800"
D NimBLEClient: "Service Discovered >> status: 0 handle: 12"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService: 0x1801"
D NimBLEClient: "Service Discovered >> status: 0 handle: 16"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService: 0x180a"
D NimBLEClient: "Service Discovered >> status: 0 handle: 35"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService: 0000ffd0-0000-1000-8000-00805f9b34fb"
D NimBLEClient: "Service Discovered >> status: 0 handle: 43"
D NimBLERemoteService: ">> NimBLERemoteService()"
D NimBLERemoteService: "<< NimBLERemoteService: f000ffc0-0451-4000-b000-000000000000"
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
D NimBLEClient: "<< retrieveServices"
D NimBLEClient: "<< << Service Discovered"
I NimBLEClient: "Found 5 services"
D NimBLEClient: ">> getService: uuid: 0xffd0"
D NimBLEClient: ">> retrieveServices"
D NimBLEClient: "Service Discovered >> status: 14 handle: -1"
D NimBLEClient: "<< << Service Discovered"
D NimBLEClient: "<< retrieveServices"
D NimBLEClient: "<< getService: not found"
SERVICE ERROR
Free Heap: 254464, debug refresh time: 2001
XBT 0: connected
XBT 1: disconnected
XBT 2: connected
XBT 3: disconnected
D NimBLEScan: ">> start(duration=1)"
D NimBLEScan: "<< start()"
I NimBLEScan: "NEW DEVICE FOUND: 58:8a:3b:16:f3:18"
D NimBLEAdvertisedDevice: "payload: 02011a020a0c0bff4c001006081eb0727131"
I NimBLEScan: "NEW DEVICE FOUND: 7c:20:5f:14:71:3f"
D NimBLEAdvertisedDevice: "payload: 02011a020a0c0aff4c0010055e1c0220e5"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 7c:20:5f:14:71:3f"
I NimBLEScan: "NEW DEVICE FOUND: 00:f0:1c:63:09:c4"
D NimBLEAdvertisedDevice: "payload: 1eff0600010920021eac695509d4fde0d399fab31b9821690ea0daa9933f20"
I NimBLEScan: "NEW DEVICE FOUND: 42:1c:9a:1d:1f:55"
D NimBLEAdvertisedDevice: "payload: 03036ffd17166ffdd57f2f55c1df324f4bc63003a25913ac4a444570"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 42:1c:9a:1d:1f:55"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 42:1c:9a:1d:1f:55"
D NimBLEAdvertisedDevice: "payload: 03036ffd17166ffdd57f2f55c1df324f4bc63003a25913ac4a444570"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 42:1c:9a:1d:1f:55"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 7c:20:5f:14:71:3f"
D NimBLEAdvertisedDevice: "payload: 02011a020a0c0aff4c0010055e1c0220e5"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 7c:20:5f:14:71:3f"
D NimBLEScan: "discovery complete; reason=0"
D NimBLEScan: ">> start(duration=1)"
D NimBLEScan: "<< start()"
I NimBLEScan: "NEW DEVICE FOUND: 5d:fa:47:e2:5f:4c"
D NimBLEAdvertisedDevice: "payload: 0201021107fc9dd0b3cb84e0840642f3f7e1e0bfcb"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 5d:fa:47:e2:5f:4c"
I NimBLEScan: "NEW DEVICE FOUND: 6b:0e:cb:a8:95:1d"
D NimBLEAdvertisedDevice: "payload: 02010213ffffff30653362353832303264656134356362020af103035634"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 6b:0e:cb:a8:95:1d"
D NimBLEAdvertisedDevice: "payload: 0709534849454c44"
I NimBLEScan: "NEW DEVICE FOUND: cc:6e:a4:ee:b1:d1"
D NimBLEAdvertisedDevice: "payload: 02011a1bff7500420401206f18070002010000000000000000000000000000"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: cc:6e:a4:ee:b1:d1"
D NimBLEAdvertisedDevice: "payload: 1c085b54565d2053616d73756e67205136205365726965732028383229"
[... 118 lines cut ...]
D NimBLEAdvertisedDevice: "payload: 02011a020a0c0aff4c0010055e1c0220e5"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 6b:0e:cb:a8:95:1d"
D NimBLEAdvertisedDevice: "payload: 02010213ffffff30653362353832303264656134356362020af103035634"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 6b:0e:cb:a8:95:1d"
D NimBLEAdvertisedDevice: "payload: 0709534849454c44"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: cc:6e:a4:ee:b1:d1"
D NimBLEAdvertisedDevice: "payload: 02011a1bff7500420401206f18070002010000000000000000000000000000"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: cc:6e:a4:ee:b1:d1"
D NimBLEAdvertisedDevice: "payload: 1c085b54565d2053616d73756e67205136205365726965732028383229"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 42:1c:9a:1d:1f:55"
D NimBLEAdvertisedDevice: "payload: 03036ffd17166ffdd57f2f55c1df324f4bc63003a25913ac4a444570"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 42:1c:9a:1d:1f:55"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: cc:6e:a4:ee:b1:d1"
D NimBLEAdvertisedDevice: "payload: 02011a1bff7500420401016fcc6ea4eeb1d1ce6ea4eeb1d0065544b6000000"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: cc:6e:a4:ee:b1:d1"
D NimBLEAdvertisedDevice: "payload: 1c085b54565d2053616d73756e67205136205365726965732028383229"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 6b:0e:cb:a8:95:1d"
D NimBLEAdvertisedDevice: "payload: 02010213ffffff30653362353832303264656134356362020af103035634"
I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 6b:0e:cb:a8:95:1d"
D NimBLEAdvertisedDevice: "payload: 0709534849454c44"
Free Heap: 254520, debug refresh time: 2001
XBT 0: connected
XBT 1: disconnected
XBT 2: connected
XBT 3: disconnected`

tonton81, 2020-12-28:
Easier to see with a snapshot i guess :P

[image]

tonton81, 2020-12-28:
Looks like the 128 bit one went in!

`0000ffd0-0000-1000-8000-00805f9b34fb`

now it's asking for the char... So i guess 16bit is broken

wakwak-koba, 2020-12-28:
I was the first person to publish this issue.

When `pClient-> getServices (true);` was not called in advance, finally, `ble_gattc_disc_svc_uuid_tx`   is called.
https://github.com/h2zero/NimBLE-Arduino/blob/5cb454bfce3bf753e26d00817b7bf66cc0de447a/src/nimble/host/src/ble_gattc.c#L1609-L1613

Here, since it is limited to `BLE_ATT_UUID_PRIMARY_SERVICE`, I guessed that `BLE_ATT_UUID_SECONDARY_SERVICE` might have been omitted from the search target, and I thought that this was unavoidable behavior, but I gave up, but my guess.

Is it a difference?

tonton81, 2020-12-28:
wakwak i tried your copy but the ringbuffer compile error didnt let me run it last night. so far i know service and characteristic work at 128bits now, because i was able to control the LED channels, just gotta change my writeValue arrays to prevent compile warnings, but so far so good

h2zero, 2020-12-28:
Lol, well that's interesting for sure. I'm not quite sure as to the best course of action yet but thank you all very much for helping out and posting data/solutions.

@tonton81 For now, using the 128 bit uuid's is the best way forward. NimBLE is very precise and does not filter off the default 128 bit uuid construction for uuid's during comparison, but this is something I can add to the library. For added simplicity you could call the getX() functions with `getX(BLEUUID((uint16_t)0xFFD0).to128());` for now. You can expect a proper fix for this soon.

@wakwak-koba 
>Here, since it is limited to BLE_ATT_UUID_PRIMARY_SERVICE, I guessed that BLE_ATT_UUID_SECONDARY_SERVICE might have been omitted from the search target

This is a very good thought, I will look at this more as well, not sure if it's related to your issue but it's worth investigaing. I will make a peripheral device with secondary services and test the effects.

h2zero, 2020-12-28:
I went over this thread again and realized I didn't respond to @ntwallace, sorry about that.

> a bug I'm having where calling createService(UUID) or createCharacteristic(UUID) with a uint16_t UUID causes an abort boot loop on the ESP

I'm sure you fixed this by now, but just in case others come accross this, the error is caused by not calling `start()` on the service before advertising start.

h2zero, 2020-12-28:
A note to add to this; The reason the original library worked in this scenario is due to it's use of `toString()` to convert the 16bit uuid to 128 and then perform a string comparison. I never noticed it before, but was a good idea for using it's code base. I'm working on something similar.

tonton81, 2020-12-28:
Waaaay better than BLEClient at fast responses to disconnects and able to reconnect consistantly (so far) and without lockups. I'm pushing the color switching at 100ms but it can definately go faster, just cleaning up the code bloat first :)
(This is without the FreeRTOS.cpp mod people do for the timeout lockup on BLEClient (which doesn't effectively work after several disconnects!)).

https://www.youtube.com/watch?v=p4Ms35I--jI&feature=youtu.be

h2zero, 2020-12-28:
Haha, glad you got it working 😄 . Looks very responsive in the video. You can push that to 7.5ms I'm sure.

Almost finished with the uuid issue, will finish testing tomorrow and push the fix. Thanks for the help!

h2zero, 2020-12-28:
Fix has been pushed in commit 9327517.

tonton81, 2020-12-28:
Ok, went to test both :)

`  NimBLERemoteService* _service = XBT_client[controller]->getService("0000ffd0-0000-1000-8000-00805f9b34fb");
  NimBLERemoteService* _service = XBT_client[controller]->getService((uint16_t)0xFFD0);
`

Both work

h2zero, 2020-12-28:
Great, thanks for confirming!

h2zero, 2020-12-28:
>By the way, for not loosing frames in BT on my device, I monitored the writeValue returns, it would toggle true or false on writes if i go below 2ms rates of writes, so I just delay(2) or delay(3) and all are successful now. Guess it's good way to monitor bandwidth thresholds :)

Yes, that would be a good way to flow control test.

tonton81, 2020-12-28:
Service is fixed, char is failing now

Serial shows "CHAR FAIL!!" before a panic

`  XBT_characteristic[controller] = _service->getCharacteristic((uint16_t)0xFFD1); /* "0000ffd1-0000-1000-8000-00805f9b34fb" */
  if (XBT_characteristic[controller] == nullptr) {
    Serial.println("CHAR FAIL!!");
    XBT_client[controller]->disconnect();
    return;
  }`

h2zero, 2020-12-28:
Yes, this is due to communicating to the NimBLE stack to ask for a 16bit uuid and it does exactly that. Since that code doesn't use the comparison check in the library it will still fail. Further changes will be necessary.

Can you decode the backtrace and post it?

tonton81, 2020-12-28:
Debug option? I am new to ESP32 due to the fact I took the first thing that I got working on the XBT controller lol

h2zero, 2020-12-28:
You will need [this](https://github.com/me-no-dev/EspExceptionDecoder), very important tool for esp32.

tonton81, 2020-12-28:
[image]

h2zero, 2020-12-28:
Ahh, there it is. Seems you have called writeValue somewhere after the getcharacteristic failed, which would make it a nullptr.

h2zero, 2020-12-28:
I will work on the next part of the fix right away. I will update here when it's done.

tonton81, 2020-12-28:
writeValue is protected by pClient's isConnected() in my sketch, that probably exaplains why the flag is 1 for isConnected before the services/characteristics is populated. If thats the case I would need to check nullptr on service and characteristic and isConnected() before doing writeValues, complex but that probably ressolves most issues with others who use BLE in general.

h2zero, 2020-12-28:
>The disconnect() though should have dropped isConnected() to false when setting a bad characteristic

This can take some time to trigger, not reliable unless you wait for it. Checking if something is nullptr before calling it's methods is best practice for sure.

Here is a patch I made to address the rest of the uuid issue, if you wouldn't mind testing it before I push?
[uuid_patch.txt](https://github.com/h2zero/NimBLE-Arduino/files/5748662/uuid_patch.txt)

tonton81, 2020-12-28:
I tried copy pasting it but failed miserably (the way i did it perhaps), has to remove some (-/+) and there was missing brackets. It's not working most likely i did something wrong here :P

h2zero, 2020-12-28:
I messed up the patch command there, lol sorry about that. I push it to a branch to make it easier.

tonton81, 2020-12-28:
Still get char faild

tonton81, 2020-12-28:
This is the Debug output:
[image]

h2zero, 2020-12-28:
Can you post the debug of when you call getcharacteristic? All I see in there is the advertising results.

tonton81, 2020-12-28:
[image]

h2zero, 2020-12-28:
Thanks, I see the issue now, will update that branch when done.

h2zero, 2020-12-28:
Just pushed a proper fix, sorry about that. I tested this against my NimBLE server device and it seems to be working correctly. Please try again if you don't mind and also try without `getServices(true)` as it should now work without.

tonton81, 2020-12-28:
yup it works with and without getServices(true)

[image]

h2zero, 2020-12-28:
Awesome, thank you for your patience. I will merge it then.

@wakwak-koba I'm curious if this will fix your issue as well? Please post back here if you test it.

tonton81, 2020-12-28:
Does NimBLE also work on Adafruit's nRF52 boards?

h2zero, 2020-12-29:
The answer to this is yes and no lol. You could run the mynewt OS with NimBLE on it as it is supported I believe (should be). However I have not yet ported this library to support it (future project).

tonton81, 2020-12-29:
Hey i think the disconnect() issues have been ressolved, i will have to let it run to see if it's good. I think the nullptr issue on writeValue during a failed characteristic match leaving isConnected() intact as true was the culprit

h2zero, 2020-12-29:
Can you post a log of the disconnect issue? I'm working on finding a solution to this but it seems to change with circumstances.

tonton81, 2020-12-29:
Hmm I will have to catch it again. All that happens is after consecutive disconnects, further disconnect()'s actually do nothing to the connection anymore

h2zero, 2020-12-29:
If I'm interpreting this correctly you are doing something like:
```
while(isConnected()) {
    disconnect();
   delay():
}
```

If so, that may cause the issue as the controller will stop responding. Only need to call it once, delay until the callback or for the supervision timeout time (+a few ms).

tonton81, 2020-12-29:
i usually disconnect() up to 4 clients (only 2 online) every 7 seconds (if theyre online)

[image]

h2zero, 2020-12-29:
This actually reminds me of something I did years ago in the old library to work around this, completely forgot about it. What I did was set a flag, isDisconnecting, and tested against that and isConnected to decide if disconnect should be called. I'm still playing with this as well but I can only reproduce the situation occasionally on peripherals with poor connections.

tonton81, 2020-12-29:
[dbgesp.txt](https://github.com/h2zero/NimBLE-Arduino/files/5749501/dbgesp.txt)

check this out:
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
D NimBLEClient: ">> disconnect()"
E NimBLEClient: "ble_gap_terminate failed: rc=2 "
D NimBLEClient: "<< disconnect()"

h2zero, 2020-12-29:
Yep rc=2 means the NimBLE host stack thinks you're already disconnected. When you called it that time it was already in progress waiting for the controller to callback when it was complete.

[9 more replies]

closed 2021-01-02 by wakwak-koba