h2zero/NimBLE-Arduino #112: reading values from BLE device (LYWSD03MMC)

michapr, 2020-08-13:
Hi,

I have tried your lib with the implementation for this BLE sensor:
https://github.com/karolkalinski/esp32-snippets/blob/master/Mijia-LYWSD03MMC-Client/Mijia-LYWSD03MMC-Client.ino
but get same problems as with original library.

```
...
17:10:08.987 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
17:10:08.987 -> Core 1 register dump:
17:10:08.987 -> PC      : 0x400d427a  PS      : 0x00060330  A0      : 0x800d441b  A1      : 0x3ffc5f10  
17:10:08.987 -> A2      : 0x00000000  A3      : 0x3ffc5f64  A4      : 0x3ffc2948  A5      : 0x0000ff00  
17:10:08.987 -> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000001  A9      : 0x3ffc5f40  
17:10:08.987 -> A10     : 0x3ffbed10  A11     : 0x3f403505  A12     : 0x3f403505  A13     : 0x0000ff00  
17:10:09.029 -> A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
17:10:09.029 -> EXCVADDR: 0x00000050  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
17:10:09.029 -> 
17:10:09.029 -> Backtrace: 0x400d427a:0x3ffc5f10 0x400d4418:0x3ffc5f50 0x400d445c:0x3ffc5fa0 0x400d448e:0x3ffc5fc0 0x400d1fb1:0x3ffc5fe0 0x400d20c4:0x3ffc6020 0x400d213e:0x3ffc6060 0x400e5ecd:0x3ffc6090 0x4008d9ad:0x3ffc60b0
17:10:09.029 -> 
17:10:09.029 -> Rebooting...
...
```
This happen sometimes after connecting to device and reading characteristics

It does not make difference using one device or more devices (same type) - or what distance to device.
It happen every 3rd ... 5th call in most cases, a known problem with original lib as I have found at other places...
Device was disconnected before and before reading next sensor was used esp_bt_controller_deinit(); - no difference.

Any idea how to solve it without rebooting device?

Thanks!
Michael

1technophile, 2020-08-13:
I'm currently integrating this sensor in the scope of [OpenMQTTGateway](https://github.com/1technophile/OpenMQTTGateway), you can find the discussion [here](https://github.com/1technophile/OpenMQTTGateway/issues/540), we are facing the same issue with this [branch](https://github.com/1technophile/OpenMQTTGateway/tree/BLEConnect) but I'm working on it and will report back on the [issue](https://github.com/1technophile/OpenMQTTGateway/issues/540).

h2zero, 2020-08-13:
In that example you are calling connect() and not checking if it returns successfully or not. Then you call getServices(), and if the client is not connected it might be the cause of this error.

1technophile, 2020-08-13:
> In that example you are calling connect() and not checking if it returns successfully or not. Then you call getServices(), and if the client is not connected it might be the cause of this error.

Planet alignment :-) , this is exactly the mistake I was doing in OpenMQTTGateway and now I have corrected it, the gateway is running since 5h without any reset.

h2zero, 2020-08-13:
@1technophile Nice, I may add some protections to the client methods for this. Printing an error would be preferable to a reboot.

h2zero, 2020-08-13:
@michapr One other issue I found:
```
void registerNotification() {

  // Obtain a reference to the service we are after in the remote BLE server.
  BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
  if (pRemoteService == nullptr) {
    Serial.print("Failed to find our service UUID: ");
    Serial.println(serviceUUID.toString().c_str());
    pClient->disconnect();
  }
```
You should consider returning at the end of this if block. If `pRemoteService == nullptr` and you do not return here you will be calling getCharacteristics on a nullptr, which will cause this exception.

michapr, 2020-08-13:
@h2zero Thanks for your answers, in my own code (with loop for 3 sensors) I made some modifications, I wasn't sure how much they affect the function because the result was always the same...
After connect() I'm checking with 
```
if (pclient->isConnected()) {
..
}
```
if is connected - and the debug log show me that it is connected - but anyway crash.

In registerNotification already have added the return after pClient->disconnect()

But what I can say - there are no so much reboots as before with original lib, so I will check more tomorrow.
Thanks!

h2zero, 2020-08-13:
@michapr After the if block I posted there is another one that needs to return also (or an else statement after). It could be calling registerForNotify() on a nullptr.

1technophile, 2020-08-13:
@michapr here is my [connect](https://github.com/1technophile/OpenMQTTGateway/blob/5536c4ea45b8713d5436da9e67572c61e4041b3f/main/ZgatewayBT.ino#L514) code for info

michapr, 2020-08-14:
@1technophile Thanks, already have seen in your code changes in OpenMQTTGateway ;)

@h2zero thank you for the hint to check it again - there was really missing a return at other place...
I made a request in the test code for battery state UUID (wrong for this device - every time get 100%, I know ...) and here was a return missing ...

Closing this for now, Thanks!
Michael

closed 2020-08-14 by michapr