h2zero/NimBLE-Arduino #113: Migration Guide for arduino-esp32 Users

vicatcu, 2020-08-13:
This is not the "pre-packaged" BLE stack in the arduino-esp32 project, but many issues in that repository have comment threads that allude to this project as an alternative. It would be very helpful to that community if a migration guide were published that explained  the differences in how to use this stack when you've already got an implementation that uses the incumbent BLE stack in the arduino-esp32 project. Pre-requisites, considerations, pitfalls, analogies, etc. would go a long way to lowering the intimidation factor of moving to a new stack.

h2zero, 2020-08-13:
This is a good suggestion. 

I have created the API differences document and refactored the original examples as a "quick start" but something more comprehensive may be in order now after many of the recent updates.

This was intended as a minimal changes needed project to make the migration easy and code familiar. But there are some things that should be documented more clearly in a "guide". 

If you or anyone have any thoughts as to specific things that should be highlighted or want to contribute to the documentation I would appreciate it.

h2zero, 2020-08-14:
@vicatcu I have created a basic migration guide, you can find it [here](https://h2zero.github.io/esp-nimble-cpp/md_docs__migration_guide.html) It's a work in progress but covers the most important parts.

h2zero, 2020-12-29:
Closing as resolved.

closed 2020-12-29 by h2zero