h2zero/NimBLE-Arduino #116: Bond pairing again fails with BLE_ERR_PINKEY_MISSING.

gkoh, 2020-08-17:
Firstly, thanks for a lovely library, really nice to use.

I ran into a problem, which took a while to track down.
If I successfully bond pair once, then try again, 'secureConnection()' fails.
I hacked in to see the taskData return code and it's BLE_ERR_PINKEY_MISSING.

After searching up and down the stack, I finally worked out it's because I never deleted the ESP32's bonding keys before re-attempting the pairing.

Once I added the following to my client removal code:
```
  {
    ble_addr_t peerAddr;
    memcpy(&peerAddr.val, m_Address.getNative(), 6);
    peerAddr.type = m_Address.getType();
    ble_gap_unpair(&peerAddr);
  }
```

it started working as expected.

To abstract this from the user, we could add a function to unpair a NimBLEAddress.
Additionally we could try to unpair automatically if secureConnection() fails, we're in the descriptor/characteristic retry loop and the error is 'PINKEY_MISSING' or 'REPEAT_PAIRING_RETRY' similar.

It's unclear to me how the REPEAT_PAIRING_RETRY error from the depths of the Nimble stack turns into PINKEY_MISSING.

h2zero, 2020-08-17:
Hi @gkoh, thanks for the detailed description. When you say "REPEAT_PAIRING_RETRY" are you referring to  BLE_GAP_EVENT_REPEAT_PAIRING? If that is the case, we can simply handle that as is done with the server code.

 The client does not have a handler for that event yet (totally forgot about that, thanks for the reminder).

>It's unclear to me how the REPEAT_PAIRING_RETRY error from the depths of the Nimble stack turns into PINKEY_MISSING.

What happens (I think), is when you're already paired and trying to re-pair it triggers the REPEAT_PAIRING event, which we are not handling at the moment. So without the handler to delete the previous pairing info we end up not sending any key/pin info, thus the error.

gkoh, 2020-08-18:
> Hi @gkoh, thanks for the detailed description. When you say "REPEAT_PAIRING_RETRY" are you referring to BLE_GAP_EVENT_REPEAT_PAIRING? If that is the case, we can simply handle that as is done with the server code.

I recall I traced it to an rc of BLE_GAP_REPEAT_PAIRING_RETRY somewhere, which I guess becomes the equivalent event.

> The client does not have a handler for that event yet (totally forgot about that, thanks for the reminder).

I'm about to lose access to the device I've been testing against, so I might not be able to confirm the exact event and error code for a while.
 
> What happens (I think), is when you're already paired and trying to re-pair it triggers the REPEAT_PAIRING event, which we are not handling at the moment. So without the handler to delete the [...]

Ah, that makes sense, thanks.

gkoh, 2020-08-18:
I've just checked over the logs I had, this is the debug snippet before the disconnection:
```
Command Status: status=0 cmd_pkts=5 ocf=0x19 ogf=0x8
LE Remote Used Features. handle=0 feat=01 00 00 00 00 00 00 00
Encrypt change: status=6 handle=0 state=0
D NimBLEClient: "Got Client event BLE_GAP_EVENT_ENC_CHANGE"
```

Unfortunately, I didn't capture what came before that, so I'm not certain we got BLE_GAP_EVENT_REPEAT_PAIRING prior to the ENC_CHANGE event.
I did note that 'status=6' is BLE_ERR_PINKEY_MISSING.

h2zero, 2020-08-18:
>I'm about to lose access to the device I've been testing against, so I might not be able to confirm the exact event and error code for a while.

That's unfortunate, however from the look of the log and description I'm quite sure of my assumption and should add the handling for that event. This should resolve the issue at best and resolve potential future issues at worst. 

If you are still able to test with this device could you try adding:
```
case BLE_GAP_EVENT_REPEAT_PAIRING: {
    rc = ble_gap_conn_find(event->repeat_pairing.conn_handle, &desc);
    assert(rc == 0);
    ble_store_util_delete_peer(&desc.peer_id_addr);

    return BLE_GAP_REPEAT_PAIRING_RETRY;
} // BLE_GAP_EVENT_REPEAT_PAIRING
```
to NimBLEClient::handleGapEvent()  and test?

gkoh, 2020-08-18:
> If you are still able to test with this device could you try adding:

Gimme a day or so, I'm going to ask for some more loan time, thanks.

gkoh, 2020-08-18:
> > If you are still able to test with this device could you try adding:
> 
> Gimme a day or so, I'm going to ask for some more loan time, thanks.

OK, managed to keep it for more testing.
I've now taken a more methodical approach to testing this and understand what is going on.

The specific scenario I encountered was:
 - delete pairing from target device
 - scan for target
 - try to bond again

And this is where it got indirect, after going over the logs carefully, NimBLEClient gets BLE_GAP_EVENT_ENC_CHANGE, with status of BLE_ERR_PINKEY_MISSING.
That error is actually telling me the key is missing from the target, not from the esp32.

I have not in fact been able to get into a BLE_GAP_EVENT_REPEAT_PAIRING situation.
I suspect this issue may be invalid, unless we wish to handle this specific case.

Nevertheless, I think a nice abstracted way to unpair manually (and hence delete the local keys) would be nice.

h2zero, 2020-08-18:
Thanks! That makes sense, the peripheral is informing us that it has deleted the bonding information. In that situation we should handle that by deleting the bond as well and re-pair. 

>Nevertheless, I think a nice abstracted way to unpair manually (and hence delete the local keys) would be nice.

This should be done also. I had intended to add bond management API’s, just hasn’t been a priority yet.

gkoh, 2020-08-19:
> Thanks! That makes sense, the peripheral is informing us that it has deleted the bonding information. In that situation we should handle that by deleting the bond as well and re-pair.

Are you suggesting the following change in NimBLEClient::handleGapEvent():
```
case BLE_GAP_EVENT_ENC_CHANGE: {
   if (event->enc_change.status == BLE_ERR_PINKEY_MISSING) {
      struct ble_gap_conn_desc desc;
      rc = ble_gap_conn_find(event->repeat_pairing.conn_handle, &desc);
      assert(rc == 0);
      ble_store_util_delete_peer(&desc.peer_id_addr);
   }
}
```

Then put a retry loop in secureConnection()?
I'm not familiar enough with the NimBLE callback flow to know where the retry after delete should really happen.
I can see that RemoteDescriptor and RemoteCharacteristic include retries with secureConnection(), but if the user calls secureConnection() directly (I do this), they would need to retry themselves.

> > Nevertheless, I think a nice abstracted way to unpair manually (and hence delete the local keys) would be nice.
> 
> This should be done also. I had intended to add bond management API’s, just hasn’t been a priority yet.

Cool, I look forward to the day it appears.
What is already here has been a godsend for rapid development.

h2zero, 2020-08-19:
>Are you suggesting the following change in NimBLEClient::handleGapEvent():
```
case BLE_GAP_EVENT_ENC_CHANGE: {
   if (event->enc_change.status == BLE_ERR_PINKEY_MISSING) {
      struct ble_gap_conn_desc desc;
      rc = ble_gap_conn_find(event->repeat_pairing.conn_handle, &desc);
      assert(rc == 0);
      ble_store_util_delete_peer(&desc.peer_id_addr);
   }
}
```

>Then put a retry loop in secureConnection()?

This is certainly an option, the code as written would need to change  `event->repeat_pairing.conn_handle` to event->enc_change.conn_handle` but otherwise should work. This way would be ideal, since it would be automatic and hidden from the application.

 If you still have the capability to test and want to try that please let me know how it works out.

gkoh, 2020-08-20:
> This is certainly an option, the code as written would need to change `event->repeat_pairing.conn_handle` to `event->enc_change.conn_handle` but otherwise should work. This way would be ideal, since it would be automatic and hidden from the application.

Agreed.

> If you still have the capability to test and want to try that please let me know how it works out.

I do and will, let me find some time to cut and test, thanks.

closed 2020-08-21 by gkoh