h2zero/NimBLE-Arduino #118: Adding bluetooth mesh

h2zero, 2020-08-27:
Just wanted to bring attention to the new [mesh](https://github.com/h2zero/NimBLE-Arduino/tree/mesh) branch. 

I've been working on adding a mesh API. So far basic on/off server and level models have been implemented and working great with nRF Mesh.

This is very much a work in progress, many features still need to be implemented and **many** models to add.
 
If anyone would like to test it out and/or contribute please feel free to do so, I'd really appreciate it!

HelloDB, 2021-08-24:
Hi, I tried to compile in platformio and he failed
```cpp
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_speed = 921600

lib_ldf_mode = deep
lib_deps =
  ; h2zero/NimBLE-Arduino @ ^1.2.0
  NimBLE-Arduino=https://github.com/h2zero/NimBLE-Arduino/archive/refs/heads/mesh.zip
```
```
lib\NimBLE-Arduino-mesh\src\NimBLEMeshElement.cpp: In member function 'NimBLEMeshModel* NimBLEMeshElement::createModel(uint16_t, NimBLEMeshModelCallbacks*)':
lib\NimBLE-Arduino-mesh\src\NimBLEMeshElement.cpp:67:137: error: cannot convert 'bt_mesh_health_srv*' to 'const bt_mesh_model_cb* const' in initialization
             m_modelsVec.push_back(bt_mesh_model{{type},0,0,0,&pModel->m_opPub,{0},{0},bt_mesh_health_srv_op,&m_pHealthModel->m_healthSrv});
                                                                                                                                         ^
lib\NimBLE-Arduino-mesh\src\NimBLEMeshElement.cpp:75:104: error: cannot convert 'NimBLEMeshModel*' to 'const bt_mesh_model_cb* const' in initialization
     m_modelsVec.push_back(bt_mesh_model{{type},0,0,0, &pModel->m_opPub,{0},{0},pModel->m_opList, pModel});
                                                                                                        ^
lib\NimBLE-Arduino-mesh\src\NimBLEMeshNode.cpp: In member function 'bool NimBLEMeshNode::start()':
lib\NimBLE-Arduino-mesh\src\NimBLEMeshNode.cpp:164:125: error: cannot convert 'bt_mesh_cfg_srv*' to 'const bt_mesh_model_cb* const' in initialization
     m_configSrvModel = new bt_mesh_model{{BT_MESH_MODEL_ID_CFG_SRV},0,0,0,nullptr,{0},{0},bt_mesh_cfg_srv_op,&m_serverConfig};
                                                                                                                             ^
*** [.pio\build\esp32dev\libf19\NimBLE-Arduino-mesh\NimBLEMeshElement.cpp.o] Error 1
*** [.pio\build\esp32dev\libf19\NimBLE-Arduino-mesh\NimBLEMeshNode.cpp.o] Error 1
```
[image]

h2zero, 2021-08-24:
Hi, thanks for reporting. I haven't done much work on this branch lately. I'll have a look when time permits.

h2zero, 2021-09-06:
Sorry for the delay, the issue should be resolved now as I have just pushed the required changes.