h2zero/NimBLE-Arduino #119: Endianness in addresses (again)

elguiri, 2020-08-30:
Hi @h2zero 

Already mentioned here (https://github.com/h2zero/NimBLE-Arduino/issues/86), addresses are stored reversed from the normal ESP way of doing them.

So the constructors of NimBLEAddress do a reverse copy of byte arrays to store the internal NimBLE format.

I'm confused by the naming of the method getNative() though. I assumed this was going to give me an address in the "native" normal ESP way, but I guess the "native" refers to NimBLE and not ESP (as there is no reverse copy on the way out)!

This code...
```
    auto addressA = NimBLEAddress("01:02:03:04:05:06");
    auto addressB = NimBLEAddress((uint8_t *)addressA.getNative());
    Log.trace("A> %s B> %s", addressA.toString().c_str(), addressB.toString().c_str());
```

Produces this (probably correctly but confused me at first)...
```
    A> 01:02:03:04:05:06 B> 06:05:04:03:02:01
```

It might be worth clarifying in the documentation what "native" is referring to. For what it's worth, I would talk about an "internal" representation of NimBLE and a "native" representation of ESP but that ship might have sailed!

Thanks once again for the great library!

h2zero, 2020-08-30:
Hi @elguiri, I agree, this should be documented. 
The getNative naming was from adapting the old library and has stuck around since. That function is needed internally in the library to return internal NimBLE representation to make calls to the stack.   

I will make a note to document this and/or possibly refactor it.

I'm curious what your use case is for this function? In your example posted the issue can be eliminated by changing the second line to:
```
auto addressB = NimBLEAddress(addressA);
```
But I'm guessing you have some other reason to use getNative?

elguiri, 2020-08-30:
Hi @h2zero 

I discovered this when trying to _not_ hold a reference to advertisedDevice (a bad idea as you might imagine).

I am making multiple client connections to peripherals that match criteria (like a particular Appearance or Service). My code keeps a reference to the address of the peripheral when first advertised, so that a client connection can then be made, and it also allows me to determine which peripheral (an instance of my class) is associated with client callbacks. I decided to store the address in six bytes rather than the NimBLEAddress object, thinking this would be more efficient. So I used getNative to extract the six bytes from the advertisedDevice address, and then later compared it to the client peer NimBLEAddress using a temporary NimBLEAddress instantiation.

To make this work at the moment, I do a reverse copy. My actual code snippet (which stores the address in ESP format ready for a NimBLEAddress constructor) is...
```
            auto address = advertisedDevice->getAddress().getNative();
            std::reverse_copy(address, address + 6, peripheral->advertisedAddress);
```
And to compare to the client (during a callback)...
```
            if (NimBLEAddress(peripheral->advertisedAddress).equals(client->getPeerAddress())) {
                return peripheral;
            }
```
In fairness, this particular check is only needed for the onConnect callback because after that my peripheral object holds a direct reference to the client which can be used for determining the association.

Now... I have a separate "problem" which might be worth opening another issue for. I tried calling the client connect method using the stored address, rather than using the advertisedDevice object, and the connect failed with a code 13. If I swapped to the connect using the advertisedDevice object at the exact same place in the code, then all is fine (as you might expect). I'll do this test in the client sample to see if I can reproduce it there (rather than it being something I have complicated in my own code).

h2zero, 2020-08-30:
I see now, thanks for sharing that. 

I'd like to suggest that you should consider saving the `NimBLEAddress` instead of the raw bytes. There isn't any real efficiency gain, probably a loss in fact. 

This serves 2 purposes, 
1. Simpler code, 
2. It stores the address type as well, which will solve your connection error issue.

elguiri, 2020-08-30:
I might need to brush up on my memory allocation. I wanted to avoid creating a new NimBLEAddress every time I wanted to save one. I know the underlying data is just 7 bytes, but there isn't a way to set the address or type of an existing address (or do a copy into). So I'd have to have a pointer to the address from my peripheral, and create a new address every time I want to save one with my peripheral. Don't they then all end up on the heap?

Ideally I would want to allocate the storage for my address as part of my peripheral object (which I never dispose), and then copy the new address details into it as required. I might be barking up the wrong tree though! Perhaps a = operator is required?

h2zero, 2020-08-31:
Sounds like you're connecting to devices with random addresses? In that case maybe you could bond with them to get the id address. However I haven't implemented the code to get the id address when scanning yet.

Yes an assignment operator is a beneficial addition for sure. It's on the list.

elguiri, 2020-08-31:
Hi @h2zero 

You learn something new (!) everyday as they say. I've discovered the C++ [placement new operator](https://www.geeksforgeeks.org/placement-new-operator-cpp/), which removes my worry about a memory leak using NimBLEAddress. The _placement new operator_ allows object construction at a known address (in my case overwriting an existing NimBLEAddress). 

So, firstly my peripheral definition now includes a pointer to a NimBLEAddress and the allocated space for it...
```
NimBLEAddress *advertisedAddress = new NimBLEAddress("");
```
When I want to store the advertisedDevice address for my peripheral...
```
peripheral->advertisedAddress = new (peripheral->advertisedAddress) NimBLEAddress(advertisedDevice->getAddress());
```

By overwriting the existing advertisedAddress memory, there is no memory leak (and no need for an assignment operator either). And it allows for simpler code and stores the address type as rightly mentioned above.

As an aside, it's a bit clumsy to have to use a constructor with an empty string in order to reserve the space. If an empty string is a valid parameter then the constructor for NimBLEAddress may as well have a parameter-less version.

After all that, I don't need to use the getNative() method but it was an interesting excursion.

If you are using Visual Studio Code, you made need to set the C/C++ Extension IntelliSense Configurations _IntelliSense mode_ to gcc-x86 to stop an error appearing about the new operator when actually it will compile.

h2zero, 2020-08-31:
I'm glad you found a solution for this. The placement new operator is quite interesting, I've never used it before.

The class does need some work in regards to your comments. I'd still like to add an assignment operator as it will be useful to future additions. An empty constructor would also be a good idea.

Thanks for sharing this!

h2zero, 2020-08-31:
Hmm, on second thought the default copy assignment operator should work just fine here, for some reason I thought it didn't. 

You should just be able to declare an instance of NimBLEAddress in your class, instead of a pointer, and use 
```
peripheral->advertisedAddress =  advertisedDevice->getAddress();
```

elguiri, 2020-08-31:
@h2zero 

Thanks for that , I didn't realise the compiler creates a default assignment operator if there are no pointers in the class (and in the case of NimBLEAddress there is just the byte array and the byte). I've simplified my code again!

If you've put the possible changes to NimBLEAddress in a separate list then please close this out.

h2zero, 2020-09-01:
Thanks @elguiri, new constructor method and getNative documentation are on my todo list.

closed 2020-09-01 by elguiri

johntaves, 2021-11-12:
I am quite confused.

void func() {
  Serial.printf("Forget: %s %d\n",cellBLE.addrs[mp->val].toString().c_str(),mp->val);
  cellBLE.addrs[mp->val] = NimBLEAddress();
  Serial.printf("Forgot: %s\n",cellBLE.addrs[mp->val].toString().c_str());
  Serial.printf("WTF: %s\n",NimBLEAddress().toString().c_str());
}

Gives me:
Forget: e0:e2:e6:52:bc:c2 0
Forgot: 1a:b0:00:00:00:00
WTF: 1a:b0:00:00:00:00
(not exactly these values, but not the correct zeros)

So I changed to this:

NimBLEAddress emptyAddress = NimBLEAddress("00:00:00:00:00:00"); /* I did not try NimBLEAddress();
void func() {
  Serial.printf("Forget: %s %d\n",cellBLE.addrs[mp->val].toString().c_str(),mp->val);
  cellBLE.addrs[mp->val] = emptyAddress;
  Serial.printf("Forgot: %s\n",cellBLE.addrs[mp->val].toString().c_str());
  Serial.printf("WTF: %s\n",emptyAddress.toString().c_str());
}

And it works as expected.
Forget: e0:e2:e6:52:bc:c2 0
Forgot: 00:00:00:00:00:00
WTF: 00:00:00:00:00:00

I am confused.

h2zero, 2021-11-13:
It appears you are storing the memory address of the `NimBLEAddress()` function in the first snippet. The second snippet would be the correct implementation for this.

johntaves, 2021-11-13:
I agree that that might be happening, but I don't see why that is correct behavior from the compiler.

cellBLE.addrs[mp->val] is an array of NimBLEAddress not NimBLEAddress *

The folowing works fine:

```
        for (;i<cellBLE.numCells && cellBLE.addrs[i] != ad->getAddress()
           && cellBLE.addrs[i] != emptyAddress;i++) ;
...
          cellBLE.addrs[i] = ad->getAddress();

```
When i had NimBLEAddress() in place of emptyAddress in that comparison, it matched (but the address was no all 0). NimBLEAddress() was consistently providing the same nonzero MAC.

I think I also saw that when I did:
`cellBLE.addrs[mp->val] = NimBLEAddress("00:00:00:00:00:00");
`
I got the all zero MAC properly, but then it failed the above comparison.

I put printf's in the NimBLE code where the empty constructor calls the other constructor and does the memset(xxx,0,6) and the memset was executed and the toString() returned the empty string as expected. But that all zero value did not make it back to cellBLE.addrs[mp->val];