h2zero/NimBLE-Arduino #120: This library is now added to Arduino library manager.

h2zero, 2020-08-31:
If you prefer to use the Arduino IDE to manage libraries you can now find this repo in it.

Go to sketch -> Include Library -> Manage Libraries and search for NimBLE and install.

corneliusmunz, 2020-08-31:
@h2zero super! Thanks for your work! Has taken a long time till the library was added by the arduino team since your request but I am happy that it is now available via the library manager 👏🏻👍🏻🏅

elguiri, 2020-09-03:
@h2zero Great work making this available to the community. The library also appears in PlatformIO (search _NimBLE-Arduino_ in Libraries), also see here https://platformio.org/lib/search?query=NimBLE-Arduino

h2zero, 2020-09-03:
My pleasure.

I was wondering if it would pop up in platformio automatically or if I had to publish it. Nice to see it was automatic.

elguiri, 2020-09-03:
> I was wondering if it would pop up in platformio automatically

PlatformIO may lag behind. It's still showing v1.0.0 but I imagine that will update soon!

elguiri, 2020-09-04:
> PlatformIO may lag behind. It's still showing v1.0.0 but I imagine that will update soon!

Yep, now showing v1.0.1 automatically!

h2zero, 2020-09-04:
Nice, 1 less thing to do 😃

h2zero, 2020-12-29:
Closing as stale.

closed 2020-12-29 by h2zero