h2zero/NimBLE-Arduino #121: Occasional delay (blocking) after calling advertising start

elguiri, 2020-09-01:
This is annoying me because it's not happening consistently (and when I turn on debug messages it never happens)!

After setting up my server, I am starting to advertise as follows...
```
    NimBLEAdvertising *advertising = NimBLEDevice::getAdvertising();
    advertising->setScanResponse(false);
    advertising->start();
```

This is the last thing I do in the setup() method before loop() takes over. Occasionally, there is an (almost exact) 10000 ms delay before the code in my loop() method starts executing. If it works "correctly", the advertising start takes less than 10 ms.

Any ideas what might be the "random" cause? Thanks!

h2zero, 2020-09-01:
Yes I experience the same when using arduino. The cause is the controller used in 1.0.4 and master. 

I've tested using 1.0.2 and the issue goes away. Not sure if there is anything that can be done to compensate but I have replaced the controller with a newer commit in my copy of esp32-arduino and seems to solve it.

elguiri, 2020-09-01:
@h2zero 

I'm using platformio in Visual Studio Code (and have the default files for my ESP32 module). Is there an easy way for me to do the change you suggest?

Cheers

h2zero, 2020-09-01:
@elguiri I don't know platformio well enough to answer that. 

However I do think I have discovered a way to deal with the issue and currently investigating the best way to implement it. 

I should be able to push a fix later tonight.

h2zero, 2020-09-02:
@elguiri Here is a patch I'm testing. If you could apply it and let me know if it works for you I'd appreciate it.
[patch_adv.patch.txt](https://github.com/h2zero/NimBLE-Arduino/files/5160103/patch_adv.patch.txt)

elguiri, 2020-09-02:
@h2zero 

I've applied the patch and will "see how it goes"; as noted the 10 second delay does not always occur.

I'd completely overlooked the NimBLEServer::start() method, as the sample server code doesn't use it. Now I understand why, it's because a call to advertising start will start the server anyway.

That means the server start method is useful when you want to start the server _without_ advertising. And that suggests it's not a good idea to put the call to ble_hs_sched_reset(BLE_HS_EDONE) at the end of server start (because then you'll get advertising starting whether you want it or not)?

Would it be better to put the call to ble_hs_sched_reset(BLE_HS_EDONE) in NimBLEAdvertising.cpp after the call to pServer->start()?

h2zero, 2020-09-02:
Thanks @elguiri, yes I had the same concerns with that patch. I need a better way to reset the controller for the first call to start advertising. Obviously we don't want to do the every time and server start was the logical place.

The host reset callback restarts scanning and advertising which is why you see that behavior. A conditional could be applied there to only start scanning if isScanning == true.

elguiri, 2020-09-02:
> The host reset callback restarts scanning and advertising which is why you see that behavior. A conditional could be applied there to only start scanning if isScanning == true.

Perhaps that wasn't clear from my post above. isScanning() didn't seem to help, because it reported false when the scan was already starting up.

h2zero, 2020-09-02:
@elguiri Yes, that gets reset when the host resets and returns incorrect because the scan hasn't stopped yet when we force the reset. 

But this may now not be needed, I believe I have figured out a better way to wake up the controller that does not involve the reset. 

Here is a new patch, if you would like to test it.
[adv_patch2.patch.txt](https://github.com/h2zero/NimBLE-Arduino/files/5163661/adv_patch2.patch.txt)

elguiri, 2020-09-02:
IN THE MIDDLE OF WRITING THIS WHEN YOU POSTED ABOVE!

Upon further investigation, isScanning() is failing because of a multi-threading timing issue.

The call to `ble_hs_sched_reset(BLE_HS_EDONE)` triggers the `NimBLEDevice::onSync` callback, and it's in that callback that the scan is _restarted_ even if it has never been started (advertising is also restarted there)! This can be seen in the log...
```
I NimBLEDevice: "NimBle host synced."
D NimBLEScan: ">> start(duration=-16843010)"
D NimBLEScan: "<< start()"
D NimBLEAdvertising: ">> Advertising start: customAdvData: 0, customScanResponseData: 0"
D NimBLEAdvertising: "<< Advertising start"
```
The scan is assumed to have started if the m_pScan is pointing at the scan object, but that singleton object gets created just to set the scan parameters and not necessarily to start it. It's quite a mess to assume restart like that, because the duration has not been set (-16843010 is random) neither the callback at that point.

elguiri, 2020-09-02:
> Here is a new patch, if you would like to test it.

Great thanks, having a look now!

h2zero, 2020-09-02:
> The scan is assumed to have started if the m_pScan is pointing at the scan object, but that singleton object gets created just to set the scan parameters and not necessarily to start it. It's quite [...]

Yes that code is incomplete, Resets are rare occurrences with multiple reasons for being triggered so I just put in some code to hopefully keep things working if they happen. I will definitely need to revisit the logic in that handler.

elguiri, 2020-09-02:
@h2zero 

Patch tests good (and unlikely to break anything else due to the code position and the function limited to advertising). onSync() handler on your todo list (sorry to keep adding to it)!

h2zero, 2020-09-02:
Great, thanks. I'll push the fix shortly.

Yes the todo list is getting longer but that's a good thing.

h2zero, 2020-09-02:
Pushed commit ae3be89 which fixes this issue.

h2zero, 2020-09-04:
Reopening this.

The patch that fixed this issue was a bandaid solution. 

I discovered the same issue while testing mesh with arduino and I believe I have found the root cause.
Arduino runs on core 1 while the stack runs on core 0. What happens is when we call start advertising we execute on core 1 but the stack needs to wait for a semaphore to be released from the controller callback running on core 0. The problem seems to be that commands sent to the controller can happen too fast and occasionally the callback gets delayed.

To test I created a task for all BLE functions pinned to core 0 and deleted the arduino loops task. 
The result is no delays of any kind and actually much better response overall.

Now the question is how to deal with it properly. The next arduino core release give the option of what core to run on but I'd prefer not to rely on that.

I'm contemplating using the IPC calls internally in the library or creating tasks that trigger once and delete if that doesn't work.

Any other suggestions would be most welcomed.

elguiri, 2020-09-05:
Hi @h2zero 

> The problem seems to be that commands sent to the controller can happen too fast and occasionally the callback gets delayed.

Do you think this is a race condition? Is the controller code imperfect and _any_ fix in NimBLE-Arduino is going to be a workaround? Does this warrant a closer look at the controller code?

> The next arduino core release give the option of what core to run on but I'd prefer not to rely on that.

It's clear there are multitasking issues to be addressed. Making everything run on one core could be detrimental to performance (especially if using WiFi as well for example). Whilst solving some problems it wouldn't guarantee thread-safe operation anyway.

> I'm contemplating using the IPC calls

I presume you are talking about [these IPC functions](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ipc.html), so the approach still needs some care avoiding a possible deadlock and managing stack usage.

h2zero, 2020-09-05:
> Do you think this is a race condition? Is the controller code imperfect and any fix in NimBLE-Arduino is going to be a workaround? Does this warrant a closer look at the controller code?

The problem occurs at [this line](https://github.com/h2zero/NimBLE-Arduino/blob/905baa97cab31e781b2496d75611f09fafbd3777/src/esp-hci/src/esp_nimble_hci.c#L94) after the semaphore is taken it seem's to not get released right away, which happens in the callback from the controller [here.](https://github.com/h2zero/NimBLE-Arduino/blob/905baa97cab31e781b2496d75611f09fafbd3777/src/esp-hci/src/esp_nimble_hci.c#L327) Looks like some sort of delay in the inter-processor communication. This only happens when sending commands to the controller in a tight loop, that's why we don't see the delay with debugging turned on, it slows things down enough. I wish we could look at the controller code but espressif has kept that as a closed source library.

I have also tested by removing the semaphore and looping on [!esp_vhci_host_check_send_available()](https://github.com/h2zero/NimBLE-Arduino/blob/905baa97cab31e781b2496d75611f09fafbd3777/src/esp-hci/src/esp_nimble_hci.c#L90) until true and the issue is also resolved, clearly not a great solution but the max loops counted on that was 15, so not too terrible either. 

> I presume you are talking about these IPC functions, so the approach still needs some care avoiding a possible deadlock and managing stack usage.

Yes those are the calls, I haven't used them before but I'll test them for interest sake.

elguiri, 2020-09-05:
@h2zero 

If the controller code is closed then whatever is done is going to be a fudge, trying not to trip the semaphore release delay. I wouldn't remove the semaphore as it is providing mutex on access to the hardware send.

Hopefully the IPC or other thread marshalling method will work, especially if it helps other things not just advertising start!

h2zero, 2020-09-05:
Agreed, not much we can do.

However I think the ipc call will work if I use it on `esp_vhci_host_send_packet`

The every controller command can be called on the core the controller runs on without affecting anything else.

I'll experiment when I get time.

h2zero, 2020-09-06:
Did some experimenting and it was successful!

Dead simple to implement and seems to be working well. Needs serious testing however to verify no bugs have been introduced.

Here is a patch for anyone interested in helping with the testing (also a bit of performance improvement).

[hci_patch.txt](https://github.com/h2zero/NimBLE-Arduino/files/5179050/hci_patch.txt)

elguiri, 2020-09-06:
Fantastic @h2zero

I'll test it (taking out the `ble_gap_adv_stop()` fudge on the advertising start)!

elguiri, 2020-09-06:
@h2zero 

Tests ok (no obvious problems although I'm not doing any performance testing). Code change looks really neat, with the refactoring to make use of the single arg call to the IPC method. I can't how it could cause a problem, it just elevates the priority of the call to the controller and runs it on the right core.

I'd be tempted to commit it to master (and revert the advertising code change)!

h2zero, 2020-09-06:
@elguiri I have found no issues with it either, but you never know. However I wonder if there should be some conditional applied to not use the IPC call if the code is already executing on core 0. I'll try to test that and see if it improves or degrades performance. I will commit it when I complete further testing.

elguiri, 2020-09-06:
@h2zero 

I think there are a couple of issues when IPC is called when not needed (already on core 0)...

1) _The IPC API is protected by a mutex hence simultaneous IPC calls are not possible_ (from the documentation). There's already a semaphore for access to the controller so this mutex could cause an unnecessary complication (you don't know what other code is using the IPC API at the same time).

2) Depending on the priority (not documented) of the high priority tasks mentioned, it may be a demotion from the default for BLE tasks (could that be priority 21)?

It will be interesting to see the results of your testing on that.

h2zero, 2020-09-06:
@elguiri My thoughts exactly, IPC should only be used if necessary. I have completed the changes and testing shows most calls do come from core 0 already. There was a bit of a performance improvement and I don't see any issues arising from this change.

Here is the updated patch  (also removes the advertising fudge)

[hci_patch2.txt](https://github.com/h2zero/NimBLE-Arduino/files/5180170/hci_patch2.txt)

h2zero, 2020-09-06:
>Depending on the priority (not documented) of the high priority tasks mentioned, it may be a demotion from the default for BLE tasks (could that be priority 21)?

Just looked this up and the priority is 24 (configMAX_PRIORITIES -1)

elguiri, 2020-09-07:
@h2zero I've applied and tested the new patch, all looks good so far thanks!

h2zero, 2020-09-07:
Thanks @elguiri I'm happy with it as well. Will commit to master shortly.

h2zero, 2020-09-07:
Going to mark this as resolved with commit 54c5c31.

closed 2020-09-07 by h2zero