h2zero/NimBLE-Arduino #122: Use std::function in typedef of callback functions to

corneliusmunz, 2020-09-01:
I am using your library to connect to multiple BLE devices (LEGO hubs) at the same time. With your support we have managed to connect up to 9 devices. 

I have now another issue when connecting to multiple remote characteristics of several devices. **With the current implementation of the callback function i can only use a static function and not a member function as a callback**. This leads to the problem, that every connected device use the same static callback function and i can not preserve state or handle the notifications of different devices in a member function of a class. 

This problem with callbacks is described in several stackoverflow posts:
https://stackoverflow.com/questions/400257/how-can-i-pass-a-class-member-function-as-a-callback#402385
https://stackoverflow.com/questions/14189440/c-callback-using-class-member/14189561#14189561
https://cognitivewaves.wordpress.com/callbacks/

**One solution is to use the std::bind and std::function functionality which was introduced in C++11**

I have tested the following changes in the NimBLERemoteCharacteristic.h file:

1. Add` #include <functional>` to be able to use the std::function functionallity
2. Change the typedef of the callback as follows
`typedef std::function<void (NimBLERemoteCharacteristic* pBLERemoteCharacteristic,
                                uint8_t* pData, size_t length, bool isNotify)> notify_callback;`

With that changes i could use the "old" way of defining the callback function without any error
`_pRemoteCharacteristic->registerForNotify(notifyCallback); // nofifyCallback is a pointer to a static function
`
and i can use now with the `std::bind` functionality the "new" way of defining the callback function which enables me to use a member function instead of a static function
`_pRemoteCharacteristic->registerForNotify(std::bind(&Lpf2Hub::notifyCallback, this, _1, _2, _3, _4)); // notifyCallback is now a member function`

Maybe you can check this improvement and add it to one of your next releases. I think its a "transparent" change where the "old" static callback functions will work without changes in the client code which uses your library. I can also state a pull request with my changes that you can check it directly.

I know that the `registerForNotify` method is marked as deprecated an i should replace it with the `subscribe` method but in principal this callback thing is also valid for the new `subscribe` method

@giancann, @lucadentella: With this issue solved, the legoino issue (https://github.com/corneliusmunz/legoino/issues/19) could also be solved but it is a prerequisite to solve it in the NimBLE library

@h2zero : Thanks in advance to have a look on that issue!

h2zero, 2020-09-01:
@corneliusmunz Thanks for bringing this up. Quite a coincidence that I was just looking into the same thing for use in the mesh extension. C++ callbacks are certainly a pain point.

Also thank you for the PR, I will review it soon.

h2zero, 2020-09-02:
Merged #123

elguiri, 2020-09-03:
@corneliusmunz Great timing, thanks for this as well, I'm going to use it in my code right now!

corneliusmunz, 2020-09-03:
Issue is resolved with #123  and release version https://github.com/h2zero/NimBLE-Arduino/releases/tag/1.0.1

elguiri, 2020-09-03:
@corneliusmunz 

Hi - this is going to be really useful but I was head scratching trying to make use of it. _1, _2 etc weren't compiling. Did you put `using namespace std::placeholders` in your code somewhere?

corneliusmunz, 2020-09-03:
@elguiri yes you are right. I have added the `using namespace std::placeholders;` using to my header file and then use the following syntax for setting the callback for notifications: 
`_pRemoteCharacteristic->registerForNotify(std::bind(&Lpf2Hub::notifyCallback, this, _1, _2, _3, _4));`
This should look quite similar in your use case because the function has always 4 parameters and for the member function an additional (normally hidden) parameter (the instance of the class where the member function is located) is used. So it should always look like the following: `<remoteCharacteristicInstance>->registerForNotify(std::bind(&<ClassName>::<memberFunctionCallbackName>, this, _1, _2, _3, _4));` where you can replace `<remoteCharacteristicInstance>`, `<ClassName>`, `<memberFunctionCallbackName>` with the names of your implementation.

You can find my implementation here: 
https://github.com/corneliusmunz/legoino/blob/da277284de1e10144cef5294c159babe3fe473e2/src/Lpf2Hub.cpp#L955
And the header file with the addtional using of std::placeholders:
https://github.com/corneliusmunz/legoino/blob/da277284de1e10144cef5294c159babe3fe473e2/src/Lpf2Hub.h#L15

h2zero, 2020-09-03:
Great description @corneliusmunz I'd like to add this to the documentation.

elguiri, 2020-09-04:
@corneliusmunz Thanks for that, it was the _1 syntax that was confusing me and stackoverflow helped with a code sample (https://stackoverflow.com/a/47946395) with an explicit mention of the placeholders namespace (and the realisation you might be _using_ that namespace).

The new syntax allows the callback to be created in advance of use, so my preferred approach is...
```
using namespace std::placeholders;
notify_callback callback = std::bind(&<ClassName>::<memberFunctionCallbackName>, this, _1, _2, _3, _4);
<remoteCharacteristicInstance>->subscribe(true, callback); // true for notifications, false for indications
```

That makes sure the callback is of the right type, and might help in the case of compilation issues. The include of the functional header is not required explicitly as it is pulled in with the NimBLEDevice header.

corneliusmunz, 2020-09-04:
@elguiri Very good hint with the not needed include of the functional header and the idea of defining the callback in advance. I will change my code in my library accordingly. 
@h2zero I would suggest to use @elguiri 's code snipped for documentation because of the use of the `subscribe` method instead of the deprecated `registerForNotify` method and the creation of the callback pointer in advance. Maybe you can add two examples in the documentation. One for the "old" way of using a callback with a function pointer to a static method and one with the example from @elguiri above

h2zero, 2020-09-04:
Reopening this for exposure so others will see it if interested. Will close again when documented.

Thanks @corneliusmunz and @elguiri for the detailed descriptions.

elguiri, 2020-09-05:
> Reopening this

@h2zero Looks to still be closed?

h2zero, 2020-09-05:
> > Reopening this
> 
> 
> 
> @h2zero Looks to still be closed?

Oops, got ahead of myself.

h2zero, 2021-01-18:
Finally documented this 😅  closing now.

closed 2021-01-18 by h2zero