h2zero/NimBLE-Arduino #124: NimBLEAdvertising: Custom advertising behaviour is a bit different

samscam, 2020-09-03:
I've noticed a difference in behaviour from the kolban/bluedroid library around custom advertising. In the old library if you either used `setAdvertisementData()` or `setScanResponseData()` but not both, the library would still try to populate the other half with things set at the BLEAdvertising level (like service uuids and name).

To be honest the old behaviour was confusing and a bit unreliable (which was part of what led me here!)

Could we document something along the lines of:

> Using `setAdvertisementData()` will entirely replace any other parameters that have been set at the `NimBLEAdvertising` level including uuids, name, appearance, etc. Set all the things you wish to advertise within the `NimBLEAdvertisementData` instead.
> 
> Calling `setScanResponseData()` on its own will have no effect - data passed in that way is only used if `setAdvertisementData()` has also been called.

Happy to turn that into a PR - perhaps with some examples for better clarity.

A harsher take (which would break compatibility) would be to say that these things can only be set in one place - either deprecating the `NimBLEAdvertisementData` class and moving the equivalent functionality into `NimBLEAdvertising` or the other way round.

h2zero, 2020-09-03:
Hi @samscam thank you for pointing out this difference, I didn't realize there was any different behavior.

The way that these were intended to work is that using setAdvertising/scanResponseData would override the default advertisement which does the shuffling between the two automatically. So these are a more "advanced" use option. 

Now that you bring it up I question wether it's being handled correctly. Something I need to look at again.

I would appreciate any PR you would like to make on this or any other matter, documentation and or code changes are most welcome.

h2zero, 2020-09-11:
Had a look at this today and I (as you) like the way things are functioning currently. Documentation is certainly needed though.

h2zero, 2020-12-29:
Closing as the documentation was completed.

closed 2020-12-29 by h2zero