h2zero/NimBLE-Arduino #125: Strange behaviour (hangs server) on certain BLE device

Calypso971, 2020-09-08:
A few days now I'm looking into a problem connecting a device which I think maybe has a limited BLE stack. Connecting and getting info with BLE Scanner on a Android phone works though.

In version 1.0.1 it failed constantly. When connecting (yes connecting) to the device, the device BLE hung itself, because after trying to connect with NimBLE and disconnecting the ESP32 from power the phone still sees the device, but cannot connect to it/get information from it.

The changes made until today in NimBLE have a different result:

* Quite often the connection didn't succeed [Using the refactored BLE examples]:
16:09:31.848 -> GAP procedure initiated: connect; peer_addr_type=0 peer_addr=c4:64:e3:92:13:7c scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
16:09:31.848 -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000d len=25
16:09:31.848 -> 0x0d 0x20 0x19 0x10 0x00 0x10 0x00 0x00 0x00 0x7c 0x13 0x92 0xe3 0x64 0xc4 0x00 0x18 0x00 0x28 0x00 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x00 
16:09:31.848 -> Command Status: status=0 cmd_pkts=5 ocf=0xd ogf=0x8
16:09:31.950 -> LE connection complete. handle=0 role=0 paddrtype=0 addr=c4.64.e3.92.13.7c local_rpa=0.0.0.0.0.0 peer_rpa=0.0.0.0.0.0 itvl=40 latency=0 spvn_tmo=256 mca=0
16:09:31.950 -> D NimBLEClient: "Got Client event "
16:09:31.950 -> D NimBLEClient: "Connection established"
16:09:31.950 -> GATT procedure initiated: exchange mtu
16:09:31.950 -> host tx hci data; handle=0 length=7
16:09:31.950 -> ble_hs_hci_acl_tx(): 0x00 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x02 0x00 0x01 
16:09:31.950 -> txed att command: mtu req; conn=0 mtu=256
16:09:31.950 -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
16:09:31.950 -> 0x16 0x20 0x02 0x00 0x00 
16:09:31.950 -> Command Status: status=0 cmd_pkts=5 ocf=0x16 ogf=0x8
16:09:32.224 -> Number of Completed Packets: num_handles=1
16:09:32.224 -> handle:0 pkts:1
16:09:32.224 -> Disconnection Complete: status=0 handle=0 reason=62
16:09:32.224 -> D NimBLEClient: "Got Client event "
16:09:32.224 -> I NimBLEClient: "disconnect; reason=574, "
16:09:32.224 -> onDisconnect

Sometimes, the connection succeed. However, getting a service or doing a discovery with discover attributes then gets an error:

16:10:00.922 -> GATT procedure initiated: exchange mtu
16:10:00.922 -> host tx hci data; handle=0 length=7
16:10:00.922 -> ble_hs_hci_acl_tx(): 0x00 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x02 0x00 0x01 
16:10:00.922 -> txed att command: mtu req; conn=0 mtu=256
16:10:00.922 -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
16:10:00.922 -> 0x16 0x20 0x02 0x00 0x00 
16:10:00.922 -> Command Status: status=0 cmd_pkts=5 ocf=0x16 ogf=0x8
16:10:00.989 -> Number of Completed Packets: num_handles=1
16:10:00.989 -> handle:0 pkts:1
16:10:01.092 -> LE Remote Used Features. handle=0 feat=fd 00 00 00 00 00 00 00 
16:10:01.092 -> ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=7 data=0x03 0x00 0x04 0x00 0x03 0xf7 0x00 
16:10:01.092 -> rxed att command: mtu rsp; conn=0 mtu=247
16:10:01.092 -> D NimBLEClient: "Got Client event "
16:10:01.092 -> I NimBLEClient: "mtu update event; conn_handle=0 mtu=247"
16:10:01.092 -> D NimBLEClient: ">> deleteServices"
16:10:01.092 -> D NimBLEClient: "<< deleteServices"
16:10:01.092 -> onConnect
16:10:01.092 -> D NimBLEClient: "<< connect()"
16:10:01.092 ->  - Connected to server
16:10:01.092 -> D NimBLEClient: ">> getService: uuid: 0x180a"
16:10:01.127 -> D NimBLEClient: ">> retrieveServices"
16:10:01.127 -> GATT procedure initiated: discover service by uuid; uuid=0x180a
16:10:01.127 -> txed att command: find type value req; conn=0 start_handle=0x0001 end_handle=0xffff attr_type=10240
16:10:01.127 -> host tx hci data; handle=0 length=13
16:10:01.127 -> ble_hs_hci_acl_tx(): 0x00 0x00 0x0d 0x00 0x09 0x00 0x04 0x00 0x06 0x01 0x00 0xff 0xff 0x00 0x28 0x0a 0x18 
16:10:01.160 -> Number of Completed Packets: num_handles=1
16:10:01.160 -> handle:0 pkts:1
16:10:01.329 -> ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x01 0x06 0x01 0x00 0x11 
16:10:01.329 -> rxed att command: error rsp; conn=0 req_op=6 handle=0x0001 error_code=17
16:10:01.329 -> D NimBLEClient: "Service Discovered >> status: 273 handle: -1"
16:10:01.329 -> E NimBLEClient: "characteristicDiscCB() rc=273 "
16:10:01.329 -> D NimBLEClient: "<< << Service Discovered"
16:10:01.329 -> E NimBLEClient: "Could not retrieve services"
16:10:01.329 -> D NimBLEClient: "<< getService: not found"

But I know for sure that this service exists and on other devices it's working without a problem. Same happens with the BLE_Client example code (this time with the discover attribute function - and again, the codes works without a problem with other devices):

16:31:46.172 -> D NimBLEClient: ">> retrieveServices"
16:31:46.172 -> GATT procedure initiated: discover all services
16:31:46.172 -> txed att command: read group type req; conn=0 start_handle=0x0001 end_handle=0xffff
16:31:46.172 -> LE Connection Update Complete. handle=0 itvl=120 latency=0 timeout=60
16:31:46.172 -> host tx hci data; handle=0 length=11
16:31:46.172 -> ble_hs_hci_acl_tx(): 0x00 0x00 0x0b 0x00 0x07 0x00 0x04 0x00 0x10 0x01 0x00 0xff 0xff 0x00 0x28 
16:31:46.172 -> D NimBLEClient: "Got Client event "
16:31:46.172 -> I NimBLEClient: "Connection parameters updated."
16:31:46.276 -> Number of Completed Packets: num_handles=1
16:31:46.276 -> handle:0 pkts:1
16:31:46.483 -> ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x01 0x10 0x01 0x00 0x11 
16:31:46.483 -> rxed att command: error rsp; conn=0 req_op=16 handle=0x0001 error_code=17
16:31:46.483 -> D NimBLEClient: "Service Discovered >> status: 273 handle: -1"
16:31:46.483 -> E NimBLEClient: "characteristicDiscCB() rc=273 "
16:31:46.483 -> D NimBLEClient: "<< << Service Discovered"
16:31:46.483 -> E NimBLEClient: "Could not retrieve services"
16:31:46.483 -> E NimBLEClient: "Error: Failed to get services"

The funny thing is though that in both cases, with the latest changes in code I can still connect from my phone to the device after the errors occur (although with the first one it's logical because the connection wasn't finalized).

Normally I would suspect the device implementation of BLE, but since the BLE Scanner on my phone doesn't seem to have a problem with it, and the behaviour changed (BLE on the serverdevice not hanging anymore) with the latest changes, I'm thinking it may be caused by something in NimBLE...

h2zero, 2020-09-08:
Hi @Calypso971 Thank you for the detail and logs (especially with the NimBLE stack logging). 

What I see here is that in the first log, before the recent code changes, the error was `BLE_ERR_CONN_ESTABLISHMENT`, which was probably a symptom that was fixed in that change. 

After the code changes you now get `BLE_ATT_ERR_INSUFFICIENT_RES`. Which is the server telling you it can't process the request for some reason. The question is what is that reason? 

I suspect it could be this:
`txed att command: read group type req; conn=0 start_handle=0x0001 end_handle=0xffff`

The server may not be able to process this request as it's asking for the whole database of attributes.
To test this you could try changing [the 0xffff on this line](https://github.com/h2zero/NimBLE-Arduino/blob/5bc570a6c2a9dd4c1e045d3a58534395ccb44fd4/src/nimble/host/src/ble_gattc.c#L1611) to a smaller value, maybe 0x0fff, and check the result.

It's not a solution to the problem, as that will prevent retrieving the full attribute set, but it would provide a clue as to the next step.

Calypso971, 2020-09-09:
@h2zero: Thanks for your suggestion. 

First of all: the first debuglines are still with the latest version, only on one of those disconnect sessions. But I can work around those with a retry I guess, so there should be a workaround for that. First thought that maybe connectivity could be an issue, but the devices are next to eachother, the phone I use for testing is much further away, and the RSSI reported is good.

Tried your suggestion of using 0x0fff, doesn't have a result (also tried to change it on line 1385) but that didn't give a result; tried it with discovery, service 0x1800, 0x180a and (this one) with a custom service that is present on the device (and can be accessed by BLE Scanner on the phone): 

08:17:58.183 -> D NimBLEClient: ">> getService: uuid: f000ffc0-0451-4000-b000-000000000000"
08:17:58.217 -> D NimBLEClient: ">> retrieveServices"
08:17:58.217 -> GATT procedure initiated: discover service by uuid; uuid=f000ffc0-0451-4000-b000-000000000000
08:17:58.217 -> txed att command: find type value req; conn=0 start_handle=0x0001 end_handle=0x0fff attr_type=10240
08:17:58.217 -> host tx hci data; handle=0 length=27
08:17:58.217 -> ble_hs_hci_acl_tx(): 0x00 0x00 0x1b 0x00 0x17 0x00 0x04 0x00 0x06 0x01 0x00 0xff 0x0f 0x00 0x28 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xb0 0x00 0x40 0x51 0x04 0xc0 0xff 0x00 0xf0 
08:17:58.251 -> Number of Completed Packets: num_handles=1
08:17:58.251 -> handle:0 pkts:1
08:17:58.320 -> ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x01 0x06 0x01 0x00 0x11 
08:17:58.320 -> rxed att command: error rsp; conn=0 req_op=6 handle=0x0001 error_code=17
08:17:58.320 -> D NimBLEClient: "Service Discovered >> status: 273 handle: -1"
08:17:58.320 -> E NimBLEClient: "characteristicDiscCB() rc=273 "
08:17:58.320 -> D NimBLEClient: "<< << Service Discovered"
08:17:58.320 -> E NimBLEClient: "Could not retrieve services"
08:17:58.320 -> D NimBLEClient: "<< getService: not found"
08:17:58.320 -> We are now connected to the BLE Server.

When I'm correct, the error returned (17) is 0x11 Unsupported Feature or Parameter Value? Basically, since it's also doing a discovery when just accessing a single service, the problem could be in the discovery call, as you are suggesting by changing the 0xffff value. But why is a BLE Scanner application on the phone then working, when that one is totally dependent of the discovery...

h2zero, 2020-09-09:
@Calypso971 I'm not sure what could be causing this issue, all I can see in the log is the response to the discovery command is error code 0x11. That code is an ATT response code, which NimBLE reports as 0x0111 (273) Aka: `BLE_ATT_ERR_INSUFFICIENT_RES`.

The peripheral is saying it's low on resources and cannot process the request, but what resources? 
There are a couple of possiblities:
- Too busy, we may need to delay or change connection parameters
- Not enough memory to handle the request, not much we can do other than wait and hope it clears 

A couple questions, what is this device you're connecting to?
Could your phone possibly be connected still while you're connecting with the esp32 (might explain the error code)?

Calypso971, 2020-09-11:
> The peripheral is saying it's low on resources and cannot process the request, but what resources?
> There are a couple of possiblities:
> 
>     * Too busy, we may need to delay or change connection parameters

Did quite some testing with those, but it didn't succeed :-(. Because it annoys me that I can't get this to work, I did a BT Snoop on my phone and examined it with Wireshark. Connection parameters of my phone are strange to say the least:
min/max are 6 (7.5 ms) and Supervision delay is 500. While in your example code you mention that the min/max shouldn't be lower as 24... Ofcourse trying these parameters still don't have the wanted result.

>     * Not enough memory to handle the request, not much we can do other than wait and hope it clears
The funny thing of this is, is that a query by phone doesn't have the problem. And the discovery call is almost the same:

Phone: 02 00 0b 00 07 00 04 00 10 01 00 ff ff 00 28
ESP32: 00 00 0b 00 07 00 04 00 10 01 00 ff ff 00 28

The first byte is the handle, so that is the only change and not strange. On the Phone results are returned, so I think the BLE call (TX) is the same, the memory usage on the same device should be too.

> A couple questions, what is this device you're connecting to?

Texas Instruments based uPC board with a Laird SaBLE-x-R2 bluetooth module

> Could your phone possibly be connected still while you're connecting with the esp32 (might explain the error code)?

Not with the phone, and also the module is programmed in such a way to only accept one connection. So if you're connected (and I am with the ESP32) it doesn't accept connection from another device.

I've also tried to do something "dirty" and request a non-existent UUID. Because this should also trigger a discovery, I'd expect to get the dreaded error also. But it didn't:

```
15:37:54.483 -> ble_hs_hci_acl_tx(): 0x00 0x00 0x0d 0x00 0x09 0x00 0x04 0x00 0x06 0x01 0x00 0xff 0xff 0x00 0x28 0x0a 0x3f 
15:37:54.515 -> Number of Completed Packets: num_handles=1
15:37:54.515 -> handle:0 pkts:1
15:37:54.585 -> ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=0x05 0x00 0x04 0x00 0x01 0x06 0x01 0x00 0x0a 
15:37:54.585 -> rxed att command: error rsp; conn=0 req_op=6 handle=0x0001 error_code=10

```
However as seen in one of the previous scans, requesting an existing UUID directly also returns the error 17. This also points into a memory issue. My first reaction also was: it's the problem of the device I'm connecting to. However, that doesn't explain why connecting with a phone and BLE Scanner just works. Also other BLE scanning applications just work.

If helpful (to view the comm flow that works) I can mail the Snoopfile from the phone - maybe there's something in the communication before the actual request of the discovery that gives a hunch of what's going on compared to the connection information from the ESP:

```
15:45:42.266 -> GAP procedure initiated: connect; peer_addr_type=0 peer_addr=c4:64:e3:92:13:7c scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
[15:45:42.300] -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000d len=25
15:45:42.300 -> 0x0d 0x20 0x19 0x10 0x00 0x10 0x00 0x00 0x00 0x7c 0x13 0x92 0xe3 0x64 0xc4 0x00 0x18 0x00 0x28 0x00 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x00 
15:45:42.300 -> Command Status: status=0 cmd_pkts=5 ocf=0xd ogf=0x8
15:45:42.676 -> LE connection complete. handle=0 role=0 paddrtype=0 addr=c4.64.e3.92.13.7c local_rpa=0.0.0.0.0.0 peer_rpa=0.0.0.0.0.0 itvl=40 latency=0 spvn_tmo=256 mca=0
15:45:42.676 -> D NimBLEClient: "Got Client event "
15:45:42.676 -> D NimBLEClient: "Connection established"
15:45:42.676 -> GATT procedure initiated: exchange mtu
15:45:42.676 -> host tx hci data; handle=0 length=7
15:45:42.676 -> ble_hs_hci_acl_tx(): 0x00 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x02 0x00 0x01 
15:45:42.676 -> txed att command: mtu req; conn=0 mtu=256
15:45:42.676 -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
15:45:42.676 -> 0x16 0x20 0x02 0x00 0x00 
15:45:42.676 -> Command Status: status=0 cmd_pkts=5 ocf=0x16 ogf=0x8
15:45:42.710 -> Number of Completed Packets: num_handles=1
15:45:42.710 -> handle:0 pkts:1
15:45:42.779 -> LE Remote Used Features. handle=0 feat=fd 00 00 00 00 00 00 00 
15:45:42.779 -> ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=7 data=0x03 0x00 0x04 0x00 0x03 0xf7 0x00 
15:45:42.779 -> rxed att command: mtu rsp; conn=0 mtu=247
15:45:42.779 -> D NimBLEClient: "Got Client event "
15:45:42.779 -> I NimBLEClient: "mtu update event; conn_handle=0 mtu=247"
15:45:42.779 -> D NimBLEClient: ">> deleteServices"
15:45:42.779 -> D NimBLEClient: "<< deleteServices"
15:45:42.779 -> D NimBLEClientCallbacks: "onConnect: default"
15:45:42.779 -> D NimBLEClient: "<< connect()"
```

h2zero, 2020-09-11:
Thank you for the details @Calypso971. Since this works with the phone and we can rule out, for now, a resource constraint on the peripheral, maybe the problem is a resource issue on the esp32. 

There are a couple settings we can try in order to verify that possibility.
In `nimconfig.h` the following values can be adjusted to increase buffers:
```
/** @brief ACL Buffer count */
#define CONFIG_BT_NIMBLE_ACL_BUF_COUNT 12

/** @brief ACL Buffer size */
#define CONFIG_BT_NIMBLE_ACL_BUF_SIZE 255

/** @brief HCI Event Buffer size */
#define CONFIG_BT_NIMBLE_HCI_EVT_BUF_SIZE 70

/** @brief Number of high priority HCI event buffers */
#define CONFIG_BT_NIMBLE_HCI_EVT_HI_BUF_COUNT 30

/** @brief Number of low priority HCI event buffers */
#define CONFIG_BT_NIMBLE_HCI_EVT_LO_BUF_COUNT 8

/**
 * @brief Sets the number of MSYS buffers available.
 * @details MSYS is a system level mbuf registry. For prepare write & prepare \n
 * responses MBUFs are allocated out of msys_1 pool. This may need to be increased if\n 
 * you are sending large blocks of data with a low MTU. E.g: 512 bytes with 23 MTU will fail. 
 */
#define CONFIG_BT_NIMBLE_MSYS1_BLOCK_COUNT 12
```
My first thought is to try increasing the HCI_EVT settings, try doubling them and see what happens.
This is all speculation without seeing logs from the peripheral however so trial and error is all I can suggest.

Calypso971, 2020-09-11:
Well without details no answer I think. This is a strange thing happening, and that only motivates me more to find out what's going on. 

Doubled the values a few times, and beside the ACL Buffer Size it didn't have any effect. However, the ACL Buffer Size doubled means a crash/reboot of the ESP, so I left that one on 255. 

Also have been playing a bit with the MTU and MSYS settings, also without result. Going to continue tomorrow.

h2zero, 2020-09-11:
>This is a strange thing happening, and that only motivates me more to find out what's going on.

I like this attitude 😄 . I'm very curious myself, tried to reproduce this issue but have been unable.

Another thing you can try is removing the call to exchange MTU [here](https://github.com/h2zero/NimBLE-Arduino/blob/213ad0d4d60aa53d880970bd6dbfa4ab6396e64b/src/NimBLEClient.cpp#L719), you would also need to then change [this](https://github.com/h2zero/NimBLE-Arduino/blob/213ad0d4d60aa53d880970bd6dbfa4ab6396e64b/src/NimBLEClient.cpp#L738) to `rc = 0; break;`.

Also I have had issues in the past with [this line](https://github.com/h2zero/NimBLE-Arduino/blob/213ad0d4d60aa53d880970bd6dbfa4ab6396e64b/src/nimble/host/src/ble_gap.c#L1691). You could try commenting that out.

Calypso971, 2020-09-12:
Succes. Bypassing the MTU exchange did it. Even though it's very early, I'm taking a glass on it and will try to look into a possible solution later this weekend. Thanks for the help upto now!

h2zero, 2020-09-12:
Glad to hear there is progress on this. Of course I now wonder why that causes the issue....

Calypso971, 2020-09-14:
Yeah, want to know it myself too ofcourse. When looking at what's happening in MTU exchange:

08:35:48.066 -> GATT procedure initiated: exchange mtu
08:35:48.066 -> host tx hci data; handle=0 length=7
08:35:48.066 -> ble_hs_hci_acl_tx(): 0x00 0x00 0x07 0x00 0x03 0x00 0x04 0x00 0x02 0x80 0x00 
08:35:48.066 -> txed att command: mtu req; conn=0 mtu=128
This is a request to change MTU to 128, right?

08:35:48.066 -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
08:35:48.099 -> 0x16 0x20 0x02 0x00 0x00 
08:35:48.099 -> Command Status: status=0 cmd_pkts=5 ocf=0x16 ogf=0x8
08:35:48.099 -> Number of Completed Packets: num_handles=1
08:35:48.099 -> handle:0 pkts:1
08:35:48.099 -> ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=7 data=0x03 0x00 0x04 0x00 0x03 0xf7 0x00 
08:35:48.099 -> rxed att command: mtu rsp; conn=0 mtu=247
This is the response to put the MTU to 247 (strange value IMHO)

08:35:48.099 -> D NimBLEClient: "Got Client event BLE_GAP_EVENT_MTU"
08:35:48.099 -> I NimBLEClient: "mtu update event; conn_handle=0 mtu=128"

And now it's updated to 128?
In my opinion this doesn't match up - I request 128, I set it to 128, and in between I get an answer saying that it is 247.
Maybe there's a logical explanation for this though.

Another time I got about the same sequence, but then it seems to have negotiated to 247 (again 247):

13:12:21.289 -> txed att command: mtu req; conn=0 mtu=256
13:12:21.289 -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2

13:12:21.359 -> rxed att command: mtu rsp; conn=0 mtu=247

13:12:21.359 -> I NimBLEClient: "mtu update event; conn_handle=0 mtu=247"

Now the stranger part: from another device (that works) I get exactly the same result. So I'm pretty stuck here.

h2zero, 2020-09-14:
>  I request 128, I set it to 128, and in between I get an answer saying that it is 247.
Maybe there's a logical explanation for this though.

Yes the explanation is that we send our MTU (128) and the peripheral sends it's MTU (247) and as per spec the lowest of the 2 values is used for the connection MTU. All seems to be working as intended there.

h2zero, 2020-12-29:
Closing this as stale. Please re-open if the issue is not yet resolved.

closed 2020-12-29 by h2zero