h2zero/NimBLE-Arduino #127: RemoteCharacteristic by handle?

gkoh, 2020-09-15:
This is perhaps related to #111, however, it is unclear to me if a 16-bit UUID is supposed to be a handle? I can't find anything definitive.
Right now, I am using the full 128-bit UUID for characteristics, but they're quite unwieldy, using the 16-bit handle would be nice.

I see in searching around there used to be a map from handle to UUID and I can certainly something similar by searching through the service iterator.

What is the current recommended practice for getting a remote characteristic by handle from a remote service?

h2zero, 2020-09-15:
Hi @gkoh the services by handle map was removed due to storing large amounts of duplicate data in a memory inefficient container. 

BLE handles are 16 bit index values assigned to each attribute by the host on the peripheral device at run time. You can access these values via `RemoteCharacteristic::getHandle()`. 

My suggestion at this time if you wish to use handles is to bond with the device and store the <handle, UUID> database in persistent storage. By doing this however you may need to consider handling the possible Service Changed notifications to re-acquire the database if it is changed by the peripheral.

May I ask what it is that makes the UUID's difficult to work with? Perhaps there is something that can be improved in this regard.

gkoh, 2020-09-16:
> May I ask what it is that makes the UUID's difficult to work with? Perhaps there is something that can be improved in this regard.

Good question, made me think about it properly (ie. am I really that pathetic that I can't type 32 characters vs 4).
My current use case is reverse engineering a BLE device, so I get packet traces and run them through Wireshark.
Wireshark, and it seems a lot of stuff in this space, tends to focus on the handles rather than the UUIDs.

During prototyping, I was using btstack, which did have handle access to characteristics, so it was natural to look at the Wireshark trace and use the handle without caring about service and characteristics UUIDs.

I think in my use case, if a call like `NimBLEClient::getCharacteristic(uint16_t handle)` was available, I'd be very comfortable.

wakwak-koba, 2020-09-17:
I am the person who posted the # 111 issue.
( Waiting for the arrival of other devices used for testing... )

The internal behavior of `pClient->getService(uuid_service)` depends on whether `pClient->getServices(true)` was pre-executed or not.
If you have not previously run `pClient->getServices(true)`, service search is performed using `ble_gattc_disc_svc_by_uuid()`.

I suspect the behavior of `ble_gattc_disc_svc_by_uuid()` is suspicious.

h2zero, 2020-09-17:
@gkoh 
> My current use case is reverse engineering a BLE device, so I get packet traces and run them through Wireshark.
Wireshark, and it seems a lot of stuff in this space, tends to focus on the handles rather than the UUIDs.

Interesting use case, yes other than the client retrieving the handles for the uuid's BLE does not use the uuid to transact. 
The difficulty in using the handles as you suggest is every service, characteristic and descriptor has a 16 bit handle and in your use case there is no guarantee that the handle you're passing is actually a characteristic handle also it could be the characteristic definition handle or value handle as well.

I would suggest writing a function that parses the vector and searching the handles for the characteristic or other attribute you're interested in. Of course you would need to call `discoverAttributes();` first.

h2zero, 2020-09-17:
@wakwak-koba 
> I suspect the behavior of ble_gattc_disc_svc_by_uuid() is suspicious.

Interesting suspicion I'm curious, do you have a log showing any info? Please post it in #111.

wakwak-koba, 2020-09-18:
I don't have any evidence yet, but I feel that the internal behavior is different between scanning the result obtained by `ble_gattc_disc_all_svcs()` and specifying it by `ble_gattc_disc_svc_by_uuid()`.
I would like to find out a little more if this is device dependent.

gkoh, 2020-09-18:
> I would suggest writing a function that parses the vector and searching the handles for the characteristic or other attribute you're interested in. Of course you would need to call `discoverAttributes();` first.

For now, everything I've needed to fiddle is definitely a characteristic, so yes, I have everything I need as the iterators are conveniently provided, so I tried:
```
NimBLERemoteCharacteristic* getCharacteristic(uint16_t handle) {
  NimBLERemoteService *pSvc = nullptr;

  for (auto it = m_Client->begin(); it != m_Client->end(); ++it) {
    if ((*it)->getStartHandle() <= handle && handle <= (*it)->getEndHandle()) {
      pSvc = *it;
      break;
    }
  }

  if (pSvc != nullptr) {
    for (auto it = pSvc->begin(); it != pSvc->end(); ++it) {
      if ((*it)->getHandle() == handle) {
        return *it;
      }
    }
  }

  return nullptr;
}
```

However, it fails to compile because `getStartHandle()` and `getEndHandle()` are private.
Of course, I could remove that optimisation and search __every__ service, but thought I'd check if there's any reason they can't be made public?

h2zero, 2020-09-18:
>  if there's any reason they can't be made public?

No reason they can't be, just didn't anticipate them being used in app code.

gkoh, 2020-09-19:
> > if there's any reason they can't be made public?
> 
> No reason they can't be, just didn't anticipate them being used in app code.

We can avoid exposing them, I can submit a PR adding this to NimBLEClient?

gkoh, 2020-09-19:
> > > if there's any reason they can't be made public?
> > 
> > 
> > No reason they can't be, just didn't anticipate them being used in app code.
> 
> We can avoid exposing them, I can submit a PR adding this to NimBLEClient?

I just went ahead and raised a PR because it was straightforward, feel free to close this all out if you don't want/need it.
I'm very happy with everything already :smiley:

gkoh, 2020-09-25:
Thanks for merging, now I'm really, really happy with choosing this library!

closed 2020-09-25 by gkoh