h2zero/NimBLE-Arduino #128: Incomplete payload in NimBLEAdvertisedDevice

corneliusmunz, 2020-09-15:
During the Scan process of my library  i detect that the behaviour of the payload in the NimBLE lib is different than in the former BLE lib. The payload contains only the last part of the overall payload and it seems to be that the payload is overwritten. The address to the pointer remains the same but the content and content length differs. 

payload of "old" non NimBLE based lib:
```
2-1-6-11-7-23-D1-BC-EA-5F-78-23-16-DE-EF-12-12-23-16-0-0-9-FF-97-3-0-41-7-80-63-0-5-12-10-0-20-0-2-A-0-B-9-6D-79-54-72-61-69-6E-48-75
```

payload of NimBLE-Arduino lib:
```
5-12-10-0-20-0-2-A-0-B-9-6D-79-54-72-61-69-6E-48-75
```

The payload is fetched after the callback is executed with the ```NimBLEAdvertisedDevice``` ```getPayload()```function

I have enabled the log files and added some log messages (payload address and payload length) with the following output
```
13:43:34.066 -> I NimBLEScan: "NEW DEVICE FOUND: 90:84:2b:03:19:7f"
13:43:34.066 -> D NimBLEAdvertisedDevice: "payload address: 3ffc86ac"
13:43:34.066 -> D NimBLEAdvertisedDevice: "payload length: 31"
13:43:34.066 -> D NimBLEAdvertisedDevice: "payload: 020106110723d1bcea5f782316deef12122316000009ff9703004107806300"
13:43:34.066 -> I NimBLEScan: "UPDATING PREVIOUSLY FOUND DEVICE: 90:84:2b:03:19:7f"
13:43:34.066 -> D NimBLEAdvertisedDevice: "payload address: 3ffc86ac"
13:43:34.102 -> D NimBLEAdvertisedDevice: "payload length: 21"
13:43:34.102 -> D NimBLEAdvertisedDevice: "payload: 051210002000020a000b096d79547261696e487562"
```
When looking at the address and the payload length (additional log messages added in the NimBLE lib) it seems that the original location is overwritten. I seems that the ```parseAdvertisement``` function is called twice before the callback function is executed. 

At the end i could solve my issue in direct access the manufacturer data in the NimBLEAdvertisedDevice struct instead of parsing the payload by my own (which was not a good idea). But i think the payload should contain the complete Advertisement payload and not only the last part. That would be my expectation but i am not a BLE specialist and do not know the spec in detail. 

Here you can find my related issue: https://github.com/corneliusmunz/legoino/issues/21#

If you need someone to test, just give me a hint :-)

h2zero, 2020-09-15:
Hi @corneliusmunz this is actually somthing I've been thinking about (I may end up rewriting the whole class). 

This is caused by the difference in how the data is sent to us by NimBLE vs Bluedroid. NimBLE sends the advertisement packet data and the scan response data in 2 separate callback calls. What I did was try to mimic Bluedroid in a way that waits for the scan response data (if applicable) before notifying the app of the existence of the data. However, due to this, the payload will be overwritten by the scan response data, with the advertisement data being lost. The simple fix is to make a copy of the data from both packets in an array but that would cause us to have copies of the data in 3 places, 2 is bad enough 😄. 

My thinking for future releases is that I will change this to just copy the data from both packets once into a vector and parse it on demand when the app asks for it. This way the payload will be correct and we can reduce the member variable count and memory footprint of the NimBLEAdvertisedDevice class. This will take some time and testing though for sure.

I recommend using the manufacturer data for now since it's the safest way.

corneliusmunz, 2020-09-15:
@h2zero Thanks for your super fast response 👍🏻💪 I think copying data is not a good idea and in the end the needed information is available in the manufacturerData and serviceData properties. Storing it in the std::string is a little bit difficult to understand but the data is available. But maybe you can remove the getPayload function because for the user it is misleading. So I would not write it new but remove the function and mark it in the migration guide.

h2zero, 2020-09-16:
>  I think copying data is not a good idea

Agreed but necessary in this case, the host stack deletes it's copy after anyway though. 

> Storing it in the std::string is a little bit difficult to understand

Yes, this is legacy code from the original library which used this method of storage. I have found it to be quite useful as it's basically a vector with extra utilities so I kept it and keeps the API the same.

> But maybe you can remove the getPayload function because for the user it is misleading.

Agreed, it doesn't really serve it's intended purpose anymore.

Thanks for the feedback!

closed 2021-01-31 by h2zero