h2zero/NimBLE-Arduino #129: Settings interval with addData

sujeendran, 2020-09-17:
Hi,
I was trying to migrate my code from Bluedroid to NimBLE, but our android app started getting GATT 133 error while connecting. I tried also connecting via LightBlue(fails with status 133) and nRFConnect(works) apps. This doesnt happen with Bluedroid.

I used to use these function earlier to allow wider range of iPhone and Android devices to connect:
BLEAdvertising::setMinPreferred(0x06)
BLEAdvertising::setMaxPreferred(0x12)

But in migration doc you mentioned now this has to be done via addData. How can i do it? I'm not well verse with BLE advertisement protocol. 

I also tried with the setMinInterval and setMaxInterval functions which causes advertising to fail.

h2zero, 2020-09-17:
Hi @sujeendran, sorry you're having this issue, it is not something I have encountered yet. I looked up the 133 error code and sadly it does not tell us much. Description from punch through:
> Some of the error status codes are documented as BluetoothGatt’s public constants, but the single most common error we’ve encountered while attempting to connect to a BluetoothDevice has to be the infamous status 133 error. Not only is this error undocumented under the BluetoothGatt list of constants, but a look inside the Android source code reveals its name as simply GATT_ERROR (0x85, which is hex for 133). You’re not alone if you think this is super vague. We strongly believe Google can do better here.

>Aside from the random occurrences of 133, we’ve generally seen error 133 happen most frequently in one of the two following scenarios:

>The BluetoothDevice we’re trying to connect to is no longer advertising or within Bluetooth range, and the connectGatt() call has timed out after about 30 seconds of trying to connect to it with the autoConnect argument set to false.
The firmware running on the BLE device has rejected the connection attempt by Android.
Regardless of which error status code you got in your onConnectionStateChange() callback, the recovery flow typically goes like this:

>Call close() on the BluetoothGatt object to indicate that we’re done with this BluetoothGatt instance, and that the system can release any pending resources.
Null out any references to this BluetoothGatt object.
If status code is either GATT_INSUFFICIENT_ENCRYPTION or GATT_INSUFFICIENT_AUTHENTICATION: call createBond() first and wait for the bonding process to succeed first before calling connectGatt() again. We’ll go through how to initiate a bond with a BLE device in a later section.
For other status codes: either surface an error to the user, or attempt to reconnect again silently for a few times before giving up. The latter is actually a viable strategy, considering the fact that error 133 can sometimes be very random and occur a few times in a row before the connection attempt eventually succeeds.  

So from that description the take away is we need get a log of what is happening from the peripheral side. If you could uncomment [this line in src/nimconfig.h](https://github.com/h2zero/NimBLE-Arduino/blob/d649b6b11f0059e5f24d1f4c2e07df1b1e01c4ab/src/nimconfig.h#L74) and get a log of the connection that would greatly help diagnose this issue.

> I used to use these function earlier to allow wider range of iPhone and Android devices to connect:
BLEAdvertising::setMinPreferred(0x06)
BLEAdvertising::setMaxPreferred(0x12)

These parameters don't do much, they simply advertise the min and max connection interval parameters that the device prefers to use which can be and usually are overridden by the connecting client anyway.  I'm not very confident that implementing those will solve the issue but we can try that after seeing the log.
> I also tried with the setMinInterval and setMaxInterval functions which causes advertising to fail.   

These will change the frequency of sending advertisements and will not affect connections. The advertising failed in this case because, if using the values above, the interval would be below spec minimum (0x20).

Does this issue occur if using the examples (the basic server example in the the refactored_examples folder would be best for this)?

sujeendran, 2020-09-17:
@h2zero Hi! Thanks for the response. Yes I have read several posts about this 133 error, its really troublesome. I will try to dig in the debug and try to find something.
In the mean time i reduced the security setting to no bond, no mitm and no sc and it worked without any issue:
NimBLEDevice::setSecurityIOCap(BLE_HS_IO_NO_INPUT_OUTPUT);
NimBLEDevice::setSecurityAuth(false, false, false);
NimBLEDevice::setSecurityInitKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);

I had tried earlier with (true,true,true) and (false,false,true) both of which ended up in 133. 
Does the third line from above actually serve any purpose if SC is set to false?

> Does this issue occur if using the examples (the basic server example in the the refactored_examples folder would be best for this)?

I havent tried yet. I will let you know if i manage to run it. The thing is I actually am running an esp-idf project that has arduino-esp32 as component for few other dependent modules. I wasn't able to build the esp-idf version of your library when I had arduino also as a component even when I set all the BT/BLE components from arduino-esp32 to be not included in the build. So i managed to add this library into the arduino-esp32 libraries and disabled the normal BLE library from building. I know it's a tricky workaround, but it builds and runs!

sujeendran, 2020-09-17:
@h2zero Would you still be able to let me know how i can manually add this data to advertising if i wanted to?
> These parameters don't do much, they simply advertise the min and max connection interval parameters that the device prefers to use which can be and usually are overridden by the connecting client [...]

h2zero, 2020-09-17:
> In the mean time i reduced the security setting to no bond, no mitm and no sc and it worked without any issue:
NimBLEDevice::setSecurityIOCap(BLE_HS_IO_NO_INPUT_OUTPUT);
NimBLEDevice::setSecurityAuth(false, false, false);
NimBLEDevice::setSecurityInitKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);

Interesting discovery, what were these set to initially? 

Is this an older phone you're connecting with? It's possible there is some problem related to secure connections (setting to to false here forces legacy pairing). The third line does have an effect in that it indicates that we will provide an IRK when pairing, used to resolve random addresses, shouldn't be needed unless you are using a random address. This might be something your android app expects by default, would be interesting to test on it's own.

> Would you still be able to let me know how i can manually add this data to advertising if i wanted to?

The answer to this depends on how your other advertising parameters are set. If you are not using `NimBLEAdvertisementData` class and passing that to `setAdvertisementData` then you will need to convert your settings to use this way of setting the data.

However if you just want to do a quick test you can create a static global in NimBLEAdvertising, for instance: `static uint8_t _itvalParams[4] = {0x00,0x06,0x00,0x12};`
Then in the NimBLEAdvertising constructor add `m_advParams.slave_itvl_range = &_itvalParams;`

>  I wasn't able to build the esp-idf version of your library when I had arduino also as a component even when I set all the BT/BLE components from arduino-esp32 to be not included in the build. 

If you use my fork of esp-nimble [here](https://github.com/h2zero/esp-nimble-component) and leave the Arduino BLE stuff stock, along with the idf version of this repo it works (that's what that repo was for). Glad you got it working though 👍. For getting the NimBLE stack logs, you'll have to enable them in menuconfig since you're using IDF.

sujeendran, 2020-09-18:
@h2zero 
>I had tried earlier with (true,true,true) and (false,false,true) both of which ended up in 133.

>Is this an older phone you're connecting with?

I am using Android 10 device with flagship specs. Also tried OnePlus 5 running on Android 9. Same result.

>However if you just want to do a quick test you can create a static global in NimBLEAdvertising, for instance: static uint8_t _itvalParams[4] = {0x00,0x06,0x00,0x12};
Then in the NimBLEAdvertising constructor add m_advParams.slave_itvl_range = &_itvalParams;

I tried it and wasnt able to build it:
[image]
Should it be added to ble_gap_adv_params struct before?

>If you use my fork of esp-nimble here and leave the Arduino BLE stuff stock, along with the idf version of this repo it works (that's what that repo was for).

I noticed the forked component is for idf 3.2+3.3. I'm running idf 4.0.1, which is why i used the cpp version of your library first from [here](https://github.com/h2zero/esp-nimble-cpp) leading to the issue i mentioned earlier while building with arduino.

h2zero, 2020-09-18:
Oops sorry about that, should have been `m_advData.slave_itval_range`.

> I noticed the forked component is for idf 3.2+3.3. I'm running idf 4.0.1

That makes sense, I'm assuming you're using the idf 4.0 arduino branch? I have not tested while using arduino with 4.0. Thanks for the heads up to try it, I'll document the changes needed.

wmarkow, 2020-11-28:
I also have the same issue. My ESP32 project uses NimBLE-Arduino library and my Android application can't connect.
For me it started to work when I did in my Android application as described [in this comment](https://github.com/android/connectivity-samples/issues/18#issuecomment-374600798)
@sujeendran, maybe it will also work for you. I'm not really happy with this solution because it will work since Android 6 and I wanted to have my app to work with Android 5.

The issue seems to be not reproducible when I use the internal BLE library from ESP32.

wmarkow, 2020-11-29:
**NimBLE_Server** example works nice for me because it uses a custom **ServerCallbacks**, it goes about this method:
```
  void onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc) {
	/** We can use the connection handle here to ask for different connection parameters.
	 *  Args: connection handle, min connection interval, max connection interval
	 *  latency, supervision timeout.
	 *  Units; Min/Max Intervals: 1.25 millisecond increments.
	 *  Latency: number of intervals allowed to skip.
	 *  Timeout: 10 millisecond increments, try for 5x interval time for best results.  
	 */
    pServer->updateConnParams(desc->conn_handle, 24, 48, 0, 60);
  };
```
When I comment it out then it doesn't work on my Android 5.1.1. I do not understand the meaning of this ```pServer->updateConnParams(desc->conn_handle, 24, 48, 0, 60)``` but at least I have found that this line helps.

To summarise, in my test environment:
 * I set a custom server callback and implement ```onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc)``` as above
 * in my custom Android app for devices with Android < API 23 I call [this connectGatt](https://developer.android.com/reference/android/bluetooth/BluetoothDevice#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback))
 * in my custom Android app with Android >= API 23 I call [this connectGatt added in API 23](https://developer.android.com/reference/android/bluetooth/BluetoothDevice#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback,%20int)) with ```BluetoothDevice.TRANSPORT_LE``` as a last parameter

With those changes together both **BLE Scanner** and my custom Android app connect to the **NimBLE_Server**.

commit 5cb454bf (2020-12-27): Re-introduce NimBLEAdvertising::setMin/MaxPreferred. Resolves #129

This implements the functionality of the original library min/max preferred connection parameters settings.
closed 2020-12-27 by h2zero