h2zero/NimBLE-Arduino #130: Getting garbage occasionally when obtaining the Device Mac address using getDevice method

davidkjackson54, 2020-09-17:
When I run this - 
**std::string deviceAddress = foundDevices.getDevice(i).getAddress().toString();**  // get MAC address

I occasionally get this garbage in the returned Mac address. Here are 2 scans taken a couple of seconds apart of the same device. The first displays fine but the second is corrupt.

Message arrived on topic: scan/tilts Message: {}
Sending..
{"tilts":[{"colour":7,**"mac":"d2:fe:0e:a1:90:45"**,"uuid":"a495bb70-c5b1-4b44-b512-1370f02d74de","temp":22.77778,"gravity":1.026,"rssi":-62}]}
Message arrived on topic: scan/tilts Message: {}
Sending..
{"tilts":[{"colour":7,**"mac":"ad"**,"uuid":"a495bb70-c5b1-4b44-b512-1370f02d74de","temp":22.77778,"gravity":1.026,"rssi":-63}]}

h2zero, 2020-09-17:
My first thought is print overwrite if using `Serial()` with debug messages also printing, sometimes these interfere. Maybe try printing the message twice and see if it occurs in both. If not I would need to find a way to reproduce this.

davidkjackson54, 2020-09-17:
The actual code is pretty simple:
I commented out the Serial.println statements.  The corrupt value is being returned in the MQTT message.
  
`std::string deviceAddress = foundDevices.getDevice(i).getAddress().toString();  // get MAC address
 int deviceRSSI = foundDevices.getDevice(i).getRSSI();                           // get RSSI         
 String x = devices[m_count]["mac"];                                                      // check if a slot is available
 if (x == "null") {
          devices[m_count]["colour"] = parseColour(devData);                  // extract Color offset
          devices[m_count]["mac"] = deviceAddress.c_str();                       // MAC address
          devices[m_count]["uuid"] = parseUUID(devData);                       // extract UUID
          devices[m_count]["temp"] = parseTemp(devData);                     // extract temperature in C
          devices[m_count]["gravity"] = parseGravity(devData);                // extract gravity
          devices[m_count]["rssi"] = deviceRSSI;                                        // RSSI
          m_count++;
        }
     }
    } 
  char buffer[1200];
  if (m_count) {
       serializeJson(outdoc, buffer);
       //Serial.println("Sending..");
       Serial.println(buffer);  
       client.setBufferSize(1200);
       client.publish("scan/tilts/done", buffer); 
       client.setBufferSize(256);                    // reset to default`
  }`

Here I ran it 4 times in sequence and I got the corruption once on the second attempt of the 4. 
{"tilts":[{"colour":7,"mac":"d2:fe:0e:a1:90:45","uuid":"a495bb70-c5b1-4b44-b512-1370f02d74de","temp":23.88889,"gravity":1.037,"rssi":-61}]}
Message arrived on topic: scan/tilts Message: {}
{"tilts":[{"colour":7,**"mac":"62ece0"**,"uuid":"a495bb70-c5b1-4b44-b512-1370f02d74de","temp":23.88889,"gravity":1.037,"rssi":-61}]}
Message arrived on topic: scan/tilts Message: {}
{"tilts":[{"colour":7,"mac":"d2:fe:0e:a1:90:45","uuid":"a495bb70-c5b1-4b44-b512-1370f02d74de","temp":23.88889,"gravity":1.037,"rssi":-61}]}
Message arrived on topic: scan/tilts Message: {}
{"tilts":[{"colour":7,"mac":"d2:fe:0e:a1:90:45","uuid":"a495bb70-c5b1-4b44-b512-1370f02d74de","temp":23.88889,"gravity":1.037,"rssi":-70}]}

h2zero, 2020-09-17:
Quite odd, the `.toString()` call should always return an 18 byte string regardless of the content, it seems to get cut off here. I will need to contemplate on this.

davidkjackson54, 2020-09-17:
I guess it depends where it finds a rogue  null character to terminate it?

h2zero, 2020-09-17:
Actually it seems the data gets corrupted somewhere before you send it, if it was just cut off the format would still be `xx:xx`, your second log does not show any ':' between the bytes. Without seeing a code snippet it's hard to diagnose.

davidkjackson54, 2020-09-17:
I hope this helps. It is pretty simple code.
`
Serial.println("Scanning...");
    
    BLEScanResults foundDevices = pBLEScan->start(SCAN_TIME);
    m_deviceCount = foundDevices.getCount();
    pBLEScan->stop();
    BLEDevice::deinit(0);

    const size_t capacity = JSON_ARRAY_SIZE(8) + JSON_OBJECT_SIZE(1) + 8*JSON_OBJECT_SIZE(6) +430;
    DynamicJsonDocument outdoc(capacity); 
    JsonArray devices = outdoc.createNestedArray("tilts");

    for (uint32_t i = 0; i < m_deviceCount; i++)  {
      BLEAdvertisedDevice Device = foundDevices.getDevice(i);
                  
      String devString = Device.toString().c_str();
      String devData = devString.substring(63);                              // a495bb70c5b14b44b5121370f02d74de004c0424c5 
      if (devData.substring(7, 32) == "0c5b14b44b5121370f02d74de") {        // Tilt found

        std::string deviceAddress = foundDevices.getDevice(i).getAddress().toString();  // get MAC address
        int deviceRSSI = foundDevices.getDevice(i).getRSSI();                           // get RSSI         

        String x = devices[m_tiltCount]["mac"];                           // check if a field is already filled
        if (x == "null") {
          devices[m_tiltCount]["colour"] = parseColour(devData);          // extract Tilt Colour offset e.g. Yellow = 7
          devices[m_tiltCount]["mac"] = deviceAddress.c_str();            // MAC address
          devices[m_tiltCount]["uuid"] = parseUUID(devData);              // extract UUID
          devices[m_tiltCount]["temp"] = parseTemp(devData);              // extract temperature in C
          devices[m_tiltCount]["gravity"] = parseGravity(devData);        // extract gravity
          devices[m_tiltCount]["rssi"] = deviceRSSI;                      // RSSI
           m_tiltCount++;
        }
     }
    } 
  char buffer[1200];
  if (m_tiltCount) {
    //serializeJson(outdoc, Serial);                  // just tempo for testing
    
      serializeJson(outdoc, buffer);
      Serial.println("Sending..");
      Serial.println(buffer);  
      client.setBufferSize(1200);
      client.publish("scan/tilts/done", buffer); 
      client.setBufferSize(256);                    // reset to default
  }  
  
  pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
`

h2zero, 2020-09-17:
Thanks for posting that.

I don't see anything immediately obvious there. I'll try to reproduce the error using your code as a template. If you could try something in the meantime I'd be interested in the results.

After the line you obtain the address string could you test the length and print the result?

davidkjackson54, 2020-09-18:
I added this code to display the MAC address before ArduinoJSON gets involved:
std::string deviceAddress = foundDevices.getDevice(i).getAddress().toString();  // get MAC address
int deviceRSSI = foundDevices.getDevice(i).getRSSI();                           // get RSSI         
Serial.println("length of DeviceAddress is ");
Serial.println(deviceAddress.length());
Serial.println(deviceAddress.c_str());

Message arrived on topic: scan/tilts Message: {}
Scanning...
length of DeviceAddress is 
**17
d2:fe:0e:a1:90:45**
Sending..
{"tilts":[{"colour":7,**"mac":"xV��i7�?�8�? "**,"uuid":"a495bb70-c5b1-4b44-b512-1370f02d74de","temp":27.22222,"gravity":1.018,"rssi":-61}]}

So something is going on with this statement:
devices[m_tiltCount]["mac"] = deviceAddress.c_str();            // MAC address

Iam going to have to look into this some more..

davidkjackson54, 2020-09-18:
Hmmm interestingly.

I commented out this  line
//std::string deviceAddress = foundDevices.getDevice(i).getAddress().toString();  // get MAC address 
 and changed it to this
String deviceAddress = foundDevices.getDevice(i).getAddress().toString().c_str();

and this changed this:
//devices[m_tiltCount]["mac"] = deviceAddress.c_str();            // MAC address
to this:
devices[m_tiltCount]["mac"] = deviceAddress;  

and so far after 10 minutes of repeated testing , the corruption is now not occurring. 

I will test it again tomorrow to be sure... 
Very odd... !

h2zero, 2020-09-18:
I'm going to assume the array element is a pointer to String? That would explain it, once the code block exits the std::string destructor is called and the pointer becomes invalid. Not sure why it works with String in that case though, might just be how it's memory is allocated/deallocated.

davidkjackson54, 2020-09-18:
I do not see any usage that has gone out of scope.  I define std.string deviceAddress then use it within an If block .
`std::string deviceAddress = foundDevices.getDevice(i).getAddress().toString();  // get MAC address
    int deviceRSSI = foundDevices.getDevice(i).getRSSI();                                     // get RSSI         

    String x = devices[m_tiltCount]["mac"];                                                          // check if a field is already filled
    if (x == "null") {
      devices[m_tiltCount]["colour"] = parseColour(devData);                             // extract offset
     devices[m_tiltCount]["mac"] = deviceAddress.c_str();                                   // MAC address
      devices[m_tiltCount]["uuid"] = parseUUID(devData);                                  // extract UUID
      devices[m_tiltCount]["temp"] = parseTemp(devData);                                // extract temperature in C
      devices[m_tiltCount]["gravity"] = parseGravity(devData);                          // extract gravity
      devices[m_tiltCount]["rssi"] = deviceRSSI;                                                  // RSSI
       m_tiltCount++;
    }`

h2zero, 2020-09-18:
I was referring to the assignment in `devices[m_tiltCount]["mac"] = deviceAddress.c_str(); `. If that is only storing a pointer to the address string (not sure, as I'm not familiar with the array structure) and is used out of scope, which it appears to be at the call `serializeJson(outdoc, buffer);` it would lead to this issue. 

Seems like a plausible cause to me, I could be wrong.

davidkjackson54, 2020-09-18:
I will check it out again. Either way it doesn't appear to be an issue with getDevice().

thanks

closed 2020-09-18 by davidkjackson54